dresden-elektronik/deconz-rest-plugin #3898: Raspbee II not recognized in the HA deCONZ addon

bsfaxi, 2020-12-13:
## Describe the question or issue you are having

I unfortunately still have the same issue here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3681 !
I'm using the latest version of deCONZ 2.7.01.
On a plain Raspbian installation, I updated my Raspbee II to the latest firmware file available "deCONZ_RaspBeeII_0x26680700.bin.GCF". It works there perfectly. The Raspbee II is recognized correctly.
On a second RPI4 (same specs) with Home Assistant deCONZ add-on, I still have the same mentioned issue's symptoms, knowing that it was working for me before the addon upgrade.

In the config file, I have the required lines:
enable_uart=1
dtoverlay=pi3-miniuart-bt
force_turbo=1

## Screenshots

The HA hardware system shows:
[image]

## Environment

 - Host system: Raspberry Pi 4B
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: 26680700
 - deCONZ version: 2.7.01
 - Device: RaspBee II
 - Do you use an USB extension cable: Non relevant
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

The deCONZ debug window shows:
11:01:53:577 HTTP Server listen on address 0.0.0.0, port: 40850, root: /usr/share/deCONZ/webapp/
11:01:53:600 CTRL. 3.27.2
11:01:53:736 dev /dev/ttyAMA0
11:01:53:736 COM: /dev/ttyAMA0 / serialno: , RaspBee
11:01:53:737 ZCLDB init file /data/.local/share/dresden-elektronik/deCONZ/zcldb.txt
11:01:53:953 parent process s6-supervise
11:01:53:953 gw run mode: docker/hassio
11:01:53:954 GW sd-card image version file does not exist: /data/.local/share/dresden-elektronik/deCONZ/gw-version
11:01:53:954 sd-card cid: 1b534d454434515430ffffffff014200
11:01:53:988 DB sqlite version 3.27.2
11:01:53:991 DB PRAGMA page_count: 33
11:01:53:992 DB PRAGMA page_size: 4096
11:01:53:992 DB PRAGMA freelist_count: 0
11:01:53:992 DB file size 135168 bytes, free pages 0
11:01:53:992 DB PRAGMA user_version: 7
11:01:53:992 DB cleanup
11:01:54:001 DB create temporary views
11:01:54:029 started websocket server at port 8081
11:01:54:042 [INFO] - Found file containing button maps. Parsing data...
11:01:54:048 [WARNING] - Button map 'sunricherCCTMap' in JSON file has no assigned ModelIDs. Skip loading button map...
11:01:54:056 [INFO] - Button maps loaded.
11:01:54:058 dlg action: Read binding table
11:01:54:058 found node plugin: libde_rest_plugin.so - REST API Plugin
11:01:54:063 found node plugin: libde_signal_plugin.so - Signal Monitor Plugin
11:01:55:715 found node plugin: libstd_otau_plugin.so - STD OTAU Plugin
11:01:55:757 dev /dev/ttyAMA0
11:01:55:757 COM: /dev/ttyAMA0 / serialno: , RaspBee
11:01:56:804 dev /dev/ttyAMA0
11:01:56:805 COM: /dev/ttyAMA0 / serialno: , RaspBee
11:01:56:836 COM check bootloader
11:01:58:840 COM timout query bootloader, assume application
11:01:59:801 dev /dev/ttyAMA0
11:02:00:284 Announced to internet https://phoscon.de/discover
11:02:00:284 discovery server date: Sun, 13 Dec 2020 10:02:00 GMT
11:02:00:285 local time seems to be ok
11:02:00:285 discovery found version 2.04.35 for update channel stable
11:02:00:403 device state timeout ignored in state 2
11:02:03:403 device state timeout ignored in state 2
...

## Additional context

bsfaxi, 2020-12-13:
I also ssh to the Home Assistant core and did the following:
# docker exec -it addon_core_deconz bash
root@core-deconz:/# ls /usr/share/deCONZ/firmware/
deCONZ_ConBeeII_0x26580700.bin.GCF  deCONZ_ConBeeII_0x26680700.bin.GCF   deCONZ_RaspBeeII_0x26680700.bin.GCF  deCONZ_Rpi_0x26390500.bin.GCF
deCONZ_ConBeeII_0x26660700.bin.GCF  deCONZ_RaspBeeII_0x26660700.bin.GCF  deCONZ_Rpi_0x26350500.bin.GCF
root@core-deconz:/# GCFFlasher_internal -f /usr/share/deCONZ/firmware/deCONZ_RaspBeeII_0x26680700.bin.GCF
GCFFlasher V3_17 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device RaspBee (RaspBee)

952: Error: failed to open device, status: Permission error while locking the device (2)
root@core-deconz:/# 

It seems the Raspbee II not recognized as a device...

bsfaxi, 2020-12-13:
If I try with the configuration "device: /dev/ttyS0" , I have the following in the log :
13:20:43:884 wait reconnect 1 seconds
13:20:43:924 dev /dev/ttyAMA0
13:20:43:925 dev /dev/ttyS0
13:20:43:925 COM: /dev/ttyS0 / serialno: , RaspBee
13:20:43:975 COM check bootloader
13:20:45:884 failed to reconnect to network try=3
13:20:45:980 COM timout query bootloader, assume application
13:20:47:410 device state timeout ignored in state 2

I logged inside the docker, and I have : 
root@core-deconz:/# GCFFlasher_internal -f /usr/share/deCONZ/firmware/deCONZ_RaspBeeII_0x26680700.bin.GCF
GCFFlasher V3_17 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device RaspBee (RaspBee)
action: reset device RaspBee
wiringPi 2.60 initialized
Bootloader version 0x00030003, app crc: 0x00000000

Nothing changed !

bsfaxi, 2020-12-13:
I finally ended up by succeeding flashing back to one old firmware available within the docker image.
deCONZ has then recognized the RaspbeeII !
And then flashing to the latest one !

root@core-deconz:/#  /usr/bin/GCFFlasher_internal.bin -d RaspBee -t 60 -f /usr/share/deCONZ/firmware/deCONZ_RaspBeeII_0x26660700.bin.GCF
GCFFlasher V3_17 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device RaspBee (RaspBee)
Bootloader version 0x00030003, app crc: 0x00000000
flashing 164106 bytes: |================|

root@core-deconz:/#  /usr/bin/GCFFlasher_internal.bin -d RaspBee -t 60 -f /usr/share/deCONZ/firmware/deCONZ_RaspBeeII_0x26680700.bin.GCF
GCFFlasher V3_17 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device RaspBee (RaspBee)
deCONZ firmware version 26660700
Bootloader version 0x00030003, app crc: 0x5CBFD214
Reboot device RaspBee (RaspBee)
action: reset device RaspBee
wiringPi 2.60 initialized
Bootloader version 0x00030003, app crc: 0x5CBFD214
flashing 164373 bytes: |================|
root@core-deconz:/#

pstranghoener, 2021-01-03:
@bsfaxi How do you flashed back to an old firmware?

bsfaxi, 2021-01-03:
> @bsfaxi How do you flashed back to an old firmware?
> 
> 

Do you mean "how I got inside the docker" running inside the HA OS?

pstranghoener, 2021-01-03:
Yes, could you explain, I have the same error .

bsfaxi, 2021-01-04:
Here you can find the official procedure to access your HA OS using port 22222: https://developers.home-assistant.io/docs/operating-system/debugging/.

On my side, I did the following:
**Generating SSH Keys, use "RSA" for HA, not "ed25519" :**
ssh-keygen -t rsa -b 4096 -C "xxxx@example.com"

Ouput:
	Generating public/private rsa key pair.
	Enter file in which to save the key (/Users/bsfaxi/.ssh/id_rsa): homeassistant.key
	Enter passphrase (empty for no passphrase): 
	Enter same passphrase again: 
	Your identification has been saved in homeassistant.key.
	Your public key has been saved in homeassistant.key.pub.
	The key fingerprint is:
	xxxxxxxxxx
	The key's randomart image is:
	+---[RSA 4096]----+
	xxxxxxx
	+----[SHA256]-----+

**Home Assistant OS access:**
This will generate 2 files: xxx.key and xxx.key.pub.
Use a USB drive formatted with FAT, ext4, or NTFS and name it CONFIG (case sensitive). Create an authorized_keys file (no extension) containing your public key, and place it in the root of the USB drive. File needs to be ANSI encoded (not UTF-8) and must have Unix line ends (LF), not Windows (CR LF). See Generating SSH Keys section below if you need help generating keys. From the UI, navigate to the Supervisor system page and choose "Import from USB". You can now access your device as root over SSH on port 22222. Alternatively, the file will be imported from the USB when the Home Assistant OS device is rebooted.
Attention: Make sure when you are copying the public key to the root of the USB drive that you rename the file correctly to authorized_keys with no .pub file extension.

You should then be able to SSH into your Home Assistant device. On Mac/Linux, use:
ssh root@homeassistant.local -p 22222
ha > login

**Accessing the Home Assistant container bash#:**
docker exec -it homeassistant /bin/bash

**Accessing the deCONZ container bash# to flash firmware:**
docker exec -it addon_core_deconz bash
Via VNC disconnected deCONZ from RaspBee → deCONZ disconnect button under "File" menu item.
root@core-deconz:/# GCFFlasher_internal -d RaspBee -t 60 -f /usr/share/deCONZ/firmware/deCONZ_RaspBeeII_0x26680700.bin.GCF 
or full path: /usr/bin/GCFFlasher_internal.bin -d RaspBee -t 60 -f /usr/share/deCONZ/firmware/deCONZ_RaspBeeII_0x26680700.bin.GCF

**PS:** if you have the message "Reboot device RaspBee (RaspBee)" continuously, that means that the Deconz app is trying to connect with the device and that you don't have a slot to execute the flashing program. In my environment, I restart the addon, quickly access the docker container and start the flashing before the Deconz app starts.

bsfaxi, 2021-01-04:
I'm re-opening this issue since I had the same issue yesterday when I upgraded the Deconz addon to the latest.
It seems that the problem is persistant after each upgrade!

bsfaxi, 2021-01-04:
> Yes, could you explain, I have the same error .

Please add a comment if this works for you.

Mimiix, 2021-01-04:
@bsfaxi I am not sure what you need from us. We don't maintain the addon so there's not much we can do here...

bsfaxi, 2021-01-04:
Hi @Mimiix, 
Sorry if I opened the issue in the wrong git. Do you advise to open it in the Home Assistant Git ?
BTW, it's the Deconz docker image that's used within HA... And with each new version upgrade (of Deconz), the problem is happening again, even with a fresh installation (seen at a friend). There is maybe a relation with the latest firmware? That's why I opened it first here...
Please advise.

Mimiix, 2021-01-04:
Can you check what happens with this firmware? http://deconz.dresden-elektronik.de/deconz-firmware/deCONZ_RaspBeeII_0x26690700.bin.GCF

bsfaxi, 2021-01-04:
Flashing is successfull and the deCONZ app connects directly. I reloaded the HA deCONZ Addon and it seems to be working great.

root@core-deconz:/usr/share/deCONZ/firmware# /usr/bin/GCFFlasher_internal.bin -d RaspBee -t 60 -f deCONZ_RaspBeeII_0x26690700.bin.GCF
GCFFlasher V3_17 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device RaspBee (RaspBee)
deCONZ firmware version 26680700
Bootloader version 0x00030003, app crc: 0x1E9498C2
Reboot device RaspBee (RaspBee)
action: reset device RaspBee
wiringPi 2.60 initialized
Bootloader version 0x00030003, app crc: 0x1E9498C2
Reboot device RaspBee (RaspBee)
action: reset device RaspBee
Bootloader version 0x00030003, app crc: 0x1E9498C2
flashing 164241 bytes: |================|deCONZ firmware version 26690700
root@core-deconz:/usr/share/deCONZ/firmware# 

deCONZ log:
19:47:21:997 wait reconnect 5 seconds
19:47:22:997 wait reconnect 4 seconds
19:47:23:997 wait reconnect 3 seconds
19:47:24:996 wait reconnect 2 seconds
19:47:25:996 wait reconnect 1 seconds
19:47:26:047 dev /dev/ttyAMA0
19:47:26:049 COM: /dev/ttyAMA0 / serialno: , RaspBee
19:47:26:059 COM check bootloader
19:47:28:073 COM timout query bootloader, assume application
19:47:28:124 Device firmware version 0x26690700 RaspBee II
19:47:28:136 unlocked max nodes: 200
19:47:28:360 Device protocol version: 0x010E
19:47:28:548 dev /dev/ttyAMA0
19:47:28:549 Current channel 25
19:47:28:554 GW firmware version: 0x26690700
19:47:28:594 CTRL ANT_CTRL 0x03
19:47:28:680 CTRL got stack debug assert code: 0x2021, type: 0x03
19:47:28:691 Device protocol version: 0x010E
19:47:28:827 Current channel 25
19:47:28:870 CTRL ANT_CTRL 0x03
19:47:28:956 CTRL got stack debug assert code: 0x2021, type: 0x03

I will keep an eye on the next Addon upgrade to see if the problem occurs again. 
Issue closed.
Thanks @Mimiix.

closed 2021-01-04 by bsfaxi

bsfaxi, 2021-01-11:
For those who stil have this kind of issue, I succeeded to solve the problem by setting the following lines to the file config.txt in the boot partition. You can do that from your laptop by inserting the SD card and modifying the file from there, or directly from HA OS if you succeed to ssh to the system on port 22222 in this subdirectory "/mnt/boot".

config.txt:

```
enable_uart=1                                                                 
dtoverlay=miniuart-bt                                                     
force_turbo=1  
```
Please pay attention to the value of the "**dtoverlay**" parameter for RPI 4B which is "**miniuart-bt**" and not "**pi3-miniuart-bt**" like we can find everywhere in many blogs.
I just taken some time to read this : https://www.raspberrypi.org/documentation/configuration/uart.md.

This has been done on a recently acquired RPI 4B by a friend. And the only working value for him was "**disable-bt**" which disables the Bluetooth. Setting the value to "**pi3-miniuart-bt**" was leading to the issue, and blocking access to the serial of the RPI.

The strange thing is that the value "**pi3-miniuart-bt**" is still working for me, and I have a RPI 4B (same model) acquired 6 months ago.
That's weird! I'm not a specialist, but I think it may be related to the hardware specs/versions/eeprom !!!.

My 2 cents.