dresden-elektronik/deconz-rest-plugin #3899: TS110F Tuya Smart Zigbee Dimmer Switch Module 1/2 Gang 220V With Neutral

KipK, 2020-12-13:
## Device

- Product name:  generic (Lonsonho ) Tuya Smart Zigbee Dimmer Switch Module 1/2 Gang 220V With Neutral
                            QS-Zigbee-D02-TRIAC-LN
- link: https://www.aliexpress.com/item/1005001590777455.html?spm=a2g0s.9042311.0.0.43fc6c37TqlQhK
- Manufacturer: Heimahn
- Model identifier: TS110F
- Device type :  Dimmer switch

## Screenshots
 
[image]
[image]
[image]
[image]
[image]

Device is recognized, pairing successfull and mostly fonctionnal.
However, scene transition, and dimming doesn't work.
Only changing brightness to a value will works ok, but dimming using a remote ( continious dimming ) will not as scene transition.

KipK, 2020-12-13:
Here are some debug logs when switching from 2 scenes.

`17:23:23:406 CTRL got nwk update id 0
17:23:55:444 enqueue event config/localtime for /config/
17:23:56:159 poll node 7c:b0:3e:aa:00:ab:9a:c8-03
17:23:56:169 Poll light node Cuisine 4
17:23:56:230 Daylight now: sunsetEnd, status: 200, daylight: 0, dark: 0
17:23:56:233 enqueue event config/localtime for /config/
17:23:56:254 Poll APS request to 0x7CB03EAA00AB9AC8 cluster: 0x0006 dropped, values are fresh enough
17:23:56:817 Poll APS request to 0x7CB03EAA00AB9AC8 cluster: 0x0008 dropped, values are fresh enough
17:23:56:832 Node data 0x60a423fffea3abbe profileId: 0x0104, clusterId: 0x0008
17:23:56:838 0x60A423FFFEA3ABBE: update ZCL value 0x01/0x0008/0x0000 after 0 s
17:23:56:882 ZCL attribute report 0x60A423FFFEA3ABBE for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
17:23:56:888 	payload: 00002045
17:23:57:103 Node data 0x7cb03eaa00ab9ac8 profileId: 0x0104, clusterId: 0x0008
17:23:57:110 0x7CB03EAA00AB9AC8: update ZCL value 0x03/0x0008/0x0000 after 0 s
17:23:57:116 ZCL attribute report 0x7CB03EAA00AB9AC8 for cluster: 0x0008, ep: 0x03, frame control: 0x18, mfcode: 0x0000 
17:23:57:122 	payload: 00002049
17:23:57:347 Node data 0x60a423fffea3abbe profileId: 0x0104, clusterId: 0x0008
17:23:57:357 0x60A423FFFEA3ABBE: update ZCL value 0x01/0x0008/0x0000 after 0 s
17:23:57:416 ZCL attribute report 0x60A423FFFEA3ABBE for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
17:23:57:425 	payload: 00002045
17:23:57:663 ZCL attribute report 0x60A423FFFEA3ABBE for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
17:23:57:926 ZCL attribute report 0x60A423FFFEA3ABBE for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
17:23:57:990 ZCL attribute report 0x60A423FFFEA3ABBE for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
17:24:00:119 Node data 0x7cb03eaa00ab9ac8 profileId: 0x0104, clusterId: 0x0300
17:24:00:132 0x7CB03EAA00AB9AC8: update ZCL value 0x03/0x0300/0x0007 after 0 s
17:24:00:136 0x7CB03EAA00AB9AC8: update ZCL value 0x03/0x0300/0x0008 after 0 s
17:24:00:140 0x7CB03EAA00AB9AC8: update ZCL value 0x03/0x0300/0x4001 after 0 s
17:24:00:237 enqueue event config/localtime for /config/
17:24:00:527 skip verify rule Rule PUSHDIMUP_START for sensor 00:17:88:01:06:37:1c:a7-02-fc00 (available = 0, node = (nil), sensorNode = (nil))
17:24:00:716 skip verify rule Rule PUSHDIMUP_STOP for sensor 00:17:88:01:06:37:1c:a7-02-fc00 (available = 0, node = (nil), sensorNode = (nil))
17:24:00:936 skip verify rule Rule PUSHDIMUP_ON for sensor 00:17:88:01:06:37:1c:a7-02-fc00 (available = 0, node = (nil), sensorNode = (nil))
17:24:01:318 poll node 60:a4:23:ff:fe:a3:ab:be-01
17:24:01:321 Poll light node bar
17:24:01:324 enqueue event config/localtime for /config/
17:24:01:483 Poll APS request to 0x60A423FFFEA3ABBE cluster: 0x0006 dropped, values are fresh enough
17:24:01:880 Poll APS request to 0x60A423FFFEA3ABBE cluster: 0x0008 dropped, values are fresh enough
17:24:02:087 add task 429977 type 25 to group 0x0005 cluster 0x0005 req.id 252
17:24:02:090 enqueue event action/scene for /groups/5
17:24:02:093 enqueue event state/bri for /lights/2
17:24:02:145 Websocket 192.168.1.18:34386 send message: {"e":"changed","id":"2","r":"lights","state":{"alert":null,"bri":90,"on":true,"reachable":true},"t":"event","uniqueid":"60:a4:23:ff:fe:a3:ab:be-01"} (ret = 148)
17:24:02:152 Websocket 192.168.1.8:49405 send message: {"e":"changed","id":"2","r":"lights","state":{"alert":null,"bri":90,"on":true,"reachable":true},"t":"event","uniqueid":"60:a4:23:ff:fe:a3:ab:be-01"} (ret = 148)
17:24:02:177 Websocket 192.168.1.18:34386 send message: {"e":"scene-called","gid":"5","r":"scenes","scid":"3","t":"event"} (ret = 0)
17:24:02:195 Websocket 192.168.1.8:49405 send message: {"e":"scene-called","gid":"5","r":"scenes","scid":"3","t":"event"} (ret = 0)
17:24:02:251 Erase task req-id: 252, type: 25 zcl seqno: 90 send time 0, profileId: 0x0104, clusterId: 0x0005
17:24:02:561 enqueue event config/localtime for /config/
17:24:02:895 Node data 0x60a423fffea3abbe profileId: 0x0104, clusterId: 0x0006
17:24:02:898 0x60A423FFFEA3ABBE: update ZCL value 0x01/0x0006/0x0000 after 0 s
17:24:02:908 enqueue event event/checkgroupanyon for /groups/65520
17:24:02:919 enqueue event event/checkgroupanyon for /groups/5
17:24:03:019 ZCL attribute report 0x60A423FFFEA3ABBE for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
17:24:03:027 	payload: 00001001
17:24:03:224 enqueue event config/localtime for /config/
17:24:03:233 Node data 0x60a423fffea3abbe profileId: 0x0104, clusterId: 0x0006
17:24:03:247 0x60A423FFFEA3ABBE: update ZCL value 0x01/0x0006/0x0000 after 0 s
17:24:03:250 enqueue event event/checkgroupanyon for /groups/65520
17:24:03:253 enqueue event event/checkgroupanyon for /groups/5
17:24:03:296 ZCL attribute report 0x60A423FFFEA3ABBE for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
17:24:03:299 	payload: 00001001
17:24:03:687 add task 429981 type 14 to 0x60A423FFFEA3ABBE cluster 0x0006 req.id 5
17:24:03:695 add task 429982 type 11 to 0x60A423FFFEA3ABBE cluster 0x0008 req.id 6
17:24:03:731 delay sending request 6 dt 0 ms to 0x60A423FFFEA3ABBE, ep: 0x01 cluster: 0x0008 onAir: 1
17:24:03:735 delay sending request 6 ep: 0x01 cluster 0x0008 to 0x60a423fffea3abbe onAir 1
17:24:03:747 poll node bc:33:ac:ff:fe:82:a4:ec-01
17:24:03:756 Poll light node Entrée 2
[... cut ...]
17:24:09:368 Node data 0x60a423fffea3abbe profileId: 0x0104, clusterId: 0x0008
17:24:09:375 0x60A423FFFEA3ABBE: update ZCL value 0x01/0x0008/0x0000 after 0 s
17:24:09:399 	payload: 0000205a
17:24:09:488 poll node 84:2e:14:ff:fe:64:e9:39-01
17:24:09:492 Poll light node On/Off plug-in unit 12
17:24:09:496 skip verify rule Rule PUSHDIMDOWN_OFF for sensor 00:17:88:01:06:37:1c:a7-02-fc00 (available = 0, node = (nil), sensorNode = (nil))
17:24:09:553 Poll APS request to 0x842E14FFFE64E939 cluster: 0x0006 dropped, values are fresh enough`

Smanar, 2020-12-13:
But It can be from the device itself ?
If it don't work with deconz, I don't think we can do something, the device is using standards zigbee commands, so I don't think it can be from exotic ones.

closed 2021-03-05 by github-actions[bot]

pafftis, 2022-02-07:
Help me please.  How to calibrate the light level?  I have a 10% llama does not burn.  And also at a value of 50-100 there is no change in the level of light.  Works in the range only from 10-50.  How to fix?