dresden-elektronik/deconz-rest-plugin #3901: deCONZ shows update available but latest update is already installed

Thomas-Vos, 2020-12-13:
## Describe the bug
After updating to the latest stable version (2.07.01 / 26680700) deCONZ stops working every morning. It shows that there is a firmware update available. See screenshot 1. (Interesting is that deCONZ help does not show the firmware version, see screenshot 2). After clicking the "Update Firmware" button it starts running the update. After that it shows "Connect to Device" and is connecting for about minute or so. Then it shows that there is a device firmware update again.

Next thing I did was close deCONZ and restarted deCONZ with debug logging enabled. This time it could not connect to the RaspBee at all. After a few minutes I closed deCONZ. See debug.txt at the bottom of this issue.

I restarted the Raspberry Pi hoping that it would solve the issue. Had some trouble starting the Raspberry Pi but after reconnecting it to power a few times it started again, see screenshot 3. After it started the RaspBee could still not connect.

This has been happening the last few days, every morning it shows this update message. The way I fix it (works every time), is flashing the same firmware again like this:
```
wget http://deconz.dresden-elektronik.de/deconz-firmware/deCONZ_RaspBeeII_0x26680700.bin.GCF
sudo GCFFlasher_internal -t 60 -f deCONZ_RaspBeeII_0x26680700.bin.GCF
```
After that I can start deCONZ but the next morning I get the same update message.

## Expected behavior
Latest firmware update is already installed so deCONZ should not ask to update the firmware.

## Screenshots
[image]
[image]
[image]

## Environment
 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: 26680700
 - deCONZ version: 2.07.01
 - Device: RaspBee II (Phoscon Gateway)
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs
[debug.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5684743/debug.txt)

## Additional context
Ran the following command when deCONZ could not connect to the RaspBee II (deCONZ was not running)
```
pi@phoscon:~ $ GCFFlasher_internal -l
GCFFlasher V3_17 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyS0       | 0x0000 | 0x0000  |            | RaspBee 
pi@phoscon:~ $ 
```

Reflashing the firmware to get deCONZ working again:
```
pi@phoscon:~ $ sudo GCFFlasher_internal -t 60 -f deCONZ_RaspBeeII_0x26680700.bin.GCF
GCFFlasher V3_17 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device RaspBee (RaspBee)
action: reset device RaspBee
wiringPi 2.52 initialized
Bootloader version 0x00030003, app crc: 0x00000000
flashing 164373 bytes: |================|deCONZ firmware version 26680700
pi@phoscon:~ $ sudo GCFFlasher_internal -l
GCFFlasher V3_17 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyS0       | 0x0000 | 0x0000  |            | RaspBee 
pi@phoscon:~ $ 

```

manup, 2020-12-19:
Confirmed, the bug is a regression in the RaspBee II firmware 0x26680700 (doesn't happen for ConBee II).

This is now fixed in firmware version 0x26690700.
Can you please update to this version as described in the Wiki.

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

Thomas-Vos, 2020-12-19:
Thank you very much. I just installed the update. I will keep you updated what happens over the next few days.

Thomas-Vos, 2020-12-22:
Fixed, thanks again @manup.

closed 2020-12-22 by Thomas-Vos