dresden-elektronik/deconz-rest-plugin #3904: Issues pairing IKEA LED1836G9 806lm bulbs (2)

cputoaster, 2020-12-13:
## Describe the bug
I bought one of these bulbs and now cannot pair it. See here: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2426, I cannot reopen the issue. Mostly used Phoscon, but in my desperation also tried the old WebUI and the Permit Join the the deconz UI. As the bug got closed, maybe someone has a relatively clear way of actually joining a bulb?

## Steps to reproduce the behavior
Exactly like the issue 2426. I also tried doing the bulb reset very close (<5cm) to the ConBee II USB stick, just to be sure.

## Expected behavior
Join like all the other bulbs

## Screenshots
N/A

## Environment
 - Host system: PC
 - Running method: Marthoc Docker container on Ubuntu host
 - Firmware version: 26680700
 - deCONZ version: 2.07.01
 - Device: ConBee II
 - Do you use an USB extension cable: yes, 30cm
 - Is there any other USB or serial devices connected to the host system? Yes If so: Which? Z-Wave controller

## deCONZ Logs
Don't see anything of a new node

## Additional context
107 Nodes, mostly Ikea bulbs and Xiaomi sensors, with a few Gledopto and Phillips outdoor floodlights.

xadox-1st, 2020-12-26:
Same for me with LED1835C6 (E14 Version).

Version 2.08.00 / 8.12.2020
Firmware 26690700

cputoaster, 2021-01-18:
I also have the problem now with LED1837R5 (GU10), still on 2.07.01 / 26680700

cputoaster, 2021-01-22:
Now managed to add 3 LED1837R5 in a row (110 nodes now, dense network), after having rebooted the server 3x in a row and then waited for 2h. May be just pure luck.
As I cannot switch of all routers (it would be very annoying to open all lamp cases etc), is there any other way that one could try adding things to dense networks? I think once they are added it seems to work ok.

cputoaster, 2021-01-29:
After trying a lot of times again, I now also managed to pair the LED1836G9. Closing this, but still don't know why it did not work for weeks with the same software.

closed 2021-01-29 by cputoaster