dresden-elektronik/deconz-rest-plugin #3905: Xiaomi Door Sensor need to resynch after every deCONZ restart

nerdmovieproduction, 2020-12-14:
## Describe the bug

Hi Everyone, I have some issue with the Xiaomi Door Sensor. Everytime deCONZ need to reboot (backup upload or complete Raspberry reboot) all my Xiaomi Door Sensor stop working, disapper from VNC and i need to resynch all of it!

## Screenshots
[image]
[image]

## Environment

 - Host system: Raspberry Pi 4
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: (26660700)
 - deCONZ version: (2.07.01)
 - Device: ConBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system?no

## deCONZ Logs

SwoopX, 2020-12-14:
Are those Mija's, the roundish ones?
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3528#issuecomment-743367042

nerdmovieproduction, 2020-12-14:
> Mija's, the roundish ones?

Yes! So i need to Retrieve missing descriptors in deCONZ GUI For all of it, right?

The only values that changes after this are: Max buffer Size, Max incoming transfert Size and Max Outgoing Transfert Size

SwoopX, 2020-12-14:
Yes, that sound as if the descriptors were updated. You might want to conduct some tests before you do it on all devices.

closed 2021-03-05 by github-actions[bot]