dresden-elektronik/deconz-rest-plugin #3906: Lights not operable after firmware update

mrfenyx, 2020-12-14:
## Describe the bug
I am currently using version 2.07.01 / firmware 26660700 and am experiencing an issue where, although the Phoscon App shows that lights are turned on and off in the GUI, they are not. This occurs identically if I am using switches or the Phoscon GUI for turning lights on and off.

## Steps to reproduce the behavior

1. Open Phoscon App and observe the lights / groups.
2. Turn on one of the lights / groups using the GUI
3. Turn off the light / group using the GUI
4. Repeat on/off using a remote / switch

## Observed behavior
The icons in front of the groups turn on/off as expected but the lights remain off.

## Expected behavior
The icons and the actual lights turn on/off when using the Phoscon App and the switches / remotes.

## Environment
 - Host system: Raspberry Pi 4B
 - Running method: Raspberry Pi OS
 - Firmware version: 26660700
 - deCONZ version: 2.07.01
 - Device: ConBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? Yes, USB-RF Stick (868-MHz-Frequency)

## Screenshots
[image]

## deCONZ Logs
[deconz.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5687807/deconz.log)

## Additional context
- 33 nodes
- 15 routers
- 18 end devices
- device brands: Ikea Tradfri, Osram Smart+, Xiaomi, Philips Hue

Mimiix, 2020-12-14:
Hi!
Hi, thanks for the extensive report!

I think there's a issue here: 
09:27:30:013 0x00158D0004B275BE error APSDE-DATA.confirm: 0xE1 on task

You are suffering from interference. You should use a extension cable. Can you try that?

mrfenyx, 2020-12-14:
Hi, thanks for the suggestion. Unfortunately I don't have an extension cable available. What I did was unplug the ConBee II from the RaspberryPi and plug it back in. Now the system works again:

[image]

What I don't understand was how come before, the switches would clearly communicate with the gateway (light icons would turn on and off) but the connected lights were unresponsive.
I will observe the issue and order an extension cable too. Hopefully it is just interference.

mrfenyx, 2020-12-14:
UPDATE: while trying out different switches and the Phoscon App I realized that sometimes the system doesn't respond to the first click or press of a button. I need to press the on or off button 2-4 times until the lights react. Might this also be caused by interference and solved by the extension cable?

Mimiix, 2020-12-14:
@mrfenyx 
That probably is the interference. Yes. To make sure of that: It woould help if you do another logfile like the one you did before.

mrfenyx, 2020-12-14:
here's the new log:
[deconz.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5688149/deconz.log)

Mimiix, 2020-12-14:
Yep! Same issue: 
09:27:14:601 0x00158D0004B277A7 error APSDE-DATA.confirm: 0xE1 on task
09:27:14:602 APS-DATA.confirm id: 126, status: 0xE1

mrfenyx, 2020-12-14:
UPDATE 2: I read a bit about the interference and especially about the one caused by USB3 devices and WiFi. The WiFi channels were ok. The issue seems to have been my other USB Stick which was actually plugged in an USB3 port. I moved it to an USB2 port and what do you know, no more interference at all. See also attached log:
[deconz_3.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5688341/deconz_3.log)

Mimiix, 2020-12-14:
Nice!

But you also have these:
10:59:45:079 0x00158D000324D242 error APSDE-DATA.confirm: 0xE9 on task
11:00:27:676 0x00158D000324D24B error APSDE-DATA.confirm: 0xE9 on task

So i'd say there still is something not right. Probably due to either distance or interference further down the line.

mrfenyx, 2020-12-14:
Hi, for some reason those routers try to connect to some quite far away other routers: from the second floor, "ignoring" the first floor where the ConBee II is, to the ground floor. That's why the connection quality is quite bad. Is there any way to "force" some routers to not connect? Also, a bit of a different question: is there any tool that can map the connections a bit nicer than the built-in visualization in deconz?

Mimiix, 2020-12-14:
On the first : There's this source routing. I am not sure how that works, this should explain a bit:  https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1261#issuecomment-687902156

However, those that do that are those xiaomi devices? They tend not to switch parent. Way around this is to pair them on their "location" . 

On the second question: Nope.

mrfenyx, 2020-12-14:
Thanks @Mimiix for the quick answers. On the first issue, they are not Xiaomi devices, they are innr lights and Osram plugs. I will look into source routing in a while and report on any updates.

damanti-me, 2020-12-15:
Hey im using the same firmeware than @mrfenyx , and also run into those issues. I just updated HA core and OS to the newest version and did not change the hardwaresetup. Using Conbee2 directly in a RBP3+. 0xE1 means according to the manual  MAC channel access failure due to interferences. 
The other thing im asking myself is why (due to the log) he is not able to cange the channel automaticly. Tring to change the channel manually to e.g. 20 with the Phoncon UI works for a second and then switches back to 15. 
Thank you for your time
Log:
13:33:03:117 Skip automatic channel change, TODO warn user
13:33:03:899 0x000D6FFFFEE400E1 error APSDE-DATA.confirm: 0xE1 on task
13:33:05:696 0x086BD7FFFE210AEB error APSDE-DATA.confirm: 0xE1 on task
13:33:07:500 0x000D6FFFFEE400E1 error APSDE-DATA.confirm: 0xE1 on task
13:33:09:296 0x086BD7FFFE210AEB error APSDE-DATA.confirm: 0xE1 on task
13:33:11:106 0x000D6FFFFEE400E1 error APSDE-DATA.confirm: 0xE1 on task
13:33:12:890 0x086BD7FFFE210AEB error APSDE-DATA.confirm: 0xE1 on task
13:33:13:117 channel 15 does not match channel mask 0x00100000 (TODO)
13:33:13:118 channel is 15 but should be 20, start channel change
13:33:13:118 Skip automatic channel change, TODO warn user
13:33:14:683 0x000D6FFFFEE400E1 error APSDE-DATA.confirm: 0xE1 on task
13:33:16:487 0x086BD7FFFE210AEB error APSDE-DATA.confirm: 0xE1 on task
13:33:16:603 Websocket disconnected 127.0.0.1:48076, state: 0, close-code: 1000, reason: 
13:33:18:275 0x000D6FFFFEE400E1 error APSDE-DATA.confirm: 0xE1 on task
13:33:18:831 New websocket 127.0.0.1:49372 (state: 3) 
13:33:18:987 Websocket disconnected 127.0.0.1:49372, state: 0, close-code: 1000, reason: 
13:33:20:091 0x086BD7FFFE210AEB error APSDE-DATA.confirm: 0xE1 on task
13:33:20:844 New websocket 127.0.0.1:49704 (state: 3) 
13:33:20:846 Remove websocket 127.0.0.1:49704 after error Unknown error, close-code: 1000, reason: 
13:33:22:459 0x000D6FFFFEE400E1 error APSDE-DATA.confirm: 0xE1 on task
13:33:23:115 channel 15 does not match channel mask 0x00100000 (TODO)

Mimiix, 2020-12-15:
@damanti-me I'd like you to open a own user question issue. Hijacking is not something i appreciate.

You probably are having USB interference. Switching channels wont matter.

closed 2021-03-05 by github-actions[bot]