dresden-elektronik/deconz-rest-plugin #3907: daylight sensor not reachable

talun2075, 2020-12-14:
[image]
[image]
Hi,
the Daylight Sensor is not Reachable.
I Use a Conbee II on a Windows System with deconz.
I add the Timezone over Phoscon App.
Gateway Informations:
2.07.01 / 8.12.2020 with Firmware 26580700

I Have found a Issue and try it do do it on this way.
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/521
But it not change the Problem.

I made a put on
/api/APIKEY/sensors/1/config

{
  "long": "4.9098",
  "lat": "52.3423",
  "sunriseoffset": 0,
  "sunsetoffset": 0
}

I got a 200 Back so the Put was successfull with this answer:
[{"success":{"/sensors/1/config/lat":"52.3423","/sensors/1/config/long":"4.9098","/sensors/1/config/sunriseoffset":0,"/sensors/1/config/sunsetoffset":0}}]

After call the Sensor Informations
/api/APIKEY/sensors/1

I got this:

{"config":{"configured":false,"on":true,"sunriseoffset":0,"sunsetoffset":0},"etag":"bd95bc37836298ba594e5c450764863c","manufacturername":"Philips","modelid":"PHDL00","name":"Daylight","state":{"dark":null,"daylight":null,"lastupdated":"none","status":null},"swversion":"1.0","type":"Daylight","uniqueid":"00:21:2e:ff:ff:06:62:03-01"}
And the Sensor is still not reachable.

Any Idea?

closed 2020-12-14 by close-issue-app[bot]

talun2075, 2020-12-14:
hm..don´t understand why is this closed? Where is the template that the bot mean?

Mimiix, 2020-12-14:
@talun2075 
[image]

You had to select one then follow the template.