dresden-elektronik/deconz-rest-plugin #3908: daylight Sesor not reachable

talun2075, 2020-12-14:
## Describe the bug
Hi,
the Daylight Sensor is not Reachable.
It was never reachable on my system. I´m new with conbee and deconz.
I Use a Conbee II on a Windows System with deconz.
I add the Timezone over Phoscon App.

I Have found a Issue and try it do do it on this way.
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/521
But it not change the Problem.

I made a put on
/api/APIKEY/sensors/1/config

{
  "long": "4.9098",
  "lat": "52.3423",
  "sunriseoffset": 0,
  "sunsetoffset": 0
}

I got a 200 Back so the Put was successfull with this answer:
[{"success":{"/sensors/1/config/lat":"52.3423","/sensors/1/config/long":"4.9098","/sensors/1/config/sunriseoffset":0,"/sensors/1/config/sunsetoffset":0}}]

After call the Sensor Informations
/api/APIKEY/sensors/1

I got this:

{"config":{"configured":false,"on":true,"sunriseoffset":0,"sunsetoffset":0},"etag":"bd95bc37836298ba594e5c450764863c","manufacturername":"Philips","modelid":"PHDL00","name":"Daylight","state":{"dark":null,"daylight":null,"lastupdated":"none","status":null},"swversion":"1.0","type":"Daylight","uniqueid":"00:21:2e:ff:ff:06:62:03-01"}
And the Sensor is still not reachable.

Any Idea?

## Steps to reproduce the behavior
No Idea

## Expected behavior
Use the Daylight Sensor ;-)

## Screenshots
[image]
[image]

## Environment

 - Host system: Windows 2019
 - Running method: Windows
Gateway Informations:
 - Firmware version: (26580700)
 - deCONZ version: (2.07.01)
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? Yes Mouse & KeyBoard

## deCONZ Logs

11:47:44:941 HTTP API PUT /api/APIKEY/sensors/1/config - 192.168.0.2
11:47:44:941 Text Data: 	{
  "long": "4.9098",
  "lat": "52.3423",
  "sunriseoffset": 0,
  "sunsetoffset": 0
}
11:47:44:941 ApiMode: 0
11:47:44:957 [{"success":{"/sensors/1/config/lat":"52.3423","/sensors/1/config/long":"4.9098","/sensors/1/config/sunriseoffset":0,"/sensors/1/config/sunsetoffset":0}}]
11:47:45:758 HTTP client hdr detected:
GET /api/APIKEY/sensors/new?_=1607941761448 HTTP/1.1
Host: 192.168.0.6:888
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:83.0) Gecko/20100101 Firefox/83.0
Accept: vnd.ddel.v1
Accept-Language: de
Accept-Encoding: gzip, deflate
Access-Control-Allow-Origin: *
X-Requested-With: XMLHttpRequest
DNT: 1
Connection: keep-alive
Referer: http://192.168.0.6:888/pwa/devices-sensors.html?_v=XXX,11,1,11&gwid=00212EFFFF066203
Sec-GPC: 1

## Additional context

Mimiix, 2020-12-14:
That's a strange one.
Does it has access to the internet?

talun2075, 2020-12-14:
The Server have access to the internet over a Pihole DNS but there is nothing logged what looks to an error.
I tryed without Pihole too.

Mimiix, 2020-12-14:
did you ever reboot the system?

ebaauw, 2020-12-14:
Last I checked the builtin _Daylight_ sensor doesn’t expose `reachable`.  Not sure how Phoscon concludes is unteachable.

The sensor won’t update unless it’s been configured.  API call to do so looks good to me.  Don’t know what’s happening.  I would check the record in the database.  Be sure to stop deCONZ and to backup the database before messing with it.

talun2075, 2020-12-14:
@ebaauw : Hi Erik, you seen I have use your Home to configure my Daylight ;-) 

I Updated the conbee II to latest firmware 26680700

In the zzl.db I found under Sensord the Daylight Sensor with this values:
{configured":false,"lat":"52.3423","long":"4.9098","on":true,"sunriseoffset":0,"sunsetoffset":0}"

state like above
fingerprint = empty
deletestate = normal
mode = 1
swversion= 1.0

Any Hint?

ebaauw, 2020-12-14:
> In the zzl.db I found under Sensord the Daylight Sensor with this values:
{configured":false,"lat":"52.3423","long":"4.9098","on":true,"sunriseoffset":0,"sunsetoffset":0}"

That's not valid JSON; I hope due to a copy/paste error?  `"configured"` should be set to `true`, once the coordinates have been set.

> state like above

That's not good.  It should be something like (but with a lot of `null` values):
```
{"dark":true,"daylight":false,"lastupdated":"2020-12-14T17:35:15.661","status":230,"sunrise":"2020-12-14T07:44:40","sunset":"2020-12-14T15:28:25"}
```

Is this for `/sensors/1` (`sid` of `1`)?

Looks like the database record has been corrupted.  I've never seen this before.  I don't know how likely it would be for just a single record to be corrupted.  I fear this might be an indication of bigger problems.  If you're running on a Raspberry Pi, I'd backup the entire uSD card and run `fsck`, before trying anything else.

If the corruption is limited to this one record, you could try and delete the record from the database.  The REST API plugin should re-create the sensor automatically on startup.

> Hi Erik, you seen I have use your Home to configure my Daylight

I'm flattered, of course, but don't you think it's a bit too early in our relationship to be moving in with me?

talun2075, 2020-12-14:
I run it under a Windows System and there are no Problems in.
I will try to delete the record and then we will seen.

Maybe you can post me your Daylight Sensor, then I can add this to the database and maybe this will help if the first will not work well.

And of course if it once work I move it away from you. I will not switch of your Lights ;-)

ebaauw, 2020-12-14:
```
$ ph get /sensors/1
{
  "config": {
    "configured": true,
    "on": true,
    "sunriseoffset": 0,
    "sunsetoffset": 0
  },
  "etag": "0b2faf904d6529dbd13e7d861f2fbac5",
  "manufacturername": "Philips",
  "modelid": "PHDL00",
  "name": "Daylight",
  "state": {
    "dark": true,
    "daylight": false,
    "lastupdated": "2020-12-14T18:48:18.417",
    "status": 230,
    "sunrise": "2020-12-14T07:44:40",
    "sunset": "2020-12-14T15:28:25"
  },
  "swversion": "1.0",
  "type": "Daylight",
  "uniqueid": "00:21:2e:ff:ff:01:02:56-01"
}
```

[image]
[image]
[image]

talun2075, 2020-12-14:
thx will try this tomorrow.

talun2075, 2020-12-15:
strange, after delete it create a new daylight sensor will be created but with really strange values.

{"dark":null,"daylight":null,"lastupdated":null,"status":null}"
"{"configured":false,"on":true,"sunriseoffset":30,"sunsetoffset":-30}"

You see lat/long is missing.

Now I have take your config and it works.
I see him and can use it.

pillemats, 2020-12-26:
Same Problem Here with win10

how To fix it? I am new with deconz an phoscon :)

thanks :-) 

btw: on ubuntu ist will work Directly but there it wont find the Stick on Startup

pillemats, 2020-12-27:
is there a way to fix ? 

I spend a lot of days in it and a boring wife :-)

Please help me

lichtl, 2020-12-28:
Having the same issue on my headless raspberry installation. In my case the sensor was created with a different ID however lat/lon fields are missing.

`   "15": {
        "config": {
            "configured": false,
            "on": true,
            "sunriseoffset": 30,
            "sunsetoffset": -30
        },
        "etag": null,
        "manufacturername": "Philips",
        "modelid": "PHDL00",
        "name": "Daylight",
        "state": {
            "dark": null,
            "daylight": null,
            "lastupdated": "none",
            "status": null
        },
        "swversion": "1.0",
        "type": "Daylight",
        "uniqueid": "00:21:2e:ff:ff:06:b0:41-01"
    },`

pillemats, 2020-12-28:
Oh interesting... ich have the problem at my windows 10 config. It is terrible without the sensor

pillemats, 2020-12-29:
@talun2075 hat du das problem in den griff bekommen und kannst mir dabei vielleicht helfen ?

SwoopX, 2020-12-30:
I can confirm this is a bug at least in the Windows version. Deconz cannot establish any SSL connection and therefore cannot successfully finish its discovery procedure. Based on this and the fact, that the discovery URL in the database and the hardcoded fallback are identical HTTPS URLs, the process fails. Part of that process is also the setup of the Daylight sensor, as latitude and longitude are chosen on the city entered for the timezone.

This is the only trace the debug output generates (DBG_INFO):
`02:05:50:212 discovery network reply error: Error creating SSL context ()`

The error originates from `discovery.cpp`, `internetDiscoveryFinishedRequest()` also holding the fallback URL.

manup, 2020-12-30:
Confirmed, the SSL error has indicated some OpenSSL DLLs where missing.
I've included them now in the v2.9.0-beta build.

In a few tests on Win10 with 2.8.0 I've seen the same error, with v2.9.0 discovery server connection works and the daylight sensor was configured properly.

https://deconz.dresden-elektronik.de/win/deCONZ_Setup_Win32_V2_09_00.exe

SwoopX, 2020-12-30:
As I already got a confirmation this is working with the new version, closing this one.

closed 2020-12-30 by SwoopX

talun2075, 2021-01-20:
> 
> 
> @talun2075 hat du das problem in den griff bekommen und kannst mir dabei vielleicht helfen ?

Hi, Sorry hatte da keine Beenachrichtigung erhalten und das jetzt nur durch zufall gesehen. Ich habe den einmal komplett über eine sql lite konfiguration überarbeitet.
Ich habe die Daten von @ebaauw genommen und das ging bei mir und danach meine Dateen angepasst.
Da ssolltest Du aber ein wenig Plan von SQL haben um das zu machen.