dresden-elektronik/deconz-rest-plugin #3911: Loosing connections to some devices (Eurotronic Spirit Zigbee)

ChristophHornung, 2020-12-14:
## Describe the bug

I am running deConz via docker in HA. I started with an Eurotronic Spirit Zigbee and some Aqara sensors and everything worked perfectly for quite some time. 
Recently I added a lot more of the Spirit Zigbee devices, the issue now is that I can pair the devices and everything works, but after some time (With maybe a restart involved) the communication to the devices no longer works. Sometimes to nearly all of them sometimes to just a few.

Symptoms:
* There is no connection line drawn between the ConBeeII and the device.
* Any action on the problematic devices results in the red bubble blinking
* During actions the log shows a lot of :
    ```
    ZCL cmd-req nwk: 0xFE5E, profile: 0x0104, cluster: 0x0000 cmd: 0x00
    reject aps request to enddevice node queue is full (4)
    ZCL read cluster: 0x0000, attribute: 0xFF22
    ```
as well as
    ```
    max transmit errors for node 0x00158D00053D2DCB, last seen by neighbors 1414 s
    ```
(I attached part of the log)
* I also see a lot of 
`    12:10:06:988 Node 0x00158D0001FF8D45 is known by 0 neighbors, last seen -1 s`

Things I tried:
* Leaving and joining network with "Save" on the network settings in between
* Starting the deCONZ add-in a bit after HA reboots

I can re-pair the device in question and they will work normally afterwards, but at least after rebooting some devices will start showing the above symptoms.

## Steps to reproduce the behavior

Seems to appear after reboots, but does not always affect the same devices. The old first Spirit Zigbee in the network seems to be immune against the problem.

## Expected behavior
Have a stable connection to all devices.

## Screenshots
[image]

## Environment
 - Host system: Raspberry Pi 3
 - Running method: Home Assistent deCONZ Add-on 6.6.0
 - Firmware version: 26660700
 - deCONZ version: (2.07.01)
 - Device: ConBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? no
 - HA Supervisor 2020.12.6 (but appeared also on previous versions)
 - Host System Home Assistant OS 5.8 (but also on previous versions)

## deCONZ Logs
[deconzLogDuringReadCluster.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5688420/deconzLogDuringReadCluster.txt)
[startup.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5688445/startup.txt)

## Additional context
9 Spirit Zigbee 
3 Aquara Sensors (Temperature + Door Open/Close + Water Sensor)

Mimiix, 2020-12-14:
I don't see a lot on the logs. 

The thing that do come to mind, is that you do not have any routers in your zigbee network. Adding those could help. 

Can you provide a log with at least 1 minute of running time? Just to see what happens.

ChristophHornung, 2020-12-14:
I copyed that from the log page on the HA addin: (I think there is a small gap)

[deconzlog2.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5688514/deconzlog2.txt)

Is there a better way to get to larger parts of the log file form the HA Addin?

Mimiix, 2020-12-14:
The debugview you had running in your screenshot :) Leave that one open for a while, then copy :)

ChristophHornung, 2020-12-14:
@Mimiix I tried that but copy does not copy to my clipboard from the remote access via the HA webpage. Is there another way to connect to it so the clipboard works?

Mimiix, 2020-12-14:
Probably VNC. But i am not sure if that still works. Should check the HA Addon documentation for that.

Mimiix, 2020-12-14:
Another thing to try first: I talked to Manup, he urgently suggests to use the latest firmware: 
http://deconz.dresden-elektronik.de/deconz-firmware

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

you can do that on a windows install ^^

ChristophHornung, 2020-12-14:
Ok, I updated to 26680700 (this is the latest right?). I will check if the problem persists and try to get a bigger chunk of the log via VNC.

ChristophHornung, 2020-12-14:
The problem still seems to persist. I opened the VNC port so a bigger chunk of the log:

[newDeconzLogFromVNC.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5688645/newDeconzLogFromVNC.txt)

Mimiix, 2020-12-14:
The only thing i can see is the 0xD0 (13:04:55:677 0x00158D00032F7E16 error APSDE-DATA.confirm: 0xD0 on task)

How far away are the devices?

ChristophHornung, 2020-12-14:
The ...E16 one is about 3 meters away, the closest one not working about 2 meters. I already tried unscrewing them from the radiator and putting them next to the ConBee, but that does not change the behavior.

When re-pairing them and putting them back on the radiator they also work, so I don't think distance is an issue.

ChristophHornung, 2020-12-14:
@Mimiix What about the "is known by 0 neighbors" entry? Isn't that strange? The ...141 is the only working one at the moment and it ready as "is known by 1 neighbor". So that seems to be a difference between the working and non working ones.

ChristophHornung, 2020-12-14:
An even larger chunk of log:

[moreLogsDeconzVNC.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5688726/moreLogsDeconzVNC.txt)

Mimiix, 2020-12-14:
I think that has to do with the fact that you don't have any routers. Just end nodes. 

 I have asked Swoop and Manup to read along. Hope they join in shortly.

ChristophHornung, 2020-12-14:
@Mimiix 
I rejoined one of the devices (the ...DCB) one. As expected it now works and I can see in the log it now also reads:

`Node 0x00158D00053D2DCB is known by 1 neighbors, last seen 7 s`

So it really seems to be that 
non working device = is known by 0 neighbors,
working device = is known by 1 neighors.

ebaauw, 2020-12-14:
Could you check whether the unreachable devices still send updates to deCONZ (does `lastupdated` change, is the manually set temperature reflected by the API)?  You can probably revive the TRV by power-cycling it, remove the batteries for a couple of seconds.

ChristophHornung, 2020-12-14:
@ebaauw Manually changed temperatures are not reflected in HA at least. Where can I see the `lastupdated` value?

Mimiix, 2020-12-14:
In HA, you probably need to listen to the deconz_event in the event listener.

Else you need to enable the debug in the HA integration.

ChristophHornung, 2020-12-14:
@ebaauw Power-Cycling does not revive them. It shows "JIN" on their display and they seem to join the network (they have an antenna icon that is display), but they show the same behavior as before.
I power-cycled the ...0A8 device and I still see the log showing `Node 0x00158D00032F70A8 is known by 0 neighbors, last seen -1 s`

ChristophHornung, 2020-12-14:
For one of the working devices I think I can see updates in the log via

```
 DB sql exec REPLACE INTO sensors (sid, name, type, modelid, manufacturername, uniqueid, swversion, state, config, fingerprint, deletedState, mode) VALUES ('11', 'Wohnzimmer Heizung Fernseher', 'ZHAThermostat', 'SPZB0001', 'Eurotronic', '00:15:8d:00:05:3d:2d:cb-01-0201', '22190930', '{"lastupdated":"2020-12-14T13:01:35.247","on":true,"temperature":2050,"valve":103}', '{"battery":100,"displayflipped":null,"heatsetpoint":2250,"hostflags":1,"locked":null,"mode":"auto","offset":0,"on":true,"reachable":true}', '{"d":769,"ep":1,"in":[1,513],"p":260}', 'normal', '1')
```

for the non working device I do not see such log entries

ChristophHornung, 2020-12-14:
@Mimiix Just, fyi regarding the connection quality. I re-paired all thermostats and the connections all look healthy except for one  which is orange (about 7 meters thorough 2 walls). So the ConBee II stick actually has quite some range.

[image]

SwoopX, 2020-12-14:
`startup.txt` contains some of those. Never seen before, that's the old NWK + 1...

```
11:47:09:306     * neighbor: 0x00158D00054060D2 (0x468D), LQI: 255, relation: 0x01 rxOnWHenIdle: 0
11:47:09:307     * different nwk address 0x468C / 0x468D
11:47:09:308 neigbor 0x00158d00054060d2 is unknown child
```

ChristophHornung, 2020-12-14:
@SwoopX The ...0D2 is the aquara environment sensor. So far I didn't have any issues with it. It sometimes doesn't show a connection line on startup but I guess because it only sends temperature updates every so often.

SwoopX, 2020-12-14:
So all Xiaomi's actually working ok?

What you experience is actually the reason why I dumped the Eurotronic. However, This experience does not have everybody and Eurotronix is apparently reworking their firmware (see the huge thread about it).

ebaauw, 2020-12-14:
>  It shows "JIN" on their display

That means it has left the network.  I've seen that maybe once or twice in over 18 months, with 8 TRVs.

ChristophHornung, 2020-12-14:
@SwoopX The aquara temp sensor and the open/close sensor work ok. Sometimes the open close sensor has no connection line but so far as soon as I open the door it sends the message. The aquara water sensor is working only partially as it is often marked as unavailable (maybe it also only reappears when sensing water - but not seeing the status of a water sensor is not really the peace of mind it should provide ;))

You dumped Eurotronic because they also showed the same behavior? So is there any logic to when the connection is lost? For me it seems to coincide with reboots. If I just try to not reboot will things continue to work? And why is the one Spirit (the first one I bought) not having the problem? I bought that one for the single purpose of seeing if things work - and it did for that one. A bit sad that after buying the rest it turns out there are problems.

ChristophHornung, 2020-12-14:
@ebaauw It shows "JIN" and then connects. (This is the same JIN it displays on setup when pairing for the first time) So just from looking at the display on the Spirit it seems to work normally (JIN - the connects - then displays temperature and the antenna symbol).

ChristophHornung, 2020-12-15:
After re-pairing all devices the problem occurred again today, and without any reboot in between. So it seems the devices drop out during normal operation.

ChristophHornung, 2020-12-15:
Two devices lost connection, I again see both of them have the "is known by 0 neighbors" log line while all other devices are known by 1 neighbor.

ChristophHornung, 2020-12-15:
This is a part of the current log with 2 devices being lost (...045 and ...0AB)
[deconzLog1512-2lost.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5696613/deconzLog1512-2lost.txt)

@Mimiix: Is there any way to constantly write to a log file? Maybe the log shows something at the moment the connections are lost.

Mimiix, 2020-12-15:
@ChristophHornung There is:
https://github.com/home-assistant/addons/blob/master/deconz/DOCS.md#advanced-debug-output-control

The following flag values: 

--dbg-info=2 --dbg-aps=2 --dbg-error=2 --dbg-zcl=1

^Do not copy paste those, but just copy the flag values.

ChristophHornung, 2020-12-15:
@Mimiix That seems to set the log levels (the same way the debug window in deConz does). 

But I don't see how to get the logs into a more consistent file. In HA I can only use the refresh button on the log page which only shoes the last few entries, and in deConz I can use the debug window but this also does have a line limit. 

Ideally I would just want to log everything to a file somewhere and then look at a few days worth of log later on.

Mimiix, 2020-12-15:
> @Mimiix That seems to set the log levels (the same way the debug window in deConz does).
> 
> But I don't see how to get the logs into a more consistent file. In HA I can only use the refresh button on the log page which only shows the last few entries, and in deConz I can use the debug window but this also does have a line limit.
> 
> Ideally I would just want to log everything to a file somewhere and then look at a few days worth of log later on.

That is something you should ask at the HA addon place. We don't maintain that.  Aren't they logged in the home-assistant.log?

easybeat, 2020-12-15:
Hi

I have the same problem with 7 devices TRV TS0601 which are losing connection after a few hours. If I wake them up by pressing some buttons on the device or changing temperature then connection is established and changes accepted.

I'm on 2.07.00 and latest firmware.

Thanks

ChristophHornung, 2020-12-16:
The next Spririt is lost from the network. This time it looks slightly different. The connection line is still drawn in dark orange but without any LQI. The log says `Node 0x00158D00053D2E1B is known by 0 neighbors, last seen 9713 s` so it seems about 2.5 hours ago the "seen" changed. 
Why does the line still exist when the log says "0 neighbors"? And what would I expect to see in the log that corresponds to the "seen" status being updated?

ChristophHornung, 2020-12-16:
I checked the other Spirits and there is one more who also displays problems, but this one is still "known by 1 neighbor", so maybe it is in the process of disconnecting. Reading values still produces the red dot to appear and not read anything. The log looks more interesting (It is the ...C02 one)
[F7C02NoReactionButStillNeighbor.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5702123/F7C02NoReactionButStillNeighbor.txt)

ChristophHornung, 2020-12-16:
@Mimiix 
In the first log the read cluster attempted seems to end (after a lot of delays) in:

````
11:49:52:616 APS-DATA.confirm id: 115, status: 0xF0 TRANSACTION_EXPIRED
11:49:52:617 APS-DATA.confirm id: 115 status: transaction expired (28356096 seconds)
````

The `28356096 seconds` part seems strange. Maybe the problem is in the message-delay and early transaction expiry?

ChristophHornung, 2020-12-16:
I tried power cycling some of the lost devices. In contrast to the one I tried yesterday all devices seem to reconnect after a short power cycle. 
I captured the log for one (E1B). After power cycling I see it blinking green again and it seems to work normally. But the green connection line between it and the ConBeeII is not drawn for some reason. (Tried turning them off and on)
[PowerCycledE1B.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5702849/PowerCycledE1B.txt)

manup, 2020-12-16:
Can you please read the basic cluster of one of those to check the device firmware version, mine looks like this (runs without problems for 1.5 years now).

[image]

I'm a a bit confused about your log it shows both firmware versions 0x26660700 and 0x26680700, is the latest firmware version also shown in *Help &rarr; About deCONZ*?

ChristophHornung, 2020-12-16:
@manup I updated the firmware after the firmware after reporting the issue, that is why the earlier logs show the older version. In Help -> About deCONZ it says 26680700. 
The basic cluster for all devices (including the first one I bought, which so far was the only one to never loose connection) looks like this:
[image]

ChristophHornung, 2020-12-16:
Lost one of the Spirits again, it shows the same symptoms as the last one. The line is still drawn, it is still in the "known by 1 neighbors" category, but all requests end in the red bubble appearing.

I changed the temperature to test that, and noticed one strange entry in the log I didn't see before:

`21:41:40:616 0x00158D00032F70A8 error APSDE-DATA.confirm: 0xF0 on task`

otherwise the log looks similar (lots of delays and then the transaction expired with a strange "seconds" value)
Full log: 
[settingTemperatureOnLostDevice.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5705430/settingTemperatureOnLostDevice.txt)

ChristophHornung, 2020-12-16:
I had a strange occurrence of the ConBeeII reconnecting to the network. I noticed it only because I had VNC open on another screen and saw that deCONZ suddenly switched to the device selection screen before returning to the Network view.

Afterwards one more device was lost and the connection lines turned red with 0/0 indicator to each of the devices. This also happened to two of the aquara devices. This never happened before (Usually the lines either stay orange or green but the device is not responding, or they don't appear at all).

I was quick enough to catch the log from the event.

[strangeConBeeDisconnect.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5705668/strangeConBeeDisconnect.txt)

This seems to be the part where the ConBeeII disconnected:

```
2:36:47:399 Node 0x00158D00053D2DCB is known by 1 neighbors, last seen 64 s
22:36:47:879 Node 0x00158D00032F70A8 is known by 1 neighbors, last seen 44 s
22:36:47:908 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
22:36:47:909 device state timeout (handled)
22:36:48:001 COM: /dev/ttyAMA0 :  (0x0000/0x0000)
22:36:48:003 dev /dev/ttyAMA0
22:36:48:004 COM: /dev/ttyACM1 : ConBee II (0x1CF1/0x0030)
22:36:48:007 dev can't create file: /dev/ttyACM1 via mknod (missing permissions)
22:36:48:077 COM: /dev/ttyAMA0 :  (0x0000/0x0000)
22:36:48:079 dev /dev/ttyAMA0
22:36:48:080 COM: /dev/ttyACM1 : ConBee II (0x1CF1/0x0030)
22:36:48:081 dev can't create file: /dev/ttyACM1 via mknod (missing permissions)
22:36:48:083 auto connect com /dev/ttyACM1
22:36:48:085 enqueue event config/localtime for /config/
22:36:48:118 COM failed to open /dev/ttyACM1: No such file or directory
22:36:48:248 networkState: CC_ReconnectNetwork
22:36:48:250 start reconnect to network
22:36:48:296 Serial com disconnected, reason: 1
22:36:48:302 device disconnected reason: 1, index: 1
22:36:48:305 Serial com disconnected, reason: 4
22:36:48:307 device disconnected reason: 4, index: 2
22:36:48:309 aps request id: 200 prf: 0x0000 cl: 0x0031 timeout (confirmed: 0) to 0x00212EFFFF05E748 (0x0000)
22:36:48:311 aps request id: 200 prf: 0x0000 cl: 0x0031 timeout NOT confirmed to 0x00212EFFFF05E748 (0x0000)
22:36:48:391 aps request id: 200 failed, erase from queue
22:36:48:393 emit artificial APSDE-DATA.confirm id: 200
22:36:48:471 aps request id: 200 failed, erase from queue
22:36:48:908 Daylight now: nightStart, status: 230, daylight: 0, dark: 1
22:36:48:912 wait reconnect 15 seconds
22:36:48:989 COM: /dev/ttyAMA0 :  (0x0000/0x0000)
22:36:48:991 dev /dev/ttyAMA0
22:36:48:992 COM: /dev/ttyACM1 : ConBee II (0x1CF1/0x0030)
22:36:48:993 dev can't create file: /dev/ttyACM1 via mknod (missing permissions)
22:36:49:061 COM: /dev/ttyAMA0 :  (0x0000/0x0000)
22:36:49:063 dev /dev/ttyAMA0
22:36:49:064 COM: /dev/ttyACM1 : ConBee II (0x1CF1/0x0030)
22:36:49:065 dev can't create file: /dev/ttyACM1 via mknod (missing permissions)
22:36:49:067 enqueue event config/localtime for /config/
22:36:49:408 void zmMaster::handleStateIdle(zmMaster::MasterEvent) not connected goto OFF state
22:36:49:411 device state timeout ignored in state 1
22:36:49:908 wait reconnect 14 seconds
22:36:49:910 enqueue event config/localtime for /config/
22:36:50:908 wait reconnect 13 seconds
```

As the results look different than anything I saw before, this might be some separate issue though? Or maybe the root cause is the same?

ChristophHornung, 2020-12-19:
Update on the situation:

I made a few changes to my setup, namely I added two devices that act as repeaters (The Lidl Led Strip and a Lidl Power Plug). The Spirits that are connected to either of them seem to no longer drop out of the network. One Spirit that is only connected to the ConBeeII still drops out. 

It still shows the same problems with the neighbor log entry. Namely: `Node 0x00158D00032F7C02 is known by 0 neighbors, last seen -37540 s`.

So to me it seems as if the problem is that when the ConBeeII is the only neighbor it sometimes looses its "neighborness" to that Spirit. 

I can also see some log entries for the neighbor relation that do not match what is displayed.

Eg. `Node 0x00158D00032F7E16 is known by 1 neighbors, last seen 115 s` but when I look at the lines going away from it it still draws one to the ConBeeII and a second one to the new Lidl Power Plug.

Also when I reconnect the dropped Spirit it again reads in the log as "is known by 1 neighbor", but no line is drawn.

Why do the drawn lines not match what is in the log? And is there any way to debug this further? I tried the "Read Neighbor Table" on the ConBeeII, but that does not seem to change anything.

It seems to me the log is more reflective of the truth, as 0 neighbor devices are the ones not functioning. Is there a bug in the line-drawing or maybe 2 conflicting neighborhood values that might also explain the dropping Spirits?

Mimiix, 2020-12-22:
@ChristophHornung In the main topic of the Eurotronic device, a dev of them is getting some firmware issues with it fixed. 

That's with i meant with https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3911#issuecomment-744406552 > Adding routers helps a bit.

ChristophHornung, 2020-12-22:
@Mimiix That is good to hear. But the neighborhood indications are a problem with deCONZ, should I open a separate bug for that?

SwoopX, 2020-12-22:
What firmware version are you running? The last one included a fix related to neighbor table entries and devices dropping when associated with the coordinator.

ChristophHornung, 2020-12-22:
@SwoopX 26680700

SwoopX, 2020-12-23:
Thanks, would have been too easy...

rPonuganti, 2020-12-25:
I have a similar problem with Eurotronic Spirit Zigbee Thermosrat. In my case it is a bit away from the Conbee but Aqara sensor from that distance works flawless, also there is a repeater (Osram Smart+ lamp) in between. So range shouldn't be a problem.

Disconnection issue is only with Eurotronics, so am not sure if problem is with this device or with Conbee.

ChristophHornung, 2020-12-25:
@rPonuganti Do you also see the "neighborhood" log entries not matching up with what is drawn in deCONZ?

rPonuganti, 2020-12-26:
Hi, at the moment device seems to be still connected. I will wait when it goes missing to read the logs.

Also at the moment, I am using deconz via ingress, unfortunately here the windows and content are not legible. 

I am trying to setup VNC viewer but so far no success.
I am using RealVNC viewer, port set to 5900 in deconz configuration and trying to access via the link " vnc://192.168.86.60:5900" but it says invalid end point.

easybeat, 2020-12-27:
@ChristophHornung I've found a solution (at the moment). Placed a Ikea repeater next to each device. They do pick up this connection. My devices are Moes TS0601 TRV and Danfoss Ally. Not a pretty solution but seems to work.

But in general this seems to be a very strange behaviour. I just can't believe that all zigbee TRV's can have the same problem. I have these Moes and Danfoss TRV's and here it looks like if they are too far away from Conbee II (in my case), they can't build up connection untill you force them with an ikea repeater. And I have lot's other router devices in my network.

Thanks

Beat

saxn-paule, 2021-01-25:
Issue shouldn't be closed because it stil exists.

Mimiix, 2021-01-26:
@SwoopX  @ebaauw  The above issue, is that due to firmware ?

easybeat, 2021-01-26:
Yes the problem still exists and it is not only with Eurotronic TRV's...I have 5 Moes TRV's and 1 Danfoss Ally and all show the same behaviour.

Thanks

ChristophHornung, 2021-01-26:
There is an updated firmware available for the Spirit [here](https://github.com/EUROTRONIC-Technology/Spirit-ZigBee/releases/tag/20190408). I will try it and report back if the problems persist.

closed 2021-03-05 by github-actions[bot]

solvisit, 2021-04-28:
Unfortunately the problem is not related to the double Eurotronic "20191014" firmware. It happens with the previous "20190408" too and with both "20191014" versions. It happens only to TRV connected directly to the coordinator. Repeaters help, but if the device decide to connect to coordinator the problem persists. Please reopen the issue.