dresden-elektronik/deconz-rest-plugin #3914: What is the difference between "Phoscon_Gateway_2020-12-03" and "Phoscon_Gateway_Homebridge_2020-12-03"?

7wells, 2020-12-14:
## Describe the question or issue you are having
Hello,

Please forgive me my maybe dumb question. What is the difference between these 2 images?
1. Phoscon_Gateway_2020-12-03
2. Phoscon_Gateway_Homebridge_2020-12-03

I cannot find an explanation on the download site.

If my (dumb?) question is in the wrong place (I am really unsure where else to put it), please kindly move it as appropriate.

And thanks for enlighten me.

## Screenshots
NA

## Environment
NA

## deCONZ Logs
NA

## Additional context
NA

manup, 2020-12-14:
The Hombrige image has a pre-installed version of Homebridge and the Hombrige-hue plugin, which can make devices paired via deCONZ available to Apple HomeKit.

Mimiix, 2020-12-16:
I guess, depends on your end usage.

closed 2021-01-12 by 7wells