dresden-elektronik/deconz-rest-plugin #3917: Tuya TRV Moes HY368: Missing state & config properties

SebLz, 2020-12-14:
## Describe the bug

When I use deconz rest API to get info about my Moes HY368 TRV sensor, I notice that its state is missing multiple informations (preset, valve (% opening), battery level,...). Same for its config object (nothing about childlock, boost-time,.. ). Those are present in zigbee2mqtt (Koenkk/zigbee2mqtt#3821). Here is what I get when querying my TRV sensor:

```
{
    "config": {
        "heatsetpoint": 1800,
        "offset": -100,
        "on": true,
        "reachable": true,
        "schedule": {
            "W124": [
                {
                    "heatsetpoint": 20,
                    "localtime": "T06:00"
                },
                {
                    "heatsetpoint": 15,
                    "localtime": "T08:00"
                },
                {
                    "heatsetpoint": 15,
                    "localtime": "T11:30"
                },
                {
                    "heatsetpoint": 15,
                    "localtime": "T12:30"
                },
                {
                    "heatsetpoint": 20,
                    "localtime": "T17:30"
                },
                {
                    "heatsetpoint": 15,
                    "localtime": "T22:00"
                }
            ],
            "W3": [
                {
                    "heatsetpoint": 20,
                    "localtime": "T134:00"
                },
                {
                    "heatsetpoint": 15,
                    "localtime": "T08:00"
                },
                {
                    "heatsetpoint": 15,
                    "localtime": "T11:30"
                },
                {
                    "heatsetpoint": 15,
                    "localtime": "T12:30"
                },
                {
                    "heatsetpoint": 20,
                    "localtime": "T17:30"
                },
                {
                    "heatsetpoint": 15,
                    "localtime": "T22:00"
                }
            ]
        },
        "schedule_on": null
    },
    "ep": 1,
    "etag": "7ef41a669c11ab3e17edd103257c0aa1",
    "lastseen": "2020-12-14T18:24Z",
    "manufacturername": "_TZE200_ckud7u2l",
    "modelid": "TS0601",
    "name": "Radiateur du bureau",
    "state": {
        "lastupdated": "none",
        "on": true,
        "temperature": 1750
    },
    "type": "ZHAThermostat",
    "uniqueid": "5c:02:72:ff:fe:8f:83:57-01-0201"
}
```

## Steps to reproduce the behavior

1. Pair HY368 with deconz
2. Use deconz rest API get sensor  

## Expected behavior

No missing properties in State and Config object in the json response

## Screenshots

## Environment

[image]
[image]

 - Host system: Odroid C4 ubuntu
 - Running method: (Home Assistant deCONZ Add-on)
 - Firmware version: 26660700
 - deCONZ version: 2.07.01 / 08/12/2020
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? yes, a bluetooth stick and a ssd

## deCONZ Logs

--
## Additional context

--

Smanar, 2020-12-14:
Hello, so first thing to check the model id, here you have "manufacturername": "_TZE200_ckud7u2l", good, take a look on that, it seem it s replaced by "heiman" on unknow situation.

Secondly, can you just restart deconz, now you have the correct manufacture name ?

SebLz, 2020-12-14:
I've restarted deconz and HA. The manufacturer name remained the same and many properties appeared !

```
{
    "config": {
        "heatsetpoint": 1800,
        "locked": null,
        "mode": "off",
        "offset": -100,
        "on": true,
        "preset": "manual",
        "reachable": true,
        "schedule": {
            "W124": [
                {
                    "heatsetpoint": 20,
                    "localtime": "T06:00"
                },
                {
                    "heatsetpoint": 15,
                    "localtime": "T08:00"
                },
                {
                    "heatsetpoint": 15,
                    "localtime": "T11:30"
                },
                {
                    "heatsetpoint": 15,
                    "localtime": "T12:30"
                },
                {
                    "heatsetpoint": 20,
                    "localtime": "T17:30"
                },
                {
                    "heatsetpoint": 15,
                    "localtime": "T22:00"
                }
            ],
            "W3": [
                {
                    "heatsetpoint": 20,
                    "localtime": "T134:00"
                },
                {
                    "heatsetpoint": 15,
                    "localtime": "T08:00"
                },
                {
                    "heatsetpoint": 15,
                    "localtime": "T11:30"
                },
                {
                    "heatsetpoint": 15,
                    "localtime": "T12:30"
                },
                {
                    "heatsetpoint": 20,
                    "localtime": "T17:30"
                },
                {
                    "heatsetpoint": 15,
                    "localtime": "T22:00"
                }
            ]
        },
        "schedule_on": null,
        "setvalve": true,
        "windowopen_set": null
    },
    "ep": 1,
    "etag": "3e71b83887913f3419759b428223f2b7",
    "lastseen": "2020-12-14T20:48Z",
    "manufacturername": "_TZE200_ckud7u2l",
    "modelid": "TS0601",
    "name": "Radiateur du bureau",
    "state": {
        "lastupdated": "none",
        "lowbattery": false,
        "on": null,
        "temperature": 1850,
        "valve": null
    },
    "type": "ZHAThermostat",
    "uniqueid": "5c:02:72:ff:fe:8f:83:57-01-0201"
}
```

Thanks for your fast and effective feedback @Smanar !

Mimiix, 2020-12-15:
Case closed?

closed 2020-12-15 by SebLz

Mimiix, 2020-12-15:
Case closed!