dresden-elektronik/deconz-rest-plugin #3922: HiHome WZB-TRVL (Smart Zigbee Radiator Thermostat)

KPWhiver, 2020-12-15:
## Device

- Product name: HiHome WZB-TRVL (Smart Zigbee Radiator Thermostat)
- Manufacturer: _TYST11_zuhszj9s
- Model identifier: uhszj9s
- Device type:
  - Thermostat

## Screenshots

[image]
(Note: this has been taken using this commit: https://github.com/KPWhiver/deconz-rest-plugin/commit/287a874cb026ab8c4d67e1a71141a07bb5deadde, without this commit it gets the name On/Off Light, but all the node/cluster info is the same.)

## Further information
In the zigbee2mqtt project this device is implemented as a whitelabel of the Saswell-SEA802, which is also discussed at the end of this issue: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3109, and support for which has been added already.
Manufacturer: _TYST11_c88teujp
Model id: 88teujp
As such I tried to implement the HiHome thermostat myself in this commit: https://github.com/KPWhiver/deconz-rest-plugin/commit/287a874cb026ab8c4d67e1a71141a07bb5deadde but although this makes the device recognized as a thermostat in de Phoscon app and in Home Assistant, I can not control the device in Home Assistant nor do I see any temperature readings or other data.

Smanar, 2020-12-15:
I have checked the PR, all seem fine, do you have the device JSON ?
If you can see it in phoscon or HA, for me it s ok.

KPWhiver, 2020-12-15:
Json for one where I already tried to change some settings through Home Assistant:
``` json
{  
    "config":{
        "heatsetpoint":1800,
        "locked":null,
        "offset":0,
        "on":true,
        "preset":"manual",
        "reachable":true,
        "schedule":{
            
        },
        "schedule_on":null,
        "setvalve":null,
        "windowopen_set":null
    },
    "ep":1,
    "etag":"cfbaafa06c6f78decbdb330b2559f397",
    "lastseen":"2020-12-15T19:04Z",
    "manufacturername":"_TYST11_zuhszj9s",
    "modelid":"uhszj9s",
    "name":"Thermostat 25",
    "state":{
        "lastupdated":"none",
        "on":null,
        "temperature":null
    },
    "swversion":"20180727",
    "type":"ZHAThermostat",
    "uniqueid":"60:a4:23:ff:fe:8e:2c:e7-01-0201"
}
```

Json for a device where did not change anything through Home Assistant:
``` json
{
    "config":{
        "heatsetpoint":null,
        "locked":null,
        "offset":0,
        "on":true,
        "preset":null,
        "reachable":true,
        "schedule":{
            
        },
        "schedule_on":null,
        "setvalve":null,
        "windowopen_set":null
    },
    "ep":1,
    "etag":"92aa8c815644d951279f554b6771a079",
    "lastseen":"2020-12-15T19:04Z",
    "manufacturername":"_TYST11_zuhszj9s",
    "modelid":"uhszj9s",
    "name":"uhszj9s",
    "state":{
        "lastupdated":"none",
        "on":null,
        "temperature":null
    },
    "swversion":"20180727",
    "type":"ZHAThermostat",
    "uniqueid":"60:a4:23:ff:fe:b5:a9:53-01-0201"
}
```

KPWhiver, 2020-12-15:
By the way, even though I set the heatsetpoint through Home Assistant (to 18 degrees), it did not actually change on the thermostat.

KPWhiver, 2020-12-15:
Here is an excerpt from the logs concerning the thermostats:
```
Dec 15 20:42:09 server conmon[29112]: 20:42:09:588 Tuya debug 4 : Address 0x60A423FFFE8E2CE7 Payload 00016501000101
Dec 15 20:42:09 server conmon[29112]: 20:42:09:589 Tuya debug 5 : Status: 0 Transid: 1 Dp: 357 (0x01,0x65) Fn: 0 Data 1
Dec 15 20:42:09 server conmon[29112]: 20:42:09:815 DeviceAnnce of SensorNode: 0x60A423FFFEB5A953 [1]
Dec 15 20:42:12 server conmon[29112]: 20:42:12:383 Tuya debug 4 : Address 0x60A423FFFEB5A953 Payload 00016501000101
Dec 15 20:42:12 server conmon[29112]: 20:42:12:383 Tuya debug 5 : Status: 0 Transid: 1 Dp: 357 (0x01,0x65) Fn: 0 Data 1
Dec 15 20:42:13 server conmon[29112]: 20:42:13:598 Tuya debug 4 : Address 0x60A423FFFE8E2CE7 Payload 00016501000101
Dec 15 20:42:13 server conmon[29112]: 20:42:13:599 Tuya debug 5 : Status: 0 Transid: 1 Dp: 357 (0x01,0x65) Fn: 0 Data 1
Dec 15 20:42:16 server conmon[29112]: 20:42:16:102 Tuya debug 4 : Address 0x60A423FFFEB5A953 Payload 00016501000101
Dec 15 20:42:16 server conmon[29112]: 20:42:16:102 Tuya debug 5 : Status: 0 Transid: 1 Dp: 357 (0x01,0x65) Fn: 0 Data 1
Dec 15 20:42:20 server conmon[29112]: 20:42:20:100 Tuya debug 4 : Address 0x60A423FFFEB5A953 Payload 00016501000101
Dec 15 20:42:20 server conmon[29112]: 20:42:20:101 Tuya debug 5 : Status: 0 Transid: 1 Dp: 357 (0x01,0x65) Fn: 0 Data 1
Dec 15 20:42:20 server conmon[29112]: 20:42:20:309 DeviceAnnce of SensorNode: 0x60A423FFFE8E2CE7 [1]
Dec 15 20:42:22 server conmon[29112]: 20:42:22:880 Tuya debug 4 : Address 0x60A423FFFE8E2CE7 Payload 00016501000101
Dec 15 20:42:22 server conmon[29112]: 20:42:22:880 Tuya debug 5 : Status: 0 Transid: 1 Dp: 357 (0x01,0x65) Fn: 0 Data 1
Dec 15 20:42:24 server conmon[29112]: 20:42:24:101 Tuya debug 4 : Address 0x60A423FFFEB5A953 Payload 00016501000101
Dec 15 20:42:24 server conmon[29112]: 20:42:24:102 Tuya debug 5 : Status: 0 Transid: 1 Dp: 357 (0x01,0x65) Fn: 0 Data 1
Dec 15 20:42:26 server conmon[29112]: 20:42:26:592 Tuya debug 4 : Address 0x60A423FFFE8E2CE7 Payload 00016501000101
Dec 15 20:42:26 server conmon[29112]: 20:42:26:593 Tuya debug 5 : Status: 0 Transid: 1 Dp: 357 (0x01,0x65) Fn: 0 Data 1
Dec 15 20:42:28 server conmon[29112]: 20:42:28:099 Tuya debug 4 : Address 0x60A423FFFEB5A953 Payload 00016501000101
Dec 15 20:42:28 server conmon[29112]: 20:42:28:100 Tuya debug 5 : Status: 0 Transid: 1 Dp: 357 (0x01,0x65) Fn: 0 Data 1
Dec 15 20:42:30 server conmon[29112]: 20:42:30:584 Tuya debug 4 : Address 0x60A423FFFE8E2CE7 Payload 00016501000101
Dec 15 20:42:30 server conmon[29112]: 20:42:30:585 Tuya debug 5 : Status: 0 Transid: 1 Dp: 357 (0x01,0x65) Fn: 0 Data 1
Dec 15 20:42:34 server conmon[29112]: 20:42:34:598 Tuya debug 4 : Address 0x60A423FFFE8E2CE7 Payload 00016501000101
Dec 15 20:42:34 server conmon[29112]: 20:42:34:599 Tuya debug 5 : Status: 0 Transid: 1 Dp: 357 (0x01,0x65) Fn: 0 Data 1
Dec 15 20:42:34 server conmon[29112]: 20:42:34:808 DeviceAnnce of SensorNode: 0x60A423FFFEB5A953 [1]
Dec 15 20:42:37 server conmon[29112]: 20:42:37:388 Tuya debug 4 : Address 0x60A423FFFEB5A953 Payload 00016501000100
Dec 15 20:42:37 server conmon[29112]: 20:42:37:389 Tuya debug 5 : Status: 0 Transid: 1 Dp: 357 (0x01,0x65) Fn: 0 Data 0
Dec 15 20:42:38 server conmon[29112]: 20:42:38:593 Tuya debug 4 : Address 0x60A423FFFE8E2CE7 Payload 00016501000101
Dec 15 20:42:38 server conmon[29112]: 20:42:38:594 Tuya debug 5 : Status: 0 Transid: 1 Dp: 357 (0x01,0x65) Fn: 0 Data 1
Dec 15 20:42:41 server conmon[29112]: 20:42:41:102 Tuya debug 4 : Address 0x60A423FFFEB5A953 Payload 00016501000100
Dec 15 20:42:41 server conmon[29112]: 20:42:41:103 Tuya debug 5 : Status: 0 Transid: 1 Dp: 357 (0x01,0x65) Fn: 0 Data 0
Dec 15 20:42:45 server conmon[29112]: 20:42:45:100 Tuya debug 4 : Address 0x60A423FFFEB5A953 Payload 00016501000100
Dec 15 20:42:45 server conmon[29112]: 20:42:45:101 Tuya debug 5 : Status: 0 Transid: 1 Dp: 357 (0x01,0x65) Fn: 0 Data 0
Dec 15 20:42:45 server conmon[29112]: 20:42:45:390 DeviceAnnce of SensorNode: 0x60A423FFFE8E2CE7 [1]
```
The Dp: 357 message seems to be whether the thermostat is on or off, if I rotate the button on it to the off position Data becomes 0, whereas otherwise it is 1.

KPWhiver, 2020-12-16:
Comparing the zigbee2mqtt implementation with this one, it seems that these "Saswell" devices do use the Tuya cluster, but they use a different set of Dp's than normal Tuya radiator thermostat. I'll try to fix this in my PR and see if that improves things.

Smanar, 2020-12-16:
Ha yes right, look at the tuya file
```
                case 0x0267: // max temperature limit
                {
                    //can be setpoint for some device
                    if (sensorNode->modelId() == QLatin1String("GbxAXL2"))
                    {
                        qint16 temp = (static_cast<qint16>(data & 0xFFFF)) * 10;
                        ResourceItem *item = sensorNode->item(RConfigHeatSetpoint);

                        if (item && item->toNumber() != temp)
                        {
                            item->setValue(temp);
                            Event e(RSensors, RConfigHeatSetpoint, sensorNode->id(), item);
                            enqueueEvent(e);

                        }
                    }
                }
                break;
```

You can add your device here, but pls use manufacturername isntead of model id

This is the part for setpoint, but it seem your device mimic the GbxAXL2

KPWhiver, 2020-12-17:
Alright, I have now changed this device to use the same dp's as is done in z2m (basically the same as GbxAXL2): https://github.com/dresden-elektronik/deconz-rest-plugin/compare/master...KPWhiver:master

I can now change the setpoint, change the windowopen mode, change it to auto/manu/off, and change the temperature offset. In all these cases confirmation is sent from the device that the change has been made. However, the device still does not send the local temperature or current setpoint, it only sends the current valve state (dp: 357) every few seconds. Any idea on how to get it to send the temperature?

The only thing I can think of is that (according to the z2m code) this device expects to be send the current time whenever it does a DeviceAnnounce. Perhaps it won't send the temperature/setpoint until the time has been sent? Unfortunately, I have no clue how to implement this in deconz.

Smanar, 2020-12-17:
Ha ?
It's not because the device is lazy and make few report ?
I don't remember there is this kind of issue for GbxAXL2

It s the tuya time command ? I have started the code here, but no one to test it ... https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3304#issuecomment-740215205

If you have a display for time on your device, It can help me to finish this code too .

KPWhiver, 2020-12-18:
They have both been connected for days and in that time I only saw the DP: 357 message. 

It is indeed the tuya time command you describe in #3304. The device does not have a display for time, but I am willing to test your code and see if it does anything. How can I test your code?

Smanar, 2020-12-18:
What have you tried ?
I prefer let this PR clean, but if you want to make test I can make another PR that mix your PR and mine (or if you want to do it on your side ?)

With both PR you will have the device working, and I hope the new debug line starting with "Tuya debug 1 : Time sync Request"

Smanar, 2020-12-19:
@KPWhiver do you want I make a branch with the 2 PR to test it ?

KPWhiver, 2020-12-19:
That would be great

Smanar, 2020-12-20:
Done, pls use the branch "tuyaspecial"
```
git pull
git branch tuyaspecial
git pull
```

Not sure this code will work, I have never tested on tuya device, it was made for a device that not use this method.
But you will have usefull log for me.

KPWhiver, 2020-12-20:
This device does not send an time request through a message, but instead expects a time when it does a deviceAnnounce. I changed the code to call SendTuyaCommand with the result of getTIme when a deviceAnnounce happens, but unfortunately nothing seems to happen, still no temperature data:
```
Dec 20 22:50:30 server conmon[14273]: 22:50:30:058 Tuya debug 4 : Address 0x60A423FFFEB5A953 Payload 00016501000100
Dec 20 22:50:30 server conmon[14273]: 22:50:30:058 Tuya debug 5 : Status: 0 Transid: 1 Dp: 357 (0x01,0x65) Fn: 0 Data 0
Dec 20 22:50:34 server conmon[14273]: 22:50:34:058 Tuya debug 4 : Address 0x60A423FFFEB5A953 Payload 00016501000100
Dec 20 22:50:34 server conmon[14273]: 22:50:34:058 Tuya debug 5 : Status: 0 Transid: 1 Dp: 357 (0x01,0x65) Fn: 0 Data 0
Dec 20 22:50:38 server conmon[14273]: 22:50:38:058 Tuya debug 4 : Address 0x60A423FFFEB5A953 Payload 00016501000100
Dec 20 22:50:38 server conmon[14273]: 22:50:38:058 Tuya debug 5 : Status: 0 Transid: 1 Dp: 357 (0x01,0x65) Fn: 0 Data 0
Dec 20 22:50:44 server conmon[14273]: 22:50:44:742 DeviceAnnce of SensorNode: 0x60A423FFFEB5A953 [1]
Dec 20 22:50:44 server conmon[14273]: 22:50:44:743 Send Tuya Command 0x24 Data: 277283b4277291c4
Dec 20 22:50:47 server conmon[14273]: 22:50:47:349 Tuya debug 4 : Address 0x60A423FFFEB5A953 Payload 00016501000100
Dec 20 22:50:47 server conmon[14273]: 22:50:47:350 Tuya debug 5 : Status: 0 Transid: 1 Dp: 357 (0x01,0x65) Fn: 0 Data 0
Dec 20 22:50:51 server conmon[14273]: 22:50:51:057 Tuya debug 4 : Address 0x60A423FFFEB5A953 Payload 00016501000100
Dec 20 22:50:51 server conmon[14273]: 22:50:51:058 Tuya debug 5 : Status: 0 Transid: 1 Dp: 357 (0x01,0x65) Fn: 0 Data 0
```

KPWhiver, 2020-12-21:
Just to be sure this device actually supports sending a temperature I ran zigbee2mqtt with my conbee 2 and paired it. Zigbee2mqtt pairs fine and receives temperature data, which means that it is doing something that deconz isn't. For reference, here are the zigbee2mqtt logs:
```
Dec 21 17:33:19 server zigbee2mqtt[6343]: Zigbee2MQTT:debug 2020-12-21 17:33:19: Can't load state from file /var/lib/zigbee2mqtt/.z2m/state.json (doesn't exist)
Dec 21 17:33:19 server zigbee2mqtt[6343]: Zigbee2MQTT:info  2020-12-21 17:33:19: Logging to console and directory: '/var/lib/zigbee2mqtt/.z2m/log/2020-12-21.17-33-19' filename: log.txt
Dec 21 17:33:19 server zigbee2mqtt[6343]: Zigbee2MQTT:info  2020-12-21 17:33:19: Starting Zigbee2MQTT version 1.16.2 (commit #unknown)
Dec 21 17:33:19 server zigbee2mqtt[6343]: Zigbee2MQTT:info  2020-12-21 17:33:19: Starting zigbee-herdsman (0.13.37)
Dec 21 17:33:19 server zigbee2mqtt[6343]: Zigbee2MQTT:debug 2020-12-21 17:33:19: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null},"backupPath":"/var/lib/zigbee2mqtt/.z2m/coordinator_backup.json","databaseBackupPath":"/var/lib/zigbee2mqtt/.z2m/database.db.backup","databasePath":"/var/lib/zigbee2mqtt/.z2m/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyACM0"}}'
Dec 21 17:33:20 server zigbee2mqtt[6343]: CREATED DECONZ ADAPTER
Dec 21 17:33:20 server zigbee2mqtt[6343]: Zigbee2MQTT:info  2020-12-21 17:33:20: zigbee-herdsman started
Dec 21 17:33:20 server zigbee2mqtt[6343]: Zigbee2MQTT:info  2020-12-21 17:33:20: Coordinator firmware version: '{"meta":{"maintrel":0,"majorrel":38,"minorrel":102,"product":0,"revision":"0x26660700","transportrev":0},"type":"ConBee2/RaspBee2"}'
Dec 21 17:33:34 server zigbee2mqtt[6343]: Zigbee2MQTT:debug 2020-12-21 17:33:34: Zigbee network parameters: {"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754}
Dec 21 17:33:34 server zigbee2mqtt[6343]: Zigbee2MQTT:info  2020-12-21 17:33:34: Currently 0 devices are joined:
Dec 21 17:33:34 server zigbee2mqtt[6343]: Zigbee2MQTT:warn  2020-12-21 17:33:34: `permit_join` set to  `true` in configuration.yaml.
Dec 21 17:33:34 server zigbee2mqtt[6343]: Zigbee2MQTT:warn  2020-12-21 17:33:34: Allowing new devices to join.
Dec 21 17:33:34 server zigbee2mqtt[6343]: Zigbee2MQTT:warn  2020-12-21 17:33:34: Set `permit_join` to `false` once you joined all devices.
Dec 21 17:33:34 server zigbee2mqtt[6343]: Zigbee2MQTT:info  2020-12-21 17:33:34: Zigbee: allowing new devices to join.
Dec 21 17:33:35 server zigbee2mqtt[6343]: Zigbee2MQTT:info  2020-12-21 17:33:35: Connecting to MQTT server at mqtt://localhost:1883
Dec 21 17:33:35 server zigbee2mqtt[6343]: Zigbee2MQTT:debug 2020-12-21 17:33:35: Using MQTT anonymous login
Dec 21 17:33:45 server zigbee2mqtt[6343]: Zigbee2MQTT:error 2020-12-21 17:33:45: Not connected to MQTT server!
Dec 21 17:33:55 server zigbee2mqtt[6343]: Zigbee2MQTT:error 2020-12-21 17:33:55: Not connected to MQTT server!
Dec 21 17:33:59 server zigbee2mqtt[6343]: Zigbee2MQTT:info  2020-12-21 17:33:59: Device '0x60a423fffeb5a953' joined
Dec 21 17:33:59 server zigbee2mqtt[6343]: Zigbee2MQTT:info  2020-12-21 17:33:59: Starting interview of '0x60a423fffeb5a953'
Dec 21 17:33:59 server zigbee2mqtt[6343]: Zigbee2MQTT:error 2020-12-21 17:33:59: Not connected to MQTT server!
Dec 21 17:33:59 server zigbee2mqtt[6343]: Zigbee2MQTT:error 2020-12-21 17:33:59: Cannot send message: topic: 'zigbee2mqtt/bridge/log', payload: '{"message":{"friendly_name":"0x60a423fffeb5a953"},"type":"device_connected"}
Dec 21 17:33:59 server zigbee2mqtt[6343]: Zigbee2MQTT:error 2020-12-21 17:33:59: Not connected to MQTT server!
Dec 21 17:33:59 server zigbee2mqtt[6343]: Zigbee2MQTT:error 2020-12-21 17:33:59: Cannot send message: topic: 'zigbee2mqtt/bridge/log', payload: '{"message":"interview_started","meta":{"friendly_name":"0x60a423fffeb5a953"},"type":"pairing"}
Dec 21 17:34:04 server zigbee2mqtt[6343]: Zigbee2MQTT:debug 2020-12-21 17:34:04: Received Zigbee message from '0x60a423fffeb5a953', type 'readResponse', cluster 'genBasic', data '{"modelId":"uhszj9s\u0000"}' from endpoint 1 with groupID null
Dec 21 17:34:04 server zigbee2mqtt[6343]: Zigbee2MQTT:debug 2020-12-21 17:34:04: Skipping message, definition is undefined and still interviewing
Dec 21 17:34:05 server zigbee2mqtt[6343]: Zigbee2MQTT:error 2020-12-21 17:34:05: Not connected to MQTT server!
Dec 21 17:34:06 server zigbee2mqtt[6343]: Zigbee2MQTT:debug 2020-12-21 17:34:06: Received Zigbee message from '0x60a423fffeb5a953', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"_TYST11_zuhszj9s"}' from endpoint 1 with groupID null
Dec 21 17:34:08 server zigbee2mqtt[6343]: Zigbee2MQTT:debug 2020-12-21 17:34:08: Received Zigbee message from '0x60a423fffeb5a953', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID null
Dec 21 17:34:09 server zigbee2mqtt[6343]: Zigbee2MQTT:debug 2020-12-21 17:34:09: Received Zigbee message from '0x60a423fffeb5a953', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":101,"fn":0,"status":0,"transid":1}' from endpoint 1 with groupID null
Dec 21 17:34:09 server zigbee2mqtt[6343]: Zigbee2MQTT:error 2020-12-21 17:34:09: Not connected to MQTT server!
[... cut ...] st_detection":"ON","linkquality":225,"local_temperature":"17.4","local_temperature_calibration":0,"preset_mode":"none","system_mode":"off","window_detection":"ON"}
Dec 21 17:37:35 server zigbee2mqtt[6343]: Zigbee2MQTT:error 2020-12-21 17:37:35: Not connected to MQTT server!
Dec 21 17:37:45 server zigbee2mqtt[6343]: Zigbee2MQTT:error 2020-12-21 17:37:45: Not connected to MQTT server!
Dec 21 17:37:55 server zigbee2mqtt[6343]: Zigbee2MQTT:error 2020-12-21 17:37:55: Not connected to MQTT server!
```

Smanar, 2020-12-21:
But there is only "Cannot send message" and "Received Zigbee message" so it don't send more request.

Do you have the link about this device on z2m ?

KPWhiver, 2020-12-21:
The device is defined here: https://github.com/Koenkk/zigbee-herdsman-converters/blob/4ec1778b372ab3cd9063140ad1d26d181324ff5a/devices.js#L12813, I'll see if I can get some logging of the outgoing messages z2m sends. However, I do notice that the device starts sending values as soon as the configuring is done. The configuring in z2m consists mostly of binding the 'genBasic' cluster, which is done here: https://github.com/Koenkk/zigbee-herdsman/blob/71e8c1663cbc41cf53a32ca5861355d58e60325c/src/adapter/deconz/adapter/deconzAdapter.ts#L654 . Perhaps this is not done correctly in deconz? I certainly can't find the 'genBasic' string in the deconz codebase.

Smanar, 2020-12-22:
Binding for the basic cluster ? And for a tuya device ?

Its something easy to do with deconz, using the bind dropbox, but not sure it will change something, do you know hw to use it ?

I m seeing 2 special options in their code, ignore_tuya_set_time and ignore_basic_report

ignore_tuya_set_time ignore the command 0x24 on Tuya cluster. So that would mean, this device don't use this command ?

BTW (probably for other device) from here https://git.koehlerweb.org/frodovdr/Sonoff-Tasmota/src/commit/ecf5ad023261a991835553a271e0fca8951c077b/tasmota/xdrv_16_tuyamcu.ino#L804 , they are using another  code, different than this one
```
#ifdef USE_TUYA_TIME
void TuyaSetTime(void) {
  if (!RtcTime.valid) { return; }

  uint16_t payload_len = 8;
  uint8_t payload_buffer[8];
  payload_buffer[0] = 0x01;
  payload_buffer[1] = (uint8_t)RtcTime.year;
  payload_buffer[2] = RtcTime.month;
  payload_buffer[3] = RtcTime.day_of_month;
  payload_buffer[4] = RtcTime.hour;
  payload_buffer[5] = RtcTime.minute;
  payload_buffer[6] = RtcTime.second;
  payload_buffer[7] = RtcTime.day_of_week;

  TuyaSendCmd(TUYA_CMD_SET_TIME, payload_buffer, payload_len);
}
#endif //USE_TUYA_TIME
```

KPWhiver, 2020-12-22:
Looking at the z2m logs again I notice that before it binds it already receives messages with a Dp of 10, which never happens with deconz (only DP 101 (357 in deconz)). I removed the z2m database and paired again and this time it even already sent a local temperature before the cluster was bound, so I now doubt this is actually necessary. Here is a log of messages that are send and received by z2m during the pairing: (apologies for the size)
```
controller:log Device announce '0x60a423fffeb5a953'
controller:log Device announce is from unknown device '0x60a423fffeb5a953'
controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":3,"manufacturerCode":null,"commandIdentifier":1},"Payload":{"status":0,"transid":1,"dp":101,"datatype":1,"fn":0,"data":{"type":"Buffer","data":[0]}}},"address":63383,"endpoint":1,"linkquality":36,"groupID":null}'
controller:log 'zcl' data is from unknown device with address '63383', skipping...
controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":4,"manufacturerCode":null,"commandIdentifier":1},"Payload":{"status":0,"transid":1,"dp":101,"datatype":1,"fn":0,"data":{"type":"Buffer","data":[0]}}},"address":63383,"endpoint":1,"linkquality":36,"groupID":null}'
controller:log 'zcl' data is from unknown device with address '63383', skipping...
controller:log Device '0x60a423fffe8e2ce7' joined
controller:log Device '0x60a423fffe8e2ce7' accepted by handler
controller:log New device '0x60a423fffe8e2ce7' joined
controller:log Creating device '0x60a423fffe8e2ce7'
controller:log Interview '0x60a423fffe8e2ce7' start
controller:device:log Interview - start device '0x60a423fffe8e2ce7'
controller:endpoint Read 0x60a423fffe8e2ce7/1 genBasic(["modelId"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null})
controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":3,"manufacturerCode":null,"commandIdentifier":1},"Payload":[{"attrId":5,"status":0,"dataType":66,"attrData":"uhszj9s\u0000"}]},"address":63383,"endpoint":1,"linkquality":36,"groupID":null}'
controller:endpoint Read 0x60a423fffe8e2ce7/1 genBasic(["manufacturerName"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null})
controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":0,"manufacturerCode":null,"commandIdentifier":1},"Payload":{"status":0,"transid":1,"dp":101,"datatype":1,"fn":0,"data":{"type":"Buffer","data":[0]}}},"address":63383,"endpoint":1,"linkquality":36,"groupID":null}'
controller:endpoint DefaultResponse 0x60a423fffe8e2ce7/1 61184(1, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null})
controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":4,"manufacturerCode":null,"commandIdentifier":1},"Payload":[{"attrId":4,"status":0,"dataType":66,"attrData":"_TYST11_zuhszj9s"}]},"address":63383,"endpoint":1,"linkquality":36,"groupID":null}'
controller:endpoint Read 0x60a423fffe8e2ce7/1 genBasic(["powerSource"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null})
controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":5,"manufacturerCode":null,"commandIdentifier":1},"Payload":[{"attrId":7,"status":0,"dataType":48,"attrData":3}]},"address":63383,"endpoint":1,"linkquality":36,"groupID":null}'
controller:endpoint Read 0x60a423fffe8e2ce7/1 genBasic(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null})
controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":1,"manufacturerCode":null,"commandIdentifier":1},"Payload":{"status":0,"transid":23,"dp":10,"datatype":1,"fn":0,"data":{"type":"Buffer","data":[1]}}},"address":63383,"endpoint":1,"linkquality":36,"groupID":null}'
[... cut ...]
controller:endpoint DefaultResponse 0x60a423fffe8e2ce7/1 61184(1, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null})
```

In the logs you can see that the device starts to send messages with a Dp other than 101 as soon as the ZCL version parameter is read, this is also the case in the previous log I showed. Reading this value manually in the deconz app does not do anything.

Smanar, 2020-12-23:
0x0101 is the on/off mode.
And deconz is not configured to use this command for this device (the only one using this command is "_TZE200_aoclfnxz"

But it seem z2m is only asking for basic cluster attribute, I don't see how it can unlock something ....

KPWhiver, 2020-12-24:
I can give you the full zigbee2mqtt log, very long, but it includes logs from the deconz adapter driver in z2m.

Going through the logs and z2m code myself I see that this is the order of the messages send and received:
```
HiHome -> DeviceAnnounce
HiHome -> Dp 101
HiHome -> Dp 101
-- PermitJoin on
HiHome -> DeviceAnnounce
HiHome <- RequestNodeDescriptor
HiHome -> NodeDescriptor
HiHome <- RequestSimpleDescriptor
HiHome -> SimpleDescriptor
HiHome <- Read modelId
HiHome -> modelId
HiHome <- Read manufacturerName
HiHome -> Dp 101
HiHome -> manufacturerName
HiHome <- Read powerSource
HiHome -> powerSource
HiHome <- Read zclVersion
HiHome -> Dp 10
HiHome -> zclVersion
HiHome <- Read appVersion
HiHome -> Dp 8
HiHome -> appVersion
HiHome <- Read stackVersion
HiHome -> Dp 130
HiHome -> stackVersion
HiHome <- Read hwVersion
HiHome -> Dp 27
HiHome -> hwVersion
HiHome <- Read dateCode
HiHome -> Dp 40
HiHome -> dateCode
HiHome <- ReadswBuildId
HiHome -> Dp 102
HiHome -> swBuildId unsupported
HiHome <- Bind cluster 0
HiHome -> Bind succeeded
-- Lots more Dp messages from HiHome
```

Smanar, 2020-12-25:
Ha yes right to check "dp":101,"datatype":1, so 0x01 0x65 and we use it too.
But I don't see something special before the 

>HiHome -> Dp 102

On deconz do you have the "debug tuya" list just after the inclusion ?

But use this change first pls
https://github.com/Smanar/deconz-rest-plugin/commit/6a9e7a4d6c81defd8a0426d7aaf0098afa87821a

I think this change will prevent device spam.

Smanar, 2020-12-25:
Ha ? it s an option ?

> 
>  Local temperature #
> 
> If you’d like to force device to send local_temperature you can use this mqtt command:
> 
>     topic: zigbee2mqtt/FRIENDLY_NAME/set/local_temperature_calibration
>     payload: YOUR_CURRENT_CALIBRATION_VALUE
> 
> YOUR_CURRENT_CALIBRATION_VALUE can be 0, but if you calibrated temperature for this device send current value. After this command thermostat responds with two messages. One for calibration change confirmation, and other with current local_temperature.

KPWhiver, 2020-12-25:
I already tried this actually by sending the temperature offset to the device through the REST API. Unfortunately, all I got back was confirmation that the temperature offset had been changed. (I also send all the other values that deconz supports, such as locked, with only a confirmation as result.

I added code to my copy of deconz to explicitly read all the Basic cluster values when the device is added, will test tonight or tomorrow.

Smanar, 2020-12-25:
Never trust confirmations from API, they just mean the request is correct and send to the device, it don't mean the device received it or support it.

KPWhiver, 2020-12-29:
Progress! Your change: https://github.com/Smanar/deconz-rest-plugin/commit/6a9e7a4d6c81defd8a0426d7aaf0098afa87821a works, I now get messages other than the on/off state (such as temperatures). The problem now is that the device sends all its messages with commandId 0x01, which means that a default response is send after every message, after which it is discarded due to the else if:
```
  }
    else if (//isXmasLightStrip(lightNode) &&
             zclFrame.commandId() == 0x01 &&
             !(zclFrame.frameControl() & deCONZ::ZclFCDisableDefaultResponse))
    {
        sendZclDefaultResponse(ind, zclFrame, deCONZ::ZclSuccessStatus);
        DBG_Printf(DBG_INFO, "Tuya debug 42 : Sending default response to 0x%016llX\n" , ind.srcAddress().ext());
    }
    else if ( (zclFrame.commandId() == 0x01) || (zclFrame.commandId() == 0x02) )
    {
```

I assume it is safe to replace the else if with an if, e.g., it is safe to first send the default response and then act upon the message?

Smanar, 2020-12-29:
Ha yes, sorry, have edited the code again, cleaner method.

But so now it s working ?

KPWhiver, 2020-12-29:
Yes, just need to make sure it gets a proper name and manufacturer when paired, I'll fix that and make a PR.

Smanar, 2020-12-29:
ok So I don't make it on my side ?

Take care I have changed a little the defaut response part.

Smanar, 2021-01-02:
@KPWhiver , hello, do you wana make the PR ? Want to add some chnage on the schedule.

And BTW happy new year ^^

KPWhiver, 2021-01-02:
Thanks, you too! Have been busy the last couple of days, will make it tomorrow.

Smanar, 2021-01-02:
Nice thx, I will probable on PR on your for schedule.
The next stable is the 8th, and as this code is a corrective, it will be probably in this version.

Smanar, 2021-01-03:
@KPWhiver if you can add this code too to the PR https://github.com/Smanar/deconz-rest-plugin/commit/d051020cf05e2ef8df123948374b4b2b02deb2e7 pls ?

It will prevent 00 value in schedule, and start the code for the daily schedule.

Smanar, 2021-01-03:
And this code too https://github.com/Smanar/deconz-rest-plugin/commit/7a820b8cc1b1c0a5d583713aa3c4a6c16b3ba96a

For this issue https://github.com/dresden-elektronik/deconz-rest-plugin/pull/3730#issuecomment-753523109

KPWhiver, 2021-01-03:
The entire tuyaspecial branch or just those two commits?

Smanar, 2021-01-03:
IDK, I don't see your PR on your account, but the last working code + thoses 2 add.

KPWhiver, 2021-01-03:
I'm adding just those two, just removing some compile errors then I\ll create the PR

Smanar, 2021-01-03:
Nice thx ^^.
Just 2 comments.

>const QDateTime epoch = QDateTime(QDate(2000, 1, 1), QTime(0, 0), Qt::UTC);

This line is better in time.cpp than tuya.cpp

And the more important part

```
    else if (//isXmasLightStrip(lightNode) &&
             zclFrame.commandId() == 0x01 &&
             !(zclFrame.frameControl() & deCONZ::ZclFCDisableDefaultResponse))
    {
        sendZclDefaultResponse(ind, zclFrame, deCONZ::ZclSuccessStatus);
    }
```

I think it s cleaner to put it inside the if ( (zclFrame.commandId() == 0x01) || (zclFrame.commandId() == 0x02) ) loop

```
    else if ( (zclFrame.commandId() == 0x01) || (zclFrame.commandId() == 0x02) )
    {
        // 0x01 Used to inform of changes in its state.
        // 0x02 Send after receiving a 0x00 command.
        
        // Send defaut response
        if ((zclFrame.commandId() == 0x01) && !(zclFrame.frameControl() & deCONZ::ZclFCDisableDefaultResponse))
        {
            sendZclDefaultResponse(ind, zclFrame, deCONZ::ZclSuccessStatus);
        }
```

Smanar, 2021-01-03:
Nice, better than my code.

KPWhiver, 2021-01-26:
Still waiting on the relevant PR to be merged

PR #4087 (2021-02-08): Add support for HiHome radiator thermostat
This PR adds support for the HiHome (closes #3922). It also includes some fixes for the Saswell SEA801 radiator thermostat.

Manufacturer Name | Model Id | Manufacturer | Product
---------------------------|-------------|--------------------|-----------------
_TYST11_zuhszj9s | uhszj9s | HiHome |  WZB-TRVL Thermostat
_TYST11_KGbxAXL2 | GbxAXL2 | Saswell | SEA801-ZIGBEE TRV
closed 2021-02-08 by manup