dresden-elektronik/deconz-rest-plugin #3926: edp re:dy plug

vpedromarques, 2020-12-15:
## Device

- Product name: edp re:dy plug.
- Manufacturer: EDP-WITHUS.
- Model identifier: ZM-SmartPlug-1.0.0.
- Device type : Please remove all unrelated device types. 
  - Other: Smart Plug with Energy Metering
[image]
[image]
[image]

 

## Screenshots

Mimiix, 2020-12-15:
Missing screenshots.

vpedromarques, 2020-12-15:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

SwoopX, 2020-12-17:
Can you please apply a load where you can anticipate how much it will draw and see what value attribute 0x0400 (Instantaneous demand) of the simple metering cluster is shown? I need to get the measurement unit (although I think it's in W already).

Also, further above in this cluster, attribute 0x0000 (current summation delivered) there's the usual multipliers/Divisors missing would it be more like 3.7 kWh or 0.37 kWh since you've taken it into operation?

vpedromarques, 2020-12-17:
> Can you please apply a load where you can anticipate how much it will draw and see what value attribute 0x0400 (Instantaneous demand) of the simple metering cluster is shown? I need to get the measurement unit (although I think it's in W already).

Connected a Hue lamp with a consumption of 8w. The reported "Instantaneous demand)" on 0x0400 is 8.

> Also, further above in this cluster, attribute 0x0000 (current summation delivered) there's the usual multipliers/Divisors missing would it be more like 3.7 kWh or 0.37 kWh since you've taken it into operation?

Made a reset to the counter of that attribute 0x0000 to 0 and with the same hue lamp after 10 minutes it turned to 1 so I would guess it's more like 0.01 kWh...

closed 2020-12-23 by manup