dresden-elektronik/deconz-rest-plugin #3927: Danfoss Ally setting Heatsetpoint accepted but device does not change valve

easybeat, 2020-12-15:
## Describe the bug

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: NUC
 - Running method: Ubuntu Virtual Machine in Proxmox
 - Firmware version: (26680700)
 - deCONZ version: (2.07.00)
 - Device: ConBee II 
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? If so: Which? yes. Aeotec Zwave Stick

## deCONZ Logs

## Additional context

easybeat, 2020-12-16:
my test was with a full value. I did set to 26 and then turned it manually down by 0.5 to 25.5

SwoopX, 2020-12-16:
Although you might not want to hear that, there's nothing we can do about it since we can't control how the device reacts on the valve by setting the heatsetpoint. This is more a case for Danfoss itself

But my TRV also doesn't jump to fully open if I set the highest temp, it takes a while...

easybeat, 2020-12-17:
@SwoopX After your input I did a test this morning and you are right. It opens but very slowly:
[image]

Like that it will take about 2 hours until the valve is open 100%!

I expect there are maybe different modes?

And yes this is not a bug from deconz.

Thanks for help.

Beat

randomuser389, 2020-12-20:
Hi 

When the device receive scheduled change from the gateway it will react much slower than when change temperature on the device.

The reason is that turn temperature on the device is seen as "user interaction".

If you wait 30 minutes to 60 minutes the devices will slowly change the motor (you can hardly hear it) 

The device has a PID regulation that give good comfort and good to keep a steady temperature without overshoot!

there as a manufactuer specific command that can be send so the device react the same like it was a "user interaction" but this is not recommended as it may cause overshoot of the temperature,

easybeat, 2020-12-22:
> Hi
>
> When the device receive scheduled change from the gateway it will react much slower than when change temperature on the device.
>
> The reason is that turn temperature on the device is seen as [...]

Hi

thank you for your information. I have TRV's from other manufacturers and they all handle it different. Whenever I send the temperature to them the valve reacts immediately. Even the Danfoss Z-Wave devices did so. So I am a bit suprised.

Anyway as my goal to operate a towel radiator in the bath this behaviour is not helpful as I want to turn this on immediately on maximum heat.

Could you please explain to me what specific command I have to use to get the TRV handle it like user interaction? Like that it would be my choice how I want to operate the TRV.

Thank you very much for your help.

Kind regards
Beat

SwoopX, 2021-01-03:
I now know how this is working and how it can be changed. The current mode of operation is a viable option and intended as such. On the other hand, I have to admit that it feels a bit strange that this works as designed any may take ages to heat up a room or at least raise temperature.

I'd assume it's the more gentle way not to overheat and reach the desired temperature on point, therby saving some energy...

Not sure yet how to accomodate that in the current code (and especially the differentiation between the 2 modes).