dresden-elektronik/deconz-rest-plugin #3929: Possible regression: Aqara sensors are disconnected after some time when connected through router

cagnulein, 2020-12-16:
## Describe the bug

My temperature sensor works fine for several weeks, but since some days it continuously disappears / disconnects.

(Last time where repairing worked there were a red line before...)

## Steps to reproduce the behavior

it happens when the aqara sensors are not directly connected to the router

## Expected behavior

aqara sensors must work with or without connecting directly to the router

## Screenshots

[image]

## Environment

Version: 2.07.01 / 8.12.2020
Firmware: 26660700

## deCONZ Logs

## Additional context

Reopening #1428
Originally posted https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1488#issuecomment-745467485

Mimiix, 2020-12-16:
@cagnulein Please fill the correct template.

Mimiix, 2020-12-16:
Closing this. Duped of #3930.

closed 2020-12-16 by Mimiix