dresden-elektronik/deconz-rest-plugin #3930: Possible regression: Aqara sensors are disconnected after some time when connected through router

cagnulein, 2020-12-16:
## Describe the bug

My temperature sensor works fine for several weeks, but since some days it continuously disappears / disconnects.

(Last time where repairing worked there were a red line before...)

## Steps to reproduce the behavior

it happens when the aqara sensors are not directly connected to the router

## Expected behavior

aqara sensors must work with or without connecting directly to the router

## Screenshots

[image]

## Environment

 - Host system: PC
 - Running method: Virtual Machine
 - Firmware version: 26660700
 - deCONZ version: 2.07.01 / 8.12.2020
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? no
 
## deCONZ Logs

## Additional context

Reopening #1488 
Originally posted https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1488#issuecomment-745467485

@Mimiix is this ok?

rexbron, 2020-12-16:
Dupe of #3929

cagnulein, 2020-12-16:
@rexbron yes, here i added some env info that @Mimiix requires

Mimiix, 2020-12-16:
You couldve edited the other issue @cagnulein  Closed it now. Continue in this one.

Smanar, 2020-12-16:
Hello what is the router brand ?
Remember Xiaomi have lot of problem with other brand https://community.hubitat.com/t/xiaomi-aqara-devices-pairing-keeping-them-connected/623

cagnulein, 2020-12-16:
@Smanar it's a conbe II.
If the device is connect directly to the router everything works great. 
The issue happens only when the device using mesh from other devices

Mimiix, 2020-12-16:
The Conbee is a coordinator, not a router. The Yellow devices are routers.

cagnulein, 2020-12-16:
sorry, the router is a LEDVANCE B40 DIM Z3

Mimiix, 2020-12-16:
According to the link @Smanar gave:

OSRAM/LIGHTIFY
Most OSRAM ZigBee products appear to be incompatible, based on various reports (see two possible exceptions in the "reported to work" section below.)

Those that seem to work:
SRAM/Sylvania LIGHTIFY
A19 RGBW color recessed light (needs confirmation)
RT Tunable White recessed light (needs confirmation)
OSRAM Smart+ Outlet - model AB3257001NJ (UK & EU only)
Sylvania SMART+ Plug 72922 (mostly working, more info here & here, but the 72922-A variant does not work)

So i think the device firmware is the issue (the Ledvance).

Smanar, 2020-12-16:
Yep, possible, but it s a bulb ? generaly bulbs are wotking fine ...
You have not another routers (from another brand) to make tests ?

cagnulein, 2020-12-16:
> Yep, possible, but it s a bulb ? generaly bulbs are wotking fine ...
> You have not another routers (from another brand) to make tests ?

yes it's a buble, i have a Heiman TS0121 but how can i bind the aqara to avoid to talking with the bulb?

Smanar, 2020-12-16:
So hard ^^, there is so much users trying to force xiaomi device to use a router instead of direct connexion.

Just for test, just unplug the bulb, plug the Heiman, and make Xiaomi inclusion (or just press shortly the button, it seem this provoke a new parent search for the device)

cagnulein, 2020-12-16:
Thanks i will test it the next days! :)

cagnulein, 2020-12-17:
Turned off the bulb, short pressed the Xiaomi button but the sensor still unavailable.
Tonight i will try to repaired it with the bulb off.

cagnulein, 2020-12-18:
hi @Smanar seems that you were right! turning off the bulb and repairing the aqara to my Heiman solves the issue! Now i turned back on the bulb and aqara stills running! Wonderful! I'm closing this one, hoping that someone will find it useful ;)

closed 2020-12-18 by cagnulein

Smanar, 2020-12-18:
But if it work now, it confirm Xioami have issue too with ledvance ....

cagnulein, 2020-12-18:
Yes i can say it without any doubts.
And i definitely say that no more Xiaomi zigbee devices will enter in my
house. I just bought 10 of these :(

SwoopX, 2020-12-18:
And another one for the category: You get what you paid for...

No, but seriously, we received word from Xiaomi that they initially were never meant to be available outside China 🤷‍♂️