dresden-elektronik/deconz-rest-plugin #3931: Unable to pair Sonoff SNZB-01

MadMonkey87, 2020-12-16:
## Describe the bug
I've tried to add a Sonoff SNZB-01 multiple times which should be supported according to https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2990

I can see the connected device in the UI, but not in the api or in the phoscon app.

## Steps to reproduce the behavior
1) go into the phoscon app
2) go to switches, click add and then other
3) press the reset button on the device until the led flashes, click often to keep the device alive

## Expected behavior
 - the device should get a name and be visible trough the api
 
## Screenshots
[image]
[image]
[image]
[image]

## Environment

 - Host system: RPI
 - Running method: Marthoc Docker container
 - Firmware version: 26350500
 - deCONZ version: (2.08.00 
 - Device: ConBee I
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? No

SwoopX, 2020-12-16:
No bug. Courtesy of Sonoff to have pretty inconsistent model identifiers, we've seen quite some hilarious variants...

MadMonkey87, 2020-12-16:
Ah I see. For my [Homey integration](https://github.com/madmonkey87/HomeyCONZ) I've already seen that there are a lot of different identifiers for these devices

closed 2021-01-12 by manup