dresden-elektronik/deconz-rest-plugin #3936: Osram Lightify FlexRGBW (Model 73661)

RichJeanes, 2020-12-16:
## Device

I have two light strips that were sold as the same product, but appear slightly differently in deCONZ.
- Product name: SYLVANIA Smart+ Adjustable White and Full Color Flex
- Manufacturer: LEDVANCE, OSRAM
- Model identifier: [blank], Osram Lightify FlexRGBW
- Device type :
  - Light

## Screenshots
Device 1:
[image]
[image]
[image]

Device 2:
[image]
[image]
[image]

Currently, the devices appear in Phoscon (via Home Assistant plug-in), but only have color and brightness control. The white LEDs are never used and the Color Temperature controls are disabled. However, in deCONZ, I can set them to Color Temperature mode and the white LEDs do turn on, but I cannot change the color temperature they are set to. If I set them to a color in Phoscon, the Color Mode attribute changes to "Current x and current y". Issuing a Move Color Temperature command will report success, but the light strip will remain at its set color until being turned off (it switches to the white color, then turns off). The white color is retained when powered on again. Even while in Color Temperature mode, I cannot adjust the temperature through deCONZ, with the strip al;ways reporting a temperature of 370.

vperez, 2020-12-18:
I purchased the same light strip recently, and I'm seeing the same behavior under Phoscon and deCONZ (Docker image). The Phoscon app has the color temperature controls disabled, but using deCONZ I was able to adjust the color temperature manually.

### Screenshots from deCONZ

[image]
[image]
[image]
[image]

RichJeanes, 2020-12-18:
@vperez Your model seems different, being the outdoor model. Mine are not the Outdoor model. My Move Color Temperature command is also different, having Rate, Min, and Max. Though I'm sure it's possible that we're both facing the same issue in the back-end.
[image]

I don't know if this is helpful, but my device does seem to be reporting its Color Temperature capability under ZLL Extensions:
[image]

vperez, 2020-12-21:
@RichJeanes you're right I did not notice that yours are not the outdoor model. I think both devices do have the ability to support changing white color temperature, but they are just not fully recognized at the moment by the Phoscon app.

RichJeanes, 2021-01-25:
This issue is not stale.

closed 2021-03-05 by github-actions[bot]