dresden-elektronik/deconz-rest-plugin #3938: Got Switches as Groups from API

talun2075, 2020-12-16:
## Describe the bug
In My Hue Essentials and in all directly API Calls I got all my Switches in Group calls.
If I delete the groups after reboot of my conbee II there are back.

## Steps to reproduce the behavior

Call: http://IP:PORT/api/APIKEY/groups

if you got your switches you have the same problem.

## Expected behavior
Switches are not showed in groups

## Screenshots
[groups.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5705491/groups.txt)

## Environment

 - Host system: Windows 2019
 - Running method: Windows
 - Firmware version: (2668070)
 - deCONZ version: (2.07.01)
 - Device: (ConBee II)
 - Do you use an USB extension cable: (yes)
 - Is there any other USB or serial devices connected to the host system? If so: yes Mouse & Keyboard

## deCONZ Logs
I got every minutes in Error:
22:09:21:431 CTRL got nwk update id 0

In Info: 22:09:21:431 CTRL got nwk update id 0
22:09:44:989 verified group capacity: 6 and group count: 2 of LightNode 0x84182600000d715e
22:09:44:989 0x84182600000d715e found group 0xFFF0
22:09:44:989 0x84182600000d715e found group 0x0006
22:09:45:805 verified group capacity: 6 and group count: 2 of LightNode 0x7cb03eaa00ae8aa8
22:09:45:805 0x7cb03eaa00ae8aa8 found group 0xFFF0
22:09:45:805 0x7cb03eaa00ae8aa8 found group 0x0005
22:09:46:477 verified group capacity: 23 and group count: 2 of LightNode 0x001788010498c411
22:09:46:477 0x001788010498c411 found group 0xFFF0
22:09:46:477 0x001788010498c411 found group 0x0008
22:09:47:288 verified group capacity: 255 and group count: 2 of LightNode 0x000b57fffe9b5bd5
22:09:47:288 0x000b57fffe9b5bd5 found group 0xFFF0
22:09:47:288 0x000b57fffe9b5bd5 found group 0x0008
22:09:47:431 ZCL attribute report 0x0017880108632462 for cluster: 0x0402, ep: 0x02, frame control: 0x08, mfcode: 0x0000 
22:09:47:967 verified group capacity: 23 and group count: 2 of LightNode 0x00178801038cf803
22:09:47:967 0x00178801038cf803 found group 0xFFF0
22:09:47:967 0x00178801038cf803 found group 0x0009
22:09:48:739 verified group capacity: 23 and group count: 2 of LightNode 0x0017880103874487
22:09:48:739 0x0017880103874487 found group 0xFFF0
22:09:48:739 0x0017880103874487 found group 0x0009
22:09:49:496 verified group capacity: 23 and group count: 2 of LightNode 0x0017880103fc61be
22:09:49:496 0x0017880103fc61be found group 0xFFF0
22:09:49:496 0x0017880103fc61be found group 0x0009
22:09:50:233 verified group capacity: 23 and group count: 2 of LightNode 0x0017880103819fe3
22:09:50:233 0x0017880103819fe3 found group 0xFFF0
22:09:50:233 0x0017880103819fe3 found group 0x0009
22:09:50:993 verified group capacity: 23 and group count: 2 of LightNode 0x00178801038cadb3
22:09:50:993 0x00178801038cadb3 found group 0xFFF0
22:09:50:993 0x00178801038cadb3 found group 0x0009
22:09:51:740 verified group capacity: 23 and group count: 2 of LightNode 0x0017880103b1f306
22:09:51:740 0x0017880103b1f306 found group 0xFFF0
22:09:51:740 0x0017880103b1f306 found group 0x000A
22:09:52:508 verified group capacity: 23 and group count: 2 of LightNode 0x0017880104561341
22:09:52:508 0x0017880104561341 found group 0xFFF0
22:09:52:508 0x0017880104561341 found group 0x000A
22:09:53:267 verified group capacity: 23 and group count: 2 of LightNode 0x0017880104d2fddb
22:09:53:267 0x0017880104d2fddb found group 0xFFF0
22:09:53:267 0x0017880104d2fddb found group 0x000A
22:09:53:989 verified group capacity: 23 and group count: 2 of LightNode 0x0017880106128ff5
22:09:53:989 0x0017880106128ff5 found group 0xFFF0
22:09:53:989 0x0017880106128ff5 found group 0x000A
22:09:54:762 verified group capacity: 23 and group count: 2 of LightNode 0x0017880103e3acfa
22:09:54:762 0x0017880103e3acfa found group 0xFFF0
22:09:54:762 0x0017880103e3acfa found group 0x0006
22:09:55:535 verified group capacity: 23 and group count: 2 of LightNode 0x0017880104d2fda4
22:09:55:535 0x0017880104d2fda4 found group 0xFFF0
22:09:55:535 0x0017880104d2fda4 found group 0x0015
22:09:56:288 verified group capacity: 23 and group count: 2 of LightNode 0x00178801044b951c
22:09:56:288 0x00178801044b951c found group 0xFFF0
22:09:56:288 0x00178801044b951c found group 0x0015
22:09:57:014 verified group capacity: 23 and group count: 2 of LightNode 0x0017880106255ae7
22:09:57:014 0x0017880106255ae7 found group 0xFFF0
22:09:57:014 0x0017880106255ae7 found group 0x0015
22:09:57:768 verified group capacity: 23 and group count: 2 of LightNode 0x00178801061299b3
22:09:57:768 0x00178801061299b3 found group 0xFFF0
22:09:57:768 0x00178801061299b3 found group 0x0015
22:09:58:520 verified group capacity: 23 and group count: 2 of LightNode 0x0017880106262936
22:09:58:520 0x0017880106262936 found group 0xFFF0
22:09:58:520 0x0017880106262936 found group 0x0015
22:09:59:323 verified group capacity: 7 and group count: 1 of LightNode 0x7cb03eaa00ad5d4a
22:09:59:323 0x7cb03eaa00ad5d4a found group 0xFFF0
22:10:00:214 verified group capacity: 255 and group count: 2 of LightNode 0x588e81fffe71431f
22:10:00:214 0x588e81fffe71431f found group 0xFFF0
22:10:00:214 0x588e81fffe71431f found group 0x0016
22:10:00:815 verified group capacity: 23 and group count: 2 of LightNode 0x0017880100c42995
22:10:00:815 0x0017880100c42995 found group 0xFFF0
22:10:00:815 0x0017880100c42995 found group 0x0006

## Additional context
If you look to the groups.txt you see there is no hidden prop on my result.
if you look to the api reverenz: https://dresden-elektronik.github.io/deconz-rest-doc/endpoints/groups/#get-group-attributes
they should have a hidden prop.
It´s not possible to create it.

Looks that there is something really wrong on my system?

Thomas-Vos, 2020-12-16:
These groups are automatically created for the switches. Hue Essentials shows them in the list as they may be useful. If you would like to hide them, please open a group in Hue Essentials, tap the three dots, and tap Hide. Alternatively you can hide groups in the app settings.

I am not sure why the hidden attribute is not there, someone else would need to answer that. Hue Essentials does not use this attribute by the way.

talun2075, 2020-12-17:
ok thanks for the answer. then I know its partial normal.

Mimiix, 2020-12-17:
Case closed?

talun2075, 2020-12-19:
hm...the missing hidden Prop is a new case or?

closed 2021-03-05 by github-actions[bot]