dresden-elektronik/deconz-rest-plugin #3939: Tuya Zigbee Curtain module QS-Zigbee-C01

adbig, 2020-12-16:
Hello,

I bought a zigbee module to automate my roller shutter garage.
Unfortunally, deconz mistakenly identified it as a TS130F device ( https://es.aliexpress.com/item/4001190275372.html ).
In reallity this is not a switch , it's this :

[image]

Bought here : https://fr.aliexpress.com/item/1005001530266723.html

I successfully manage to go up and down with the detected TS130F configuration but the problem I encounter is that each time I execute the command to go UP or down, the opening/closing time is only 10 secondes. My roller shutter  takes approximatively 30 secondes to Open/close so I have to execute 3 times the opening/closing command.

## Device

- Product name:  Zigbee curtain module
- Manufacturer: _TZ3000_vd43bbfq
- Model identifier: TS130F
- Device type : 
  - Other: Curtain / Roller shutter module

## Screenshots

[image]

[image]

thanks !

Adrien

Mimiix, 2020-12-16:
Hi!

Can you provide all screenshots of the clusters?

adbig, 2020-12-17:
Hi, 

[image]

[image]

[image]
[image]

Smanar, 2020-12-17:
For me it s the normal working mode, you probalby need to calibrate the device, you have the command on the cluster 0x0102, and the instructions on your manual.

adbig, 2020-12-17:
Hi @Smanar  and thank you for your response.

I discovered this thread that is very close to my problem. https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3134

After setting in the XML file the calibration access  to RW I was finally able to calibrate my module !!!!!

```
<attribute-set id="0xf001" description="Tuya Window Covering Setting">
                        <attribute id="0xf001" name="Calibration" type="enum8" default="0" required="m" access="rw">
```
I don't know why this setting is in ReadOnly Mode because it is needed for any new/unconfigured module.

Normally, I use Jeedom to control my domotic, and frankly, it's not very easy to setup this essential feature  ... Do you know if it would be possible to have this kind of feature (calibration) available with the Jeedom DeCONZ plugin ? (I don't know who are the developpers of this plugin).

Thank you for your support !

Adrien

Smanar, 2020-12-18:
They are already in RW mode on the last version, wich one version are using ?

And I m agree with you, so there is an hidden command (not documented yet) for that

`PUT, PATCH /api/<apikey>/lights/<id> with {"calibration":true}`

adbig, 2020-12-18:
Hello,

I'm using 2.07.00-raspbian-buster-beta package but I just see there is a new revision 2.08.00-raspbian-buster-beta.

Adrien

Smanar, 2020-12-18:
Ha yes, the change as done 19 days ago, so it 's recent.

adbig, 2020-12-19:
SO, it'sgood for me. I close this issue. thank you again for your help !

closed 2020-12-19 by adbig

Mimiix, 2020-12-21:
@adbig Great:)

Sjonnie2018, 2021-01-08:
Hi All,

I have this device (TS130F) and have read through all the mentioned threads to find out how I can change the close and open times to more then 10 seconds. Frankly, I am stumped with all the technical info and am looking for some guidance so I can get the module to close/open my roller shutters for 40 seconds.

Could somebody please explain to me in simple words how I can get the device to extend the power when opening or closing my shutters?

Working with Deconz (-gui) version 2.07.01

Thanks a million in advance!

Greetz,

Sjonnie

Smanar, 2021-01-08:
Do you still have the manual gived with the device ?
I haven't it to make tries, but I think you can do exactly like in the manual except you use deconz GUI to set the "set calibration mode" instead of the app.

Sjonnie2018, 2021-01-08:
Hi Smanar,

Thanks for your reply and support!

I have the manual but it does not give any information about changing the timings for the module. I can not find "set calibration mode" in deconz-gui. Maybe I overlooked. If so, please tell me where I can find it.

Greetz,

Sjonnie

adbig, 2021-01-08:
Hi,

First, you must have the latest version of Deconz because in the previous versions I had to modify the "general.xml" file to allow the switch to calibration mode.
This section was in read only mode : 
`                        <attribute id="0xf001" name="Calibration" type="enum8" default="0" required="m" access="rw">
`

At the the bottom of this screen you'll see "Calibration : end"

[image]

- First ensure you curtain is fully closed.
- Set the calibration  to "start" and click Write. Now you are in Calibration mode.
- Above, click on the Exec button in the function  Up / Open. In the calibration mode the 30 secondes duration is disable.
- Once your curtain is fully open click on the "Exec" button in the function "Stop"
- Click on the Exec button in the function Down / Close
- Once your curtain is fully closed click on the "Exec" button in the function "Stop"
- Set the calibration  to "end" and click Write

Enjoy, your module is calibrated !

(sorry for appromative my english ...)

Adrien

Sjonnie2018, 2021-01-09:
@adbig : You Sir, are now officially declared Sjonnie's Hero of The Day 2021! :)

Very clear explanation and instructions. Calibrating the module was very easy following your guidance.

Now I can safely order these for my other roller shutters.

One note for future generations: after the last "write" action the device shows "start" in the calibration field. You can remedie that by doing a read of the attributes.

Again Sir: Kudos!

adbig, 2021-01-09:
Glad it helped you !

incognitomailincognitomail, 2021-08-09:
@adbig can you explain how to connect to curtain module, I didn't find calibration dialogs in DeconZ plugin
And how I can edit general.xml file

Sjonnie2018, 2021-08-09:
You will have to open deconz-gui (in your case: connect to the device in the last screenshot). After that you will find the node in the gui. Select the right most "radio button" on the top right of the node and check the cluster info.

incognitomailincognitomail, 2021-08-09:
@Sjonnie2018 What is Conbee (I have ZigBee cc2531 usb dongle).
Can you please send me toturial

Sjonnie2018, 2021-08-09:
Hi, I figured from your screenshot that you had a Conbee (2) stick from Dresden Electronics. I haven't got a clue how to connect Deconz(-GUI) to your cc2531 stick. Sorry! I don't quite understand how and why you would use Deconz if you do not have a device from Dresden Electronics. Maybe I am missing something or do not fully understand you situation. If so, please enlighten me.

The tutorial I used is the one adlbg wrote down in post 8 of this thread.

Mimiix, 2021-08-09:
@incognitomailincognitomail Deconz only work with Conbee / Raspbee devices created by the vendor.

incognitomailincognitomail, 2021-08-09:
@Mimiix thanks.
I can use Conbee stick to calibrate the curtain module only, and after the calibration to use my ZigBee cc2531 usb dongle)

Sjonnie2018, 2021-08-09:
I am a bit stumped here. You have a Conbee II stick to configure the roller shutter module and after that you want to use your cc2531 to control it? If that's the case the information in this particular thread will help you achieve your goal. Connect your Conbee II stick to the device running Deconz (using a USB extension cord) and follow the steps in this thread. Good luck!

Mimiix, 2021-08-09:
> I am a bit stumped here. You have a Conbee II stick to configure the roller shutter module and after that you want to use your cc2531 to control it? If that's the case the information in this [...]

I am wondering if that would work. To pair it to the cc stick he would need to reset the device. Wondering if the device keeps the settings.

Sjonnie2018, 2021-08-09:
I honestly don't have a clue! I wouldn't even bother if I was him and just buy a Conbee II. Not overly expensive and as a bonus he gets support from this community :)

Smanar, 2021-08-09:
And for information, if It work, it s possible use the API, if you don't have the GUI
`PUT, PATCH /api/<apikey>/lights/<id> with {"calibration":true}`

incognitomailincognitomail, 2021-08-10:
ok thanks for all.
Can I calibrate the roller shutter in other way? (without Conbee)

adbig, 2021-08-10:
With the tuya smart life app .... If you Read the doc provided with your module it's explained...

incognitomailincognitomail, 2021-08-10:
@adbig can you please send link with the details
Additional Q,
With smart life I can use my ZigBee cc2531 usb dongle?

Mimiix, 2021-08-10:
@incognitomailincognitomail This repository is for deCONZ only. This issue is about the specific device for deCONZ. Keep it on topic here. With teh CC Stick, your kind of on your own.

@adbig Does the device work properly with deCONZ?

adbig, 2021-08-10:
Yes the device works very Fine with decone since monthes.

incognitomailincognitomail, 2021-08-24:
Hi 

I don't see all module only "hone Automation configuration tool"
@adbig Please advise 

[image]

Smanar, 2021-08-24:
Your devices are not included at all.
And as your network is new, can be a deconz issue, start from the start, check in phoscon if there is a firmware number, test with USB extension, check logs, checks config, .....

2chilled, 2021-09-20:
Hi, using REST API I can activate calibration mode using:
`curl "http://openhabian:80/api/abc/lights/22" -d '{"calibration":"Start"}' -XPUT`

But how can I deactivate it? I've tried setting calibration to "End", but that does not have any effect.

Smanar, 2021-09-20:
"calibration" work with true and false.

2chilled, 2021-09-21:
@Smanar I tried with true and false, but it seems to be still in calibration mode after putting false. This is the sequence I tried:
```
curl "http://openhabian:80/api/abc/lights/22" -d '{"calibration":true}' -XPUT
curl "http://openhabian:80/api/abc/lights/22/state" -d '{"lift":0}' -XPUT
curl "http://openhabian:80/api/abc/lights/22/state" -d '{"lift":"stop"}' -XPUT
curl "http://openhabian:80/api/abc/lights/22/state" -d '{"lift":100}' -XPUT
curl "http://openhabian:80/api/abc/lights/22/state" -d '{"lift":"stop"}' -XPUT
curl "http://openhabian:80/api/abc/lights/22" -d '{"calibration":false}' -XPUT
```
The stop commands were issued when the curtain was fully open or closed. After that sequence when setting "lift" to 50, the curtain always goes to end position. What am I missing?

AdrienBigot, 2021-09-21:
@Smanar are you sure it is start and stop  ?
In the XML file I read : 
```
     </attribute-set>
                <attribute-set id="0xf001" description="Tuya Window Covering Setting">
                        <attribute id="0xf001" name="Calibration" type="enum8" default="0" required="m" access="rw">
                                <value name="Start" value="0"></value>
                                <value name="End" value="1"></value>
                        </attribute>
                        <attribute id="0xf002" name="Motor Reversal" type="enum8" default="0" required="m" access="rw">
                                <value name="Off" value="0"></value>
                                <value name="On" value="1"></value>
                        </attribute>
        </attribute-set>

@2chilled why don't you try with Deconz GUI ?

mahe-ymc, 2021-09-21:
@AdrienBigot I'm running on a headless raspi without any X libs installed. Do you think I'd have more luck running the GUI version and connecting via vnc?

2chilled, 2021-09-21:
@AdrienBigot I'm running Conbee2 on a headless raspi without any X libs installed. If calibration can't be done with REST API I'll bite the bullet and change that. I think I need to connect via vnc then?

Smanar, 2021-09-21:
Sure, from the code

```
        qint64 value = 0x00;
        if (map["calibration"].toBool())
        {
            value = 0x01;
        }
```

So the code send the value 0 or 1 according to True or False (true = 0x01 = End and False = 0x00 = Start, and BTW they are reversed .....)
You have not an error message as return when you make the request ?

2chilled, 2021-09-21:
@Smanar No, I'm getting 200 response code. I guess your `toBool` ist not very strict with it's inputs ;)

Ok, does that mean I have to activate calibration with `false` and deactivate it with `true`?

2chilled, 2021-09-21:
Just tried with switching false to true and vice versa. No change. Device stays in calibration mode as it does not stop moving. Power off and power on leads to the initial state, that is 10 seconds duration until stop. Do I really need to use the GUI for calibrating?

Smanar, 2021-09-21:
You have a 200 code, but there is a message too ?
IDK, I have checked the code again, and all seem fine, you haven't access to log ?

AdrienBigot, 2021-09-21:
@2chilled I have the same setup as you with a headless Debian buster on a raspberry Pi2 and I use deCONZ gui over ssh.

On the machine running deCONZ : 
- Ensure you have X11forwarding to yes in your /etc/ssh/sshd_config 
- Stop deCONZ deconz service first :  systemctl stop deconz.service

On your client : 
ssh -X user@your_server_address.
Once you're logged, just launch deCONZ binary (/usr/bin/deCONZ for me). It opens a new windows with your deCONZ GUI :-D   (don't try with root user. On my side I must ssh with my "pi" user (who usually also run deconz.service).

Adrien

2chilled, 2021-09-21:
@AdrienBigot Thank you very much for that howto! Just realized that I already have all tools in front of my eyes, but I didn't see them. X11Forwarding, that's the key :) My rollerblind works now and calibrating is easy peasy using the GUI. Thanks again, appreciate it :)

AdrienBigot, 2021-09-21:
Happy it works for you !

But it's strange that we cannot do the calibration with API. I admit I never try do do it in this way and don't want to reset my working config :-D
I know somebody that will buy this device. I'll try to test the calibration with API on my side and will give a feedback here.

Adrien.

2chilled, 2021-09-21:
I still have a bunch of these devices to install, I'll try to post API Info requested by @Smanar this evening.

Smanar, 2021-09-21:
But I realy don't understand, all seem fine, can you take a look in log in same time, you are making the request (with aps/info/info_l2)

The synthax is good because it work to enable it, but it s like `if (map["calibration"].toBool())` is never trigger, so the value is always 0x00

2chilled, 2021-09-21:
I can have a look tomorrow, but where is that log located? I see nothing in journal of deconz systemd unit.

Smanar, 2021-09-21:
If you have headless mode, you have only 1 solution, use command line https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/deCONZ-debug-flags

2chilled, 2021-10-04:
I have a new issue with this device to report. Sometimes, I don't have a pattern yet, I cannot write the state of the device. I provide a communication flow:
```
curl "http://192.168.0.207:80/api/2D58761B15/lights/51/state" -XPUT -d'{"lift": 10}'
[{"success":{"/lights/51/state/lift":10}}]

curl "http://192.168.0.207:80/api/2D58761B15/lights/51" | json_pp
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   364  100   364    0     0  72800      0 --:--:-- --:--:-- --:--:-- 72800
{
   "etag" : "95ff636c780d8c1d6751efc547f0b537",
   "hascolor" : false,
   "lastannounced" : null,
   "lastseen" : "2021-10-04T07:38Z",
   "manufacturername" : "_TZ3000_vd43bbfq",
   "modelid" : "TS130F",
   "name" : "Rolladen Wohnzimmer Süd OG 1",
   "state" : {
      "bri" : 0,
      "lift" : 0,
      "on" : false,
      "open" : true,
      "reachable" : true
   },
   "swversion" : null,
   "type" : "Window covering device",
   "uniqueid" : "cc:86:ec:ff:fe:be:71:0a-01"
}
```

When I hit the manual switch and stop it again, the device is writeable again through Zigbee. Please tell me what info you need in addition to tackle this. Thank you!

Smanar, 2021-10-04:
I m not sure to understand the issue.
You are making the request `curl "http://192.168.0.207:80/api/2D58761B15/lights/51/state" -XPUT -d'{"lift": 10}'`
You have the return "success" but the device don't move and the json stay at lift=0 ?

2chilled, 2021-10-05:
@Smanar Yes, you understood correctly. This problem occurs with 2 devices installed on top of each other. I also tested them using deconz GUI and I'm getting timeouts there. What I also observed in the GUI is that both devices have bad connection and somehow they're not connecting to the nearest Zigbee routers around them. For example I have a plug 1m away from them which is very well connected but they just don't connect to it. I guess I will pull them out the wall and check if it helps or really replace them in the end.

For the deconz-rest-plugin the isolated issue I'm observing that it delivers false positives. It shouldn't return success if it wasn't successful. But it does in my case.

2chilled, 2021-10-05:
@Smanar Update: I removed both devices from deconz and added them again. After that everything works as it should :) But the last point from above remains in general.

Smanar, 2021-10-05:
Just for information about the deconz working mode. When you make a request, you will have a "success" if deconz can send the request, it don't mean the device will use it.

- You make the http request
- Deconz send the zigbee request
- You have the "success" message
- The device execute the request
- The device send report
- deconz Update the value

So bad connexion can explain your issue, but the question is why you have bad connexion.

2chilled, 2021-10-05:
Ok, then deconz behaves like expected. Yeah, that one still remains. At least I know a workaround for next time ;)

69franky, 2021-10-11:
> @69franky Your method to terminate calibration mode worked for me for some of my qs-zigbee-c01 devices but not for all of them. Maybe it also depends on actual manufacturer. I have both "_TZ3000_vd43bbfq" and "_TZ3000_fccpjz5z" installed.

I am evaluating only one device, which is from  "_TZ3000_fccpjz5z", and I could calibrate it. All this does not sound like a sustainable solution. Keep you posted.

2chilled, 2021-10-11:
@Smanar My repeater is in place for 2 days now and the problem still persists. It seems like these devices have problems with connecting with my other routers. There are 3 devices suffering from the issue described above and all of them have no connection to the Conbee2 coordinator. Is there anything I can do about it? Do you know of an alternative rollershutter device which also supports the lift API?

Smanar, 2021-10-31:
@qba667 you have tried the code modification ?

AoSpades, 2021-11-12:
Hi there! I've got a similar device which I'm trying to use over Zigbee2Tasmota and the console. 

Is there any way to enter calibration mode using a ZbSend command on these devices?

Mimiix, 2021-11-12:
@AoSpades thats kinda off-topic..

[26 more replies]