dresden-elektronik/deconz-rest-plugin #3942: Would it be possible to report group.action changes through websocket

kryzys69, 2020-12-17:
## Describe the question or issue you are having

Hi,
I see that one can get changes not only in _state_, but also _config_ sections of light or sensors. And just like in the title I would like to ask if it would be possible to receive also changes in _group.action_. Of course I mean through websocket, because you can always get this changes by polling the device.

Sorry for the second questions, the first was closed automatically :|

## Screenshots

## Environment

    Host system: PC
    Running method: Windows / Virtual Machine
    Firmware version: 26240500
    deCONZ version: 2.06.00 / 15.11.2020
    Device: ConBee I
    Do you use an USB extension cable: yes
    Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs

## Additional context

ebaauw, 2020-12-17:
Note that the `action` of a group is write-only: it reflects the target state for the lights in a group, but not the actual state.  It simply echoes the last values PUT to the `action`.  If you change individual lights through the API, or if you control (lights in the) group directly through a controller, `action` is **not** updated.  I don't think creating web socket notifications for `action` makes much sense, except when you want to monitor what values other API clients PUT to `action`.

kryzys69, 2020-12-17:
Hi, thank you for the response. I know that it is the "last group command" if we can call it this way. And my case is... you have written it in the last sentence. I would like to be notified if other client changed group state.

kryzys69, 2021-01-07:
I noticed that it is partly implemented. The "recall scene" event is already fired and it is the part of "group's action". 
Adding the possibility to receive group's action changes would greatly improve integration quality because I could get rid of polling groups.

closed 2021-03-05 by github-actions[bot]