dresden-elektronik/deconz-rest-plugin #3943: Apex Smart Plug

bomaskin, 2020-12-17:
## Device

- Product name: Apex Smart Plug
- Manufacturer: Datek
- Model identifier: PoP
- Device type : 
   - Other: Smart plug with power metering

## Screenshots

[image]

[image]

[image]

Mimiix, 2020-12-17:
@SwoopX This seems to be missing measurement features? Should i add device enhancement?

SwoopX, 2020-12-17:
(Now) all is there, you can leave the label as it is

haarvig, 2020-12-17:
I work for Datek who has developed this Smart Plug. Please let me know if you need a full Zigbee cluster dump from the device. I have it in a word document, but can send in any format.

SwoopX, 2020-12-17:
It would be really awesome if you could share it as PDF. Feel free to also reach out to me on Discord if required/desired.

haarvig, 2020-12-17:
[mains_power_outlet.pdf](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5709674/mains_power_outlet.pdf)
Here it is. We also have other new sensors coming we would like to have supported.

I´m not on Discord for now.

Mimiix, 2020-12-17:
> Here it is. We also have other new sensors coming we would like to have supported.

Feel free to reach out trough Discord :) Would love to have you around for user questions and integrations.

closed 2021-02-15 by manup