dresden-elektronik/deconz-rest-plugin #3948: Enable selection of interface for deconz network communication

SwoopX, 2020-12-17:
## Feature request type
Enhance deconz by the ability to choose the network interface via which the traffic should be send over. This should explicitly include the loopback interface.

## Description
Details will be added later.

## Considered alternatives
Blocking all traffic from unwanted interfaces via firewall, but that requires an additional component come into play, additional administrative efforts and applies additional system load (being especially notable on weak systems).

## Additional context
Selection of interfaces (all, lo, ifX, ifY, ifZ) should ideally be feasible from command line, deconz GUI and via REST API.

carver7, 2021-01-07:
I'd like to see this too, please. Having unprotected devices on a LAN is getting more risky with time, as East-West style attacks are becoming more and more common, and the concept of a network perimeter firewall sufficiently being able to guard a vulnerable "trusted network" with devices communicating to each other with no encryption is simply becoming less and less effective over time. That applies at home and in the corporate world's on-premises ￼intranets.

A rest api over https really would protect your customers a lot down the road￼, and allowing for the enabling the restrictions described by the op here￼, would at least allow some people to figure out the reverse proxy approach￼.

Thank you for consideration.

manup, 2022-10-07:
I think that works now :)

closed 2022-10-07 by manup