dresden-elektronik/deconz-rest-plugin #3951: Loosing devices after each reboot

dmatora, 2020-12-17:
## Describe the bug

On each reboot I lose all kind of devices I'm using (lamps, sensors, switches, etc). Some of them usually recover (at least when I wake them manually) some require adding them over (even if I wait for an hour or for a day)

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

- Raspberry PI 4 (2.0A power supply)
- Raspbee (latest? firmware 26390500)
- deCONZ 2.07.01 / 08.12.2020 inside Docker (HASS.io add-on version 6.6.1)
- Home Assistant 0.118.2
- No USB devices, I believe bluetooth is disabled (which was necessary to get Raspbee working) not sure how to check WiFi, I'm using Ethernet for Wan/Lan connection.
[image]

## deCONZ Logs

## Additional context

- Tried restoring backups but still had to add devices over.
- Tried saving network settings after leaving network and then joining network, like suggested [here](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Network-lost-and-configuration-restore-does-not-help), but still loosing devices after each reboot.

Mimiix, 2020-12-17:
Can you provide some logs?

dmatora, 2020-12-17:
I can see the logs over VNC but when I right click and select "copy" I can not paste them after that. It looks like VNC client doesn't provided by the addon doesn't share clipboard. How do I grab those?

Mimiix, 2020-12-17:
I have no clue. You should use VNC (as they provided that before).

dmatora, 2020-12-18:
Is there a way to connect to deconz over VNC using desktop VNC client while running it as Hass.io addon?

Mimiix, 2020-12-18:
There was,  but i am not sure. You should check the documentation of the addon.

SwoopX, 2020-12-21:
Are those only Xiaomi devices and/or other devices?

dmatora, 2021-01-25:
Yes, most of my devices are Xiaomi
I switched to another Zigbee hub and problem went away
Probably will still use deconz for Tuya TRV since I don't see some of the basic functions on another hub, will update if issue persist with Tuya TRV (currently waiting for them to be installed)

SwoopX, 2021-01-25:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Xiaomi:-Lost-devices-or-functional-issues

dmatora, 2021-01-26:
@SwoopX article talks about battery powered devices, the issue I was having was affecting all kind of devices (including light bulbs)

Smanar, 2021-01-26:
No one know how to obtain deconz log on HA ? Without using the VNC ? There is not an option inside it ?

Will be usefull to check them for database corruption, badly integration, or other.

closed 2021-03-05 by github-actions[bot]