dresden-elektronik/deconz-rest-plugin #3952: Blitzwolf BW-SHP13 doesn't show power.

caro7372, 2020-12-17:
## Describe the bug

## Steps to reproduce the behavior

## Environment

 - Host system: (Raspberry Pi)
 - Running method: (Home Assistent deCONZ Add-on)
 - Firmware version: (26660700)
 - deCONZ version: (2.07.01)
 - Device: (ConBee II)
 - Do you use an USB extension cable: (yes) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which? Yes, DSMR cable and SSD (USB).

Mimiix, 2020-12-17:
Closing.

- Template not filled.
- Device is supported and working.
-

closed 2020-12-17 by Mimiix

caro7372, 2020-12-17:
So 3 plugs are working and one is not.
What do you suggest other than closing this.
I tried to give as much as info as I could give.

Mimiix, 2020-12-17:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3788

I suggest searching this git.

caro7372, 2020-12-17:
Dennis,

thank you very much for pointing me in the right direction.
I appreciate it very much.
Enjoy Christmas and New-Years eve.

Stay safe.

Mimiix, 2020-12-18:
Thank you @caro7372 :)

Enjoy Christmas and have a happy new year! :)

caro7372, 2020-12-18:
And last but not least, I've got the plug finally in HA.
Did some reset etc in Deconz and it finally showed up with power reading too.