dresden-elektronik/deconz-rest-plugin #3960: Missing nodes after restart host

pluusla, 2020-12-18:
## Describe the bug
I use Homeassistant (Hassio) on my Raspberry Pi 4 as operating system in combination with Conbee 2 and some xiaomi sensors. My host (Raspberry pi 4) is rebooted every night. After each reboot deconz loses the sensors, but they are still displayed in the phoscon app.

I have 2 xiaomi mi door sensors and 2 xiaomi motion sensors

## Steps to reproduce the behavior

1- Add several sensors to a new installation
2- When they are already added, changed the name, restart the deconz host.
3- When you return to deconz, the nodes are missing

I also tried this workaround: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3528#issuecomment-743367042
But without any luck... 

## Expected behavior

  Even after restart, the nodes shouldn't vanished
## Screenshots

After restart deconz:
[image]

sensors after restart phoscon:
[image]

after reconnect the sensor:
[image]

## Environment

 - Host system: Raspberry Pi 4
 - Running method: Home Assistent deCONZ Add-on 
 - deCONZ version: 2.07.01
 - Device: ConBee II 
 - Do you use an USB extension cable: no 
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs

Any tips how i can copy the logs from deconz (noVNC). It seems i can't copy anything from the debug view...

Mimiix, 2020-12-18:
After a reboot, it can take up to 1 hour for them to rejoin. 

> Any tips how i can copy the logs from deconz (noVNC). It seems i can't copy anything from the debug view...

That's probably a bug in the addon. Nothing we can do about it :(

SwoopX, 2020-12-18:
@pluusla Are you really sure you followed that correctly? Forst off, as far as I can tell no need to reset the device, so you can start from step 5 onwards. Also, you will most likely need 2 persons for that: one continuously pushing the button on the device, the other one doing the required steps in deconz GUI.

If the last screenshot taken is the result after you've done the procedure, you've done it either wrongly or unsuccessfully.

pluusla, 2020-12-19:
@SwoopX 
That was the important hint for me! I did indeed need a second person. After that, the nodes were recognized with all the details and remained after the reboot. Great thing, thanks a lot!

closed 2020-12-19 by pluusla