dresden-elektronik/deconz-rest-plugin #3961: Source Routing feature (beta) via command line parameters

tpnprins, 2020-12-18:
## Describe the question or issue you are having
I'm running deCONZ in a headless set-up and would like to use the source routing beta feature (which I tested using the GUI). Can this be activated through CLI parameters, also for the maximum hops & minimum LQI settings? If not, I'd like to kindly request this feature as a product enhancement. Thanks!

## Screenshots

N/A

## Environment

 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: 0x26680700
 - deCONZ version: 2.8.0 beta
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

N/A

## Additional context

N/A

manup, 2020-12-18:
Currently it can only enabled via the GUI or by configure it manually in the `config.ini` file with the follwing section:

```
[source-routing]
enabled=true
max-hops=5
min-lqi=130
```

A proper REST-API for source routing and perhaps command line switch will be part of future updates.

tpnprins, 2020-12-18:
Thanks @manup, clear!

closed 2020-12-18 by tpnprins