dresden-elektronik/deconz-rest-plugin #3967: Device name MS-104Z 1 Gang 2 Way Zigbee Switch 3.0 Moes

testpaul999, 2020-12-19:
## Device

- Product name: The device name as shown on the product or package.
MS-104Z
- Manufacturer: As per deCONZ GUI Basic cluster.
Moes
- Model identifier: As per deCONZ GUI Basic cluster.
Switch Module 
- Device type : Please remove all unrelated device types. 
  - Switch
 

## Screenshots

[image]
[image]
[image]

Smanar, 2020-12-19:
Lol, by capture, it s deconz capture we need ^^
But this device is already suported no ? https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3595

testpaul999, 2020-12-19:
Ok, thx try once more to connect.

closed 2020-12-19 by testpaul999