dresden-elektronik/deconz-rest-plugin #3969: Symfonisk controller broken in v2.8.0

ebaauw, 2020-12-19:
## Describe the bug

No button events are generated for the IKEA Symfonisk controller.

## Steps to reproduce the behavior

Press or dial the controller:

## Expected behavior

Duh

## Screenshots

n/a

## Environment

 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: 26390500)
 - deCONZ version: 2.8.0
 - Device: RaspBee I
 - Do you use an USB extension cable: n/a
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs

```
Dec 19 15:02:18 pi2 deCONZ[21000]: 15:02:17:941 APS-DATA.indication srcAddr: 0x1c8e, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0008, lqi: 255, rssi: -58
Dec 19 15:02:18 pi2 deCONZ[21000]: 15:02:17:943 	asdu: 1174030000
Dec 19 15:02:18 pi2 deCONZ[21000]: 15:02:17:947 [INFO] - No button map for: SYMFONISK Sound Controller endpoint: 0x01 cluster: LEVEL_CONTROL (0x0008) command: STOP (0x03) payload[0]: 000
```

## Additional context

See #1898.

manup, 2020-12-27:
Fortunately I have this thing here, will do some tests.

manup, 2020-12-28:
Fixed in latest PR. As workaround before the next release you can manually edit the `modelid` in the button_maps.json file.

Replace "SYMFONISK" with "SYMFONISK Sound Controller".

kptkip, 2021-01-02:
thnx @manup.
The workaround does it.

Mimiix, 2021-01-04:
This fixed in the latest?
 @ebaauw @manup

kptkip, 2021-01-04:
In 2.9.0 not.

manup, 2021-01-04:
This was after 2.9.0, so will be included in the next release.

labaland, 2021-01-13:
> Fixed in latest PR. As workaround before the next release you can manually edit the `modelid` in the button_maps.json file.
> 
> Replace "SYMFONISK" with "SYMFONISK Sound Controller".

Where can i find that file? i run deconx in docker =)

ebaauw, 2021-03-19:
It works in 2.9.3.

closed 2021-03-19 by ebaauw

AdamBremholm, 2021-08-28:
Having same problem  in 2.11.05

joggs, 2021-12-20:
Same problem 2.13.01

Mimiix, 2021-12-20:
There were some fixes in the last version i think

joggs, 2021-12-20:
The only thing the controller does in 2.13.01 is to show battery life. no events are triggered. When running deconz in docker, there are no button_maps.json that will keep modifications after a restart of the container, so right now, the symfonisk controller  is unusable.

Mimiix, 2021-12-21:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4956 is open still

@thenon can you confirm the above?

TheNON75, 2021-12-21:
Please allow me some days to chase up if all ok, I have some concerns, but currently very afk and with no physical access to my zigbee playground

der-david, 2022-01-06:
> The only thing the controller does in 2.13.01 is to show battery life. no events are triggered. When running deconz in docker, there are no button_maps.json that will keep modifications after a restart of the container, so right now, the symfonisk controller is unusable.

I have had the same issues and I have resolved the problems (for now) by updating the **remotes** firmware to `2.3.080`:
```
ConBee II
Version: 2.13.04 / 12/18/2021
Firmware: 26660700
```

**Assuming your remote is not paired:**
1. Remove battery from Symfonisk
2. In Phoscon _Add new switch_ > _IKEA_ > _TRÅDFRI Dimmer_
3. Insert battery and press pairing button on the back four times quickly (close to your coordinator) and the red light should indicate the joining sequence
4. Ignore the Phoscon web interface and observe deCONZ GUI
5. The device should appear and after a little while you should see it being named "SYMFONISK Sound Controller"
6. Select the device and read the "Basic" attributes in the Cluster Info panel, "SW Build ID" should say `2.1.022` or similar
7. Open the STD OTAU Plugin panel
8. It should say "OTAU enabled"
8. Choose the file [`10043101-3.1-TRADFRI-dimmer-2.3.080.ota.ota.signed`](http://fw.ota.homesmart.ikea.net/global/GW1.0/01.17.019/bin/10043101-3.1-TRADFRI-dimmer-2.3.080.ota.ota.signed) or whichever is the [latest version for 4554](http://fw.ota.homesmart.ikea.net/feed/version_info.json) at the time you are reading this
9. Make sure you have selected the right device
10. Press _Update_
10. The "Progress" column should say "Queued", try if _Query_ starts the update process
11. If not, remove the battery from Symfonisk again and reinsert it after a few seconds
12. I had to retry this a couple of times with different _Update_ and _Query_ sequences
13. The update progress should be displayed in the "Progress" column, the whole process can take up to a few hours depending on your mesh
14. Once the update was complete I had to remove the node and rejoin it
15. In Phoscon, I can now see all of the events by going to _Help_ > _API Information_ > _Switches_ > _My Dimmer_ > _Events_

**For Home-Assistant users:**
At least in Home-Assistant (`2021.12.8`) the remotes left and right events are reversed, but ATM I don't really care about that.
In case you have old deCONZ devices in your HA instance, you can head to `/developer-tools/service` and run the services `deconz.device_refresh` and `deconz.remove_orphaned_entries`.

joggs, 2022-01-07:
That worked perfectly Der-David!
2.5 hours and an almost drained device battery later, the update completed successfully and I can now see events in home assistant.
Big thanks for your detailed response!