dresden-elektronik/deconz-rest-plugin #3970: KlikaanKlikUit smoke detector

JMBokhorst, 2020-12-19:
Hello,

I recently bought a KlikAanKlikUit smoke detector ([link to device](https://klikaanklikuit.nl/product/slimme-draadloze-rookmelder/)). Which I think is a rebrand of the Heiman HS1SA smoke sensor. 

I could pair the device and it is picked up by deconz as a Heiman smoke sensor. The issue I have is that it is _not reachable_ directly after pairing. I tried to pair the device again with the same outcome. I tried to test the device by running it in test-mode but I don't get an update/notification via deconz. 

If I check the device via the GUI, I do see it is connected but I don't see a lot of information. 

Does somebody has experience with this device? Or can somebody help me getting it to work correctly? 

## Screenshots
[image]
[image]
[image]
[image]
[image]

## Environment

 - Host system: PC
 - Running method: Marthoc Docker container 
 - Firmware version: (26807000)
 - deCONZ version: (2.07.01)
 - Device: ConBee II
 - Do you use an USB extension cable: No
 - Is there any other USB or serial devices connected to the host system? If so: Which? Yes, a DSMR slimmer meter cable (RJ-11 to USB). 

Hope you guys can help me

SwoopX, 2021-01-17:
What are the values of the IAS Zone cluster? Scroll down and click on read, make sure the device is awake by pushing it's physical button.

JMBokhorst, 2021-01-17:
Sorry for my ignorance, but I'm not sure where I should scroll down and click on read. I assume the in the tab 'cluster info', there is a read button but it doesn't seem to update the values (even though I selected the smoke detector node):

[image]

SwoopX, 2021-01-17:
The screenshot is not from the IAS zone cluster. However, that's the button also available there.

As said, you need to wake the device prior to pressing read. You do that by pushing the physical button one or twice (like you would do for a reset, just not that long).

JMBokhorst, 2021-01-17:
With the devices woken, I see this as information below the IAS zone info. Not sure if this is what you mean, I cannot click on it and press read (might be the VNC viewer). 

[image]

JMBokhorst, 2021-01-17:
I think I have found it (let me know if this isn't it) 

[image]
[image]

SwoopX, 2021-01-17:
You need to enroll it manually (an improvement for this has just been merged):

1. Open deCONZ.
2. Select the Coordinator (blue node) on the map and copy value of the IEEE field.
3. Select the Motionsensor, click the "ias zone" cluster. 
4. Scroll down to the attributes list to attribute 0x0010
5. double click the "value"
6. put the IEEE value you copied earlier in the textbox above. 
7. Make sure the Sensor is awake. This can be done by clicking the reset button once.
8. Click "Write"

JMBokhorst, 2021-01-17:
I will try that now and will report my findings, thanks!

JMBokhorst, 2021-01-18:
I have added the IEEE field to the sensor. And now it seems to be set, but it still is unavailable.

SwoopX, 2021-01-18:
Has the zone state (0x0000 in your last screenshot) changed to enrolled?

JMBokhorst, 2021-01-18:
No, it's still on 'Not enrolled' 

[image]

SwoopX, 2021-01-18:
Hm, that's still odd. Could you update to 2.09.01 once it is available for docker and then re-pair the device? It has some overhauled code taking care of the automatic enrollement and making it more robust.

JMBokhorst, 2021-01-18:
After updating the node is indeed enrolled, however in the webGUI it still says "Not reachable". Can it be that the did a bit more than just rebrand the device? 

[image]
[image]

SwoopX, 2021-01-18:
Could you share the details from the basic cluster please? I'd particularly interested in the model identifier and manufacturer name to double check the code.

JMBokhorst, 2021-01-18:
Of course, please find attached the information: 

[image]

SwoopX, 2021-01-18:
> Although the information says it's a heiman sensor, the brand on the box is KlikAanKlikUit.

That's not really important here. Code-wise, it looks fine. Could you also share how the device looks like in the REST API?

JMBokhorst, 2021-01-18:
Using a GET with the REST API might have triggered something(?). Now it seems to be reachable: 

[image]

Thanks for the help! =D

JMBokhorst, 2021-02-01:
The sensors is working fine since I have updated to version `2.09.01` and manually set the `ias zone`. Therefore I will close this issue.

closed 2021-02-01 by JMBokhorst

SwoopX, 2021-02-01:
Just FYI: with 2.09.01, the enrollment became way more robust and it shouldn't happen again that a device doesn't get enrolled automatically ;)

Great it works now 👍

JMBokhorst, 2021-02-01:
Super! Thanks for the help :)

Gunthervc, 2022-07-20:
> The sensors is working fine since I have updated to version `2.09.01` and manually set the `ias zone`. Therefore I will close this issue.

Hi, how did you do this update? i have same issue with Heiman version 2019.8.29