dresden-elektronik/deconz-rest-plugin #3971: Exclude "some" lights from group "0"

popy2k14, 2020-12-19:
## Feature request type
extend group "0" handling

## Description
I am using group "0" to turn off all lights when all leaved the place, pressed an special master button or @night 01:00.
During christmas time i have 4x zigbee wall plugs with cristmas lights attached to it.
This i want to exclude and NOT turn off.

## Considered alternatives
Sure i can make an new "My All" group and just put my needed lights in it. Sadly if i'll add any new light to my setup it does'nt gets added automatically into my "My All" group.

An alternative would be the user can setup other groups, where also new lights would be added if newly paired.

## Additional context
Just some ideas

Mimiix, 2020-12-21:
> Sure i can make an new "My All" group and just put my needed lights in it. Sadly if i'll add any new light to my setup it does'nt gets added automatically into my "My All" group.

That would be way better. Instead of modifying behaviour, just have a setting like that.

However, i think this should be doable by using the rest api. IF you have a script, run it everyday at x date. That script checks if there's new devices. If there are, do a post to add them to group X. Just a workaround idea.

SwoopX, 2020-12-21:
@popy2k14 please note that this group is supposed to contain all main powered devices by design. We will not bend that according to you request, as this can be achieved by other means. Iirc, the standard even prescribes that.

The alternate method you mentioned is the intended way.

closed 2020-12-21 by SwoopX

popy2k14, 2020-12-21:
Thx, guys for your ideas and hints.
Was just an idea, but i think i will go the workaround with an script.

SwoopX, 2020-12-21:
A little bit of housekeeping and maintenance resides with the user, a cross that we all have to carry 😉