dresden-elektronik/deconz-rest-plugin #3973: ORVIBO Zigbee Dry Contact CM10ZW support

samsam-rolon, 2020-12-19:
## Device

- Product name: ORVIBO - Module commutateur contact sec Zigbee.
- Manufacturer: ORVIBO 
- Model identifier: CM10ZW
- Device type : 
  - Door Lock

Be careful : the device is configured in the "dry contact mode" but they have 3 mode available with a physical switch on the device :
[image]
file:///C:/Users/Samuel/AppData/Local/Temp/Module-commutateur-contact-sec-CM10ZW-1.pdf
## Screenshots

[image]
[image]
[image]
[image]
[image]
[image]

Smanar, 2020-12-19:
Hello, but you miss something ?
This device don't work natively ?

samsam-rolon, 2020-12-20:
Hum ... no. I didn't see it in API

Smanar, 2020-12-20:
Ha ?
This device is detected as doorlocker, need to check because with the cluster 0x0006 you need to have a light entry.

Smanar, 2020-12-20:
I have made something to correct name and make the device visible in api
On the same code than for the temperature device

```
git clone --branch orvibo https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

samsam-rolon, 2020-12-21:
[image]
Some KO related to a missing `;` line 2213 just before the case (on the break 1 line over)
I've corrected it but the GUI still not showing me a friendly name ; neither the API
[image]

samsam-rolon, 2020-12-21:
Okay, better now :
I see it in the API under lights now. It's okay to change states with API now 
`
  "10": {
    "etag": "8d805b9e009a3651c7d94a84a9bae608",
    "hascolor": false,
    "lastannounced": "2020-12-21T15:54:12Z",
    "lastseen": "2020-12-21T15:54Z",
    "manufacturername": "Unknown",
    "modelid": null,
    "name": "Unknown 10",
    "state": {
      "alert": "none",
      "on": false,
      "reachable": true
    },
    "swversion": null,
    "type": "Unknown",
    "uniqueid": "00:12:4b:00:22:31:80:70-01"
  },
 "22": {
    "etag": "0dd7696faf61ac6c30666afe1fbd510e",
    "hascolor": false,
    "lastannounced": "2020-12-21T15:54:12Z",
    "lastseen": "2020-12-21T15:53Z",
    "manufacturername": "Unknown",
    "modelid": null,
    "name": "Unknown 22",
    "state": {
      "alert": "none",
      "on": false,
      "reachable": true
    },
    "swversion": null,
    "type": "Unknown",
    "uniqueid": "00:12:4b:00:22:31:80:70-02"
  },
  "23": {
    "etag": "32d8c61ae7ef49768db4209135fbd762",
    "hascolor": false,
    "lastannounced": "2020-12-21T15:54:12Z",
    "lastseen": "2020-12-21T15:54Z",
    "manufacturername": "Unknown",
    "modelid": null,
    "name": "Unknown 23",
    "state": {
      "alert": "none",
      "on": false,
      "reachable": true
    },
    "swversion": null,
    "type": "Unknown",
    "uniqueid": "00:12:4b:00:22:31:80:70-03"
  },
`

Smanar, 2020-12-21:
Thx, I have corrected the code, and have added something to have a type instead of "unknow"
But I don't understand why you have no model id and no manufacture name ? It same after a reboot ?

samsam-rolon, 2020-12-21:
> 
> 
> Thx, I have corrected the code, and have added something to have a type instead of "unknow"
> But I don't understand why you have no model id and no manufacture name ? It same after a reboot ?
> 
> And why 3 endpoint ? you have only 1 output ?

I've try to unpair and repair it to join the Zigbee network and same issue : unknow
There is 3 endpoint because I have 3 dry contact relay available on the device

Smanar, 2020-12-21:
Ok.
Have corrected a typo in the name, but it don't explain why you don't have model id or manufacture name ....

nicph, 2021-01-09:
Same situation as in the issue here. Is there any progress on the dry contact mode of the CM10ZW ? Can we do something to help ?

Smanar, 2021-01-09:
@nicph you are using the recompiled code ?

nicph, 2021-01-10:
No, i did not find the branch mentionned above on this repo, but i just noticed it was on a fork. I'll try

nicph, 2021-01-10:
Ok I recompiled the plugin and now I can reach the 3 entities from the API, thank you. Is there any plan to merge this here ?

Smanar, 2021-01-10:
Yep, I will make the PR today, but there is still something strange, the missing model id for exemple.
You have it on the JSON on your side ?

nicph, 2021-01-10:
Great news !

No, i have `"modelid": null,` for the 3 endpoints.

samsam-rolon, 2021-01-10:
Same issue on my side :
``` json
{
  "etag": "788c35fb76a9f1ca4381f40f8b099bd5",
  "hascolor": false,
  "lastannounced": "2021-01-10T12:06:49Z",
  "lastseen": "2021-01-10T12:08Z",
  "manufacturername": "Unknown",
  "modelid": null,
  "name": "Door Lock Unit 22",
  "state": {
    "alert": "none",
    "on": false,
    "reachable": true
  },
  "swversion": null,
  "type": "Door Lock Unit",
  "uniqueid": "00:12:4b:00:22:31:80:70-01"
},
{
  "etag": "8c48f6a3e1cc1f9d956bf7c6747c62c5",
  "hascolor": false,
  "lastannounced": "2021-01-10T12:32:31Z",
  "lastseen": "2021-01-10T12:38Z",
  "manufacturername": "Unknown",
  "modelid": null,
  "name": "Door Lock Unit 23",
  "state": {
    "alert": "none",
    "on": false,
    "reachable": true
  },
  "swversion": null,
  "type": "Door Lock Unit",
  "uniqueid": "00:12:4b:00:22:31:80:70-02"
},
{
  "etag": "352d6926639e25e1f2b1ff7196b9bc58",
  "hascolor": false,
  "lastannounced": "2021-01-10T12:39:11Z",
  "lastseen": "2021-01-10T12:38Z",
  "manufacturername": "Unknown",
  "modelid": null,
  "name": "Door Lock Unit 24",
  "state": {
    "alert": "none",
    "on": false,
    "reachable": true
  },
  "swversion": null,
  "type": "Door Lock Unit",
  "uniqueid": "00:12:4b:00:22:31:80:70-03"
}
``` 
[image]
[image]
[image]

Smanar, 2021-01-10:
It s realy strange, someone can share the log (info / info_l2) during the inclusion ?

samsam-rolon, 2021-01-10:
Sorry, why ?

samsam-rolon, 2021-01-10:
@Smanar I didn't have it. 
I need to re-include ... Sorry, why ?

Smanar, 2021-01-11:
Not realy important, just because I don't understand why manufacture and model id are not working.
And more important, just to be sure deconz is not spamming requests to retreive them. For that, you can just take a look on deconz / help / log if you have the desktop.

samsam-rolon, 2021-01-12:
Stupid question :
Is any way to let works the "delay" part ?
I would like to configure it as a "1 push and relase button" for my garage door 😄 
[image]

Smanar, 2021-01-12:
Perhaps here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3394
It seem there is an hiden command "on with delay"

samsam-rolon, 2021-01-12:
With the GUI, the "On time" or "Off wait time" is not working

Smanar, 2021-01-12:
Ha ok, ok so look at the error message, but it s probably because the device don't support it.

nicph, 2021-01-14:
> It s realy strange, someone can share the log (info / info_l2) during the inclusion ?

Attached [here](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5811848/orvibo.log)

Smanar, 2021-01-14:
Thx @nicph but it s worse now ^^

The device is managed like tuya one (because of the manufacture number = 0x0000) so the API search a manufacture name in the database, and find it
>00:48:27:684 DB SELECT manufacturername FROM nodes WHERE mac LIKE '00:12:4b:00:22:02:89:5a%' COLLATE NOCASE: ORVIBO

So the code set it, from the start. So why you loose it during the inclusion ...

You have not waht happen after ? The device was not yet included on your logs.

nicph, 2021-01-14:
> Thx @nicph but it s worse now ^^

This may be related to the fact I'm not using the patched version anymore ? Because the docker container restarted at some time.

nicph, 2021-02-02:
@Smanar i had the opportunity to reset my setup, here are the logs with the patched version active
[deconz-0201-orvibo.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5909482/deconz-0201-orvibo.log)

Smanar, 2021-02-02:
Ok, so now it s working like the samsam-rolon one.
The code ask for model id, but It never have godd response, realy strange.
It seem the task is created, and immediatly erased ....
```
22:33:49:882 read attributes of 0x00124B002202895a cluster: 0x0000: [
22:33:49:886 0x0004
22:33:49:888 ]
22:33:49:891 add task 6500 type 19 to 0x00124B002202895a cluster 0x0000 req.id 103
22:33:50:145 Erase task req-id: 103, type: 19 zcl seqno: 139 send time 0, profileId: 0x0104, clusterId: 0x0000
```

And BTW why you have 3 endpoint ?

nicph, 2021-02-03:
> And BTW why you have 3 endpoint ?

That's because the device has 3 distinct relays

Smanar, 2021-02-15:
BTW the manufacture name and the model id is still missing ?
I need to correct that for the PR will be validated.

samsam-rolon, 2021-02-15:
Yes it's working for me only after re-inclusion

Smanar, 2021-02-16:
Ha ? ^^ seriously ? you have a complete json now ?
Can you share it pls ?

Smanar, 2021-02-22:
@samsam-rolon we are waiting for the JSON to validate the PR.
Of if you have the values gived by your third app ? just need the model ID and the manufacture name.

samsam-rolon, 2021-02-22:
euhm, sorry, I didn't understand what you are waiting for ? I was thinking all was complete for the PR !

Smanar, 2021-02-22:
Yes, but manup don't want tp merge the PR, if you still have empty value.
And I want to check if the convertion have worked, I m trying to set my own model id, instead of the magic string on the firmware a7z7z4d44ze74z41s5qsz1s1x41d5c.

So I just need thoses 2 values.

herveja, 2021-02-23:
Hi Smanar,
I'm bit a beginer here, but if I could help, here is what my json look like for the ORVIBO dry contact CM10ZW
I'm waiting for the merge to see how it goes in HA. For now it's not realy usefull :)

 "7": {
        "etag": "9fb62ad1e409eaa37dac6e05a0e4de9b",
        "hascolor": false,
        "lastannounced": "2021-02-22T17:47:59Z",
        "lastseen": "2021-02-22T17:47Z",
        "manufacturername": "ORVIBO",
        "modelid": "82c167c95ed746cdbd21d6817f72c593",
        "name": "Window covering controller 7",
        "state": {
            "bri": 0,
            "lift": 0,
            "on": false,
            "open": false,
            "reachable": false
        },
        "swversion": "v3.1.06",
        "type": "Window covering controller",
        "uniqueid": "00:12:4b:00:22:31:80:5a-01"
    }

Smanar, 2021-02-23:
Nice Thx a lot, have made a typo

>"modelid": "82c167c95ed746cdbd21d6817f72c593",

And I m using 

>else if (str == QLatin1String("82c167c95ed756cdbd21d6817f72c593"))

I have updated the code, but no need to retest it, I think it will be enought for manup. The code will be probably merged in the next week version.

samsam-rolon, 2021-03-02:
> "modelid": "82c167c95ed746cdbd21d6817f72c593",

Yead I Confirm this model ID

closed 2021-03-03 by manup

samsam-rolon, 2021-04-21:
Is any one get some issues with this devices ? Like not responding ... :(

Smanar, 2021-04-21:
I haven't the device, but it s strange, it s a router.
Something special visible in deconz GUI or in log ?

samsam-rolon, 2021-04-21:
It's a router device, yeah ! Sometimes in GUI led is flashing RED instead of Green/Blue
Nothing in logs ... but i'm running it with those options : 
```
environment:
      - DECONZ_WEB_PORT=38080
      - DECONZ_WS_PORT=38081
      - DEBUG_INFO=1
      - DEBUG_APS=0
      - DEBUG_ZCL=0
      - DEBUG_ZDP=0
      - DEBUG_OTAU=0
      - DECONZ_VNC_MODE=1
      - DECONZ_VNC_PORT=35900
      - DECONZ_VNC_PASSWORD=HIDE
      - DECONZ_UPNP=1
      - DECONZ_DEVICE=/dev/ttyACM0
```

samsam-rolon, 2021-04-21:
Also, I have a CC2530 + CC2591 for extending zigbee range to this area

Smanar, 2021-04-21:
Can be a range issue ?
Mesh issue ?

You can try to set other value for reporting (will work up to deconz change them again)

samsam-rolon, 2021-04-21:
> 
> 
> Can be a range issue ?
> Mesh issue ?
> 
IDK : the device is well connected to my extender router so in theory no
[image]

> You can try to set other value for reporting (will work up to deconz change them again)

Any advices ?

Smanar, 2021-04-22:
Not me, and as nothing appear in logs ...
The connexion look good on deconz, but it s just an indication. I think your garage door is in metal ? I have ever see some issues about sensor sensible to metal support ?

samsam-rolon, 2021-04-23:
Yes my garage door is in metal but the extender and the dry contact are not near the door but on the shortest path "on the fly" to receive good signal ...

samsam-rolon, 2021-04-23:
I will analyse log with debug options after 1 week

0rsa, 2022-02-21:
Hello,
I just received my CM10ZW support and I'm back to the initial problem.
I can see an unknown device in deconz GUI and there is no entry in the API.
Cluster basic info is a bit different from author:
[image]
[image]

Smanar, 2022-02-21:
Different model id mean different device for deconz.
But I have take a look faslty on code, there is nothing special for this device, I think you can use it with DDF.

Do you know what is DDF ? https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet
You just need to drag and drop 3 "On/Off plug-in unit" with the good endpoint.

0rsa, 2022-02-23:
Thank you @Smanar, your help is really appreciated.
I successfully edited DDF to map a On/Off light with good endpoints.
It is working but I'm disappointed by the module behaviour itself, I mean you have to toggle on to activate dry contact and you have to toggle off to "open" dry contact. There is no security, the module can keep the contact dry forever.

I will stay with my Tuya dry contact which has a dry contact which automatically deactivates after a few hundred ms even if it not officially supported #5685

mc4one, 2022-04-15:
Hi Guys,

I have a issue like some people there, but I don't understand all :/ I'm french, maybe my english is too bad :D 

I'm on Home Assistant, using Conbee2 and the add-on Deconz.

I tried several times to reset the modul (4 press 5 press long) but when I start a discorey (light or switch) nothing was detected. But on Deconz I see a node with number. In the parameter I see that it's the Orvibo.
For now I have nothing plug into (like my garage door to make a try).
On Home assistant I have aapparel "no name" with no entities.

Like you see here, and l ike some people, manufacturer and model identifiier are empty :

[image]

But on the node I see Door lock Controller :

[image]

And If I check on "Basic" and "cluster info" I see the name and model identifer...

[image]

Any idea to help me ? Thank you :)

Smanar, 2022-04-15:
So I have a bad new and a good new.

You have a new device, nothing to see with the one from this issue (not same model id, not same manufacture number), so natively It will be never detected.

So same answer than here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3973#issuecomment-1047104875
And like @0rsa haven't share his DDF, need to start from scratch.

But don't worry it s fast and easy for this kind of device, you have a video on the link, but he have give some information on the second post.

If it don't work, can you share the DDF you have to correct it.

mc4one, 2022-04-15:
Thank's @Smanar for your help.

I succeeded something :)

Now I have the Manufacturer and model identifier.
Just one of the 3 dry contact are visible on the light device like switch. And I have 1/3 entitie visible.

I think I forget something on the DDF.

Here my DDF preview (sorry copy past doesn't work :/) : 
[image]
[image]
[image]

This screen shot is what I drag and Drop : 

[image]

Smanar, 2022-04-16:
>I think I made a mistake about the endpoint. I don't really undestand what I need to do.

Yep exaclty, on your DDF you are creating 3 device, but all with the same uniqueID (uuid) it's based on $adress.ext and the endpoint.

So just use 0x01 0x02 and 0x03

If it don't work after that can be conflict, so we will use the value instead of "auto" on some endpoint field.
And the device will probably work but you will not have return, will be normal, still missing binding/report. Not realy usefull on this kind of device, there is is no manual action possible, but can be usefull if for exemple you restart deconz.

mc4one, 2022-04-16:
I make a try yesterday before your message. But not in the right place :D 

Now I think it's good :)
[image]

And to show for other users where to change the value : 
[image]

And on Haas I have 3 entities on the device : 
[image]

@Smanar can I change the name of the product ? Just to have something clear ?
[image]

And my last question (I hope :D) Where can I configure a quick on/off to have a good dry contact for my garage door.

Smanar, 2022-04-16:
>can I change the name of the product ? Just to have something clear ?

Yeah and it s a good idea, you just need to keep the "long number" for the model ID.

>Where can I configure a quick on/off to have a good dry contact for my garage door.

Will be in the request you make you need to send for exemple to the /state the body `{"on": true, "ontime": 3}`

You are using it without binding/report ?

mc4one, 2022-04-16:
> Will be in the request you make you need to send for exemple to the /state the body `{"on": true, "ontime": 3}`
> 
> You are using it without binding/report ?

I don't find the right way to modify the /state  :(
And, I don't know what is binding/report. I have something with "binding" on DDF but nothing on it and I can't do anything.

Smanar, 2022-04-17:
I don't use HA, so I think it s better to ask on the plugin support if there is a way to use the "ontime" parameter. It's used generaly for bulb.
Will be more robust than using an automation, but yes using automation works too.

>I just see somethinh, @0rsa using "On/Off light" and me using "On/Off plug-in unit". Which is the better ?

Not a big change on the result, but "On/Off light" is generaly for light and "On/Off plug-in unit" for plug. Your device have the type "Door lock controller", this type is not yet in the DDF core (I will make a PR in 10mn). It s possible the parameter "ontime" is only used for a type and not another on HA, but I can't test on my side.

BTW, you have no way to share the DDF as copy paste don't work ? Can't save it in a "shared folder". I can try to make it with a test editor, but easy to make typo error.

mc4one, 2022-04-19:
It's very difficult, I can't save the file in a other place than the purpose of Deconz :/

And by the way, I finally understand with your link the possibility to drag and drop on Binding section. But... I can't do this. Home assistant use VNC to control Deconz and when I switch between the Zigbee mapping and DDF it's a new window. I can't reduc them and make a drag & drop.

It's never simple...

[31 more replies]