dresden-elektronik/deconz-rest-plugin #3974: device state timeout ignored in state 2, device disconnected reason: 1, index: 0

SuperSpe, 2020-12-19:
## Describe the question or issue you are having
I have installed raspbee for about two weeks and everything worked correctly, i had 3 sensors. one day out of the blue i can't connect to deconz or phoscon app anymore. I try everything, even fresh HASSIO install...nothing change. I think the Raspbee is broken. I try /dev/ttyAMA0 and /dev/ttyS0. same.

## Screenshots

## Environment
 - Host system: Raspberry Pi 
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: stock
 - deCONZ version: 6.6.1
 - Device: RaspBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] firmware.sh: executing... 
[00:02:46] INFO: GCFFlasher V3_17 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyAMA0     | 0x0000 | 0x0000  |            | RaspBee 
[cont-init.d] firmware.sh: exited 0.
[cont-init.d] nginx.sh: executing... 
[cont-init.d] nginx.sh: exited 0.
[cont-init.d] novnc.sh: executing... 
[cont-init.d] novnc.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[00:02:47] INFO: Starting udevd...
[00:02:47] INFO: Running the deCONZ OTA updater...
[00:02:47] INFO: Running the IKEA OTA updater...
[00:02:47] INFO: Websockify waiting for VNC to start
[00:02:47] INFO: Running the OSRAM LEdvance OTA updater...
[00:02:47] INFO: Starting VNC server (local/yes)...
[00:02:48] INFO: Starting websockify...
WebSocket server settings:
  - Listen on 127.0.0.1:5901
  - Flash security policy server
  - Web server. Web root: /usr/share/novnc
  - No SSL/TLS support (no cert file)
  - proxying from 127.0.0.1:5901 to 127.0.0.1:5900
[00:02:51] INFO: deCONZ waiting for VNC to start
[00:02:51] INFO: Starting the deCONZ gateway...
QCoreApplication::arguments: Please instantiate the QApplication object first
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
[00:02:58] INFO: Starting Nginx...
2020/12/20 00:02:58 [notice] 328#328: using the "epoll" event method
2020/12/20 00:02:58 [notice] 328#328: nginx/1.14.2
2020/12/20 00:02:58 [notice] 328#328: OS: Linux 4.19.127-v7
2020/12/20 00:02:58 [notice] 328#328: getrlimit(RLIMIT_NOFILE): 1048576:1048576
2020/12/20 00:02:58 [notice] 328#328: start worker processes
2020/12/20 00:02:58 [notice] 328#328: start worker process 1093
QObject::connect: Cannot queue arguments of type 'QTextBlock'
(Make sure 'QTextBlock' is registered using qRegisterMetaType().)
QObject::connect: Cannot queue arguments of type 'QTextCursor'
(Make sure 'QTextCursor' is registered using qRegisterMetaType().)
  - No SSL/TLS support (no cert file)
  - proxying from 127.0.0.1:5901 to 127.0.0.1:5900
[00:02:51] INFO: deCONZ waiting for VNC to start
[00:02:51] INFO: Starting the deCONZ gateway...
QCoreApplication::arguments: Please instantiate the QApplication object first
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
[00:02:58] INFO: Starting Nginx...
2020/12/20 00:02:58 [notice] 328#328: using the "epoll" event method
2020/12/20 00:02:58 [notice] 328#328: nginx/1.14.2
2020/12/20 00:02:58 [notice] 328#328: OS: Linux 4.19.127-v7
2020/12/20 00:02:58 [notice] 328#328: getrlimit(RLIMIT_NOFILE): 1048576:1048576
2020/12/20 00:02:58 [notice] 328#328: start worker processes
2020/12/20 00:02:58 [notice] 328#328: start worker process 1093
QObject::connect: Cannot queue arguments of type 'QTextBlock'
(Make sure 'QTextBlock' is registered using qRegisterMetaType().)
QObject::connect: Cannot queue arguments of type 'QTextCursor'
(Make sure 'QTextCursor' is registered using qRegisterMetaType().)
00:02:58:375 HTTP Server listen on address 0.0.0.0, port: 40850, root: /usr/share/deCONZ/webapp/
00:02:58:929 CTRL. 3.27.200:02:59:084 dev /dev/ttyAMA0
00:02:59:085 COM: /dev/ttyAMA0 / serialno: , RaspBee
00:02:59:086 ZCLDB init file /data/.local/share/dresden-elektronik/deCONZ/zcldb.txt
00:02:59:339 parent process s6-supervise
00:02:59:340 gw run mode: docker/hassio
00:02:59:340 GW sd-card image version file does not exist: /data/.local/share/dresden-elektronik/deCONZ/gw-version
00:02:59:341 sd-card cid: 035344534130384780ffffffff0133a9
00:02:59:343 DB sqlite version 3.27.2
00:02:59:346 DB PRAGMA page_count: 33
00:02:59:347 DB PRAGMA page_size: 4096
00:02:59:347 DB PRAGMA freelist_count: 0
00:02:59:347 DB file size 135168 bytes, free pages 0
00:02:59:348 DB PRAGMA user_version: 7
00:02:59:348 DB cleanup
00:02:59:349 DB create temporary views
00:02:59:358 started websocket server at port 8081
00:02:59:363 [INFO] - Found file containing button maps. Parsing data...
00:02:59:369 [WARNING] - Button map 'sunricherCCTMap' in JSON file has no assigned ModelIDs. Skip loading button map...
00:02:59:386 [INFO] - Button maps loaded.
00:02:59:389 dlg action: Read binding table
00:02:59:390 found node plugin: libde_rest_plugin.so - REST API Plugin
00:02:59:395 found node plugin: libde_signal_plugin.so - Signal Monitor Plugin
00:03:06:856 OTAU: create 1189-001A-00102428.zigbee
00:03:06:936 OTAU: create 1189-000F-00102428.zigbee
00:03:07:036 OTAU: create 1189-0011-00103101.zigbee
00:03:07:456 OTAU: create 110C-0008-01020509.zigbee
00:03:07:537 OTAU: create 110C-006B-01020510.zigbee
00:03:07:637 OTAU: create 110C-0062-01020510.zigbee
00:03:07:712 OTAU: create 110C-0063-01020510.zigbee
00:03:07:782 OTAU: create 110C-0013-01020510.zigbee
00:03:07:793 found node plugin: libstd_otau_plugin.so - STD OTAU Plugin
00:03:07:865 dev /dev/ttyAMA0
00:03:07:865 COM: /dev/ttyAMA0 / serialno: , RaspBee
00:03:08:260 dev /dev/ttyAMA0
00:03:08:594 Announced to internet https://phoscon.de/discover
00:03:08:596 discovery server date: Sat, 19 Dec 2020 23:03:08 GMT
00:03:08:597 	 local time seems to be ok
00:03:08:598 discovery found version 2.04.35 for update channel stable
00:03:08:932 dev /dev/ttyAMA0
00:03:08:933 COM: /dev/ttyAMA0 / serialno: , RaspBee
00:03:08:982 COM check bootloader
00:03:10:983 COM timout query bootloader, assume application
00:03:12:526 device state timeout ignored in state 2
00:03:15:526 device state timeout ignored in state 2
00:03:18:527 device state timeout ignored in state 2
00:03:18:595 dev /dev/ttyAMA0
00:03:21:526 device state timeout ignored in state 2
00:03:24:527 device state timeout ignored in state 2
00:03:27:027 start reconnect to network
00:03:27:527 device state timeout ignored in state 2
00:03:28:591 dev /dev/ttyAMA0
00:03:30:527 device state timeout ignored in state 2
00:03:32:278 try to reconnect to network try=1
00:03:33:527 device state timeout ignored in state 2
00:03:36:527 device state timeout ignored in state 2
00:03:37:526 try to reconnect to network try=2
00:03:38:594 dev /dev/ttyAMA0
00:03:42:027 device state timeout ignored in state 2
00:03:42:526 try to reconnect to network try=3
00:03:47:027 device state timeout ignored in state 2
00:03:47:527 try to reconnect to network try=4
00:03:48:592 dev /dev/ttyAMA0
00:03:52:027 device state timeout (handled)
00:03:52:088 dev /dev/ttyAMA0
00:03:52:089 COM: /dev/ttyAMA0 / serialno: , RaspBee
00:03:52:091 device disconnected reason: 1, index: 0
00:03:52:101 device disconnected reason: 1, index: 1
00:03:52:527 failed to reconnect to network try=5
00:03:53:026 wait reconnect 15 seconds
00:03:53:527 void zmMaster::handleStateIdle(zmMaster::MasterEvent) not connected goto OFF state
00:03:53:528 device state timeout ignored in state 1
00:03:54:027 wait reconnect 14 seconds
00:03:55:026 wait reconnect 13 seconds
00:03:56:027 wait reconnect 12 seconds
00:03:57:026 wait reconnect 11 seconds
00:03:57:526 failed to reconnect to network try=6
00:03:58:027 wait reconnect 10 seconds
00:03:58:587 dev /dev/ttyAMA0
00:03:59:026 wait reconnect 9 seconds
## Additional context
even if the config is /dev/ttyS0, the log keep saying /dev/ttyAMA0

Mimiix, 2020-12-21:
Hi :) 

Did you click "firmware update"  in the addon?

SuperSpe, 2020-12-21:
Do you mean in the phoscon app?
Cause I can reach it, when I connect it says "no network found"

Mimiix, 2020-12-21:
Yes, did you click that button prior having this issue?

SuperSpe, 2020-12-21:
I am using a Raspbee II, so no USB is directly attached to the GPIO.
Since there was no update button when it was working, I updated the firmware manually to HYPER WORM, but after that all worked flawless.
I did another test, using another raspberry but the result is the same.
thanks

SuperSpe, 2020-12-21:
so I manually downgrade to deCONZ_RaspBeeII_0x26660700.bin.GCF (was deCONZ_RaspBeeII_0x26690700.bin.GCF before) and now is working,
what is going on?

Phreet, 2021-01-11:
Had the same issue ("device state timeout ignored in state 2") on Raspberry 4B and a RaspBee2 device for Home-Assistant and deCONZ add-on, Since i manually disabled WiFi and Bluetooth on Raspberry, i found out that following dtoverlay fixed the issue for me:

```
dtoverlay=disable_bt
dtoverlay=disable_wifi
dtoverlay=miniuart-bt <- this is the fix
```

With "dtoverlay=miniuart-bt" i added a new device under system ("/dev/ttyS0") and when sticking to "/dev/ttyAMA0" in deCONZ add-on, it found my RaspBee2 and worked flawlessly.
In addition: i setup RaspbianOS in parallel to update the RaspBee2 firmware to latest (deCONZ_RaspBeeII_0x26690700.bin.GCF  )

closed 2021-03-05 by github-actions[bot]