dresden-elektronik/deconz-rest-plugin #3977: triggers events (sensors/wireless switches) not getting published via REST API for hours after every restart

s0ftice, 2020-12-20:
## Describe the bug
I have a weird issue for the last 2-3 weeks. After every deCONZ restart, the REST API does not send events triggered by sensors and wireless switches to Home Assistant (HA) anymore. Consequently, the status of my sensors (IKEA and Xiaomi) is stuck in HA at what they were before rebooting the deCONZ Raspberry Pi.
I enabled debug logging in HA and where I usually see the state update events immediately, e.g. upon pressing a button on a wireless switch, nothing is showing up.
However, I can see the correct sensors state in Phoscon, it updates immediately. I can also see the blue activity circle in deCONZ GUI when I press a wireless switch.

HA is successfully connected to the REST API as I can send manual triggers from HA to deCONZ. For example turning on/off ZigBee lights, power sockets, etc. via HA works fine immediately after restarting deCONZ.
Seems like in my case, the REST API forgets to publish triggers events (sensors/wireless switches) after a restart, while at the same time it is still able to receive and handle events from consumers (here: HA).

This lasts for several hours until it "magically heals itself" and I can see the the described events in the HA log and their respective states in my HA dashboards again.

What I tried to fix the issue (but did not help)
- Restarting HA
- Restarting the deCONZ device
- Restarting the deCONZ application
- Moving deCONZ from a RPi2 and RPi3 (same microSD card)

Appreciate any help I can get.

## Steps to reproduce the behavior
Restart deCONZ: either closing the UI application via VNC or rebooting the RPi, both will lead to the described behaviour.

## Expected behavior
Events triggered by sensors and wireless switches should be published on the REST API. Especially as I can see these events appearing in the deCONZ GUI and deCONZ debug log.

## Screenshots
none. Happy to provide if needed but don't think it's helpful in my case. The log file attached is probably much more important.

## Environment
 - Host system: Raspberry Pi 3
 - Running method: Raspbian from https://phoscon.de/en/conbee/sdcard, (Home Assistant 2020.12.1 is running on different hardware - both are connected via LAN cable)
 - Firmware version: 26680700
 - deCONZ version: 2.07.01
 - Device: ConBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs
This is what I can see in the HA log when it's working fine (pressing button on a wireless switch).
`Dec 20 09:51:49 server 072194f2a485[495]: 2020-12-20 09:51:49 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"5","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2020-12-20T01:51:49.374"},"t":"event","uniqueid":"bc:33:ac:ff:fe:xx:xx:xx-01-1000"}`

But I do not see above entries for several hours after restarting deCONZ. However, I see other REST API events from deCONZ after a restart of deCONZ immediately. For example:
`Dec 20 10:23:35 server 072194f2a485[495]: 2020-12-20 10:23:35 DEBUG (MainThread) [pydeconz.websocket] {"attr":{"id":"7","lastannounced":"2020-12-08T01:02:07Z","lastseen":"2020-12-20T02:23Z","manufacturername":"IKEA of Sweden","modelid":"TRADFRI Signal Repeater","name":"repeat02","swversion":"2.2.005","type":"Range extender","uniqueid":"68:0a:e2:ff:fe:xx:xx:xx-01"},"e":"changed","id":"7","r":"lights","t":"event","uniqueid":"68:0a:e2:ff:fe:xx:xx:xx-01"}`

Hence, HA is connected to the REST API and is able to receive certain events from deCONZ/Phoscon. Just not events from sensors or wireless switches.

I attached the deCONZ log recorded immediately after I restarted deCONZ. I pressed the button of an IKEA switch (E1743) several times, which is visible in the log.

[deconz.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5720431/deconz.log)
## Additional context
none

majorsl, 2020-12-24:
A couple of us have very similar issues with sensors, see here: #3805 ...although I know I didn't wait for hours for it to heal.  That might be something I will try.

s0ftice, 2020-12-25:
yes looks quite similar. let me know if waiting helps for you. mine can take up to 5-6 hours until everything works again. but it's pretty binary, either none of them (sensors and wireless switches) work, or then suddenly all of them work (reliably, it does not stop working until the next reboot)

Mimiix, 2020-12-28:
If phoscon receive the right status, then there's something going on at ha I think. Phoscon uses the same api and websockets.

@kane610 is that the correct assumption?

Either way, I've only seen this if ha was started before the addon/deconz was ready (started)

Kane610, 2020-12-28:
I can only say what I see. If some events get to hass but not others it looks to me like a source issue

Mimiix, 2020-12-31:
Closed on request

closed 2020-12-31 by Mimiix