dresden-elektronik/deconz-rest-plugin #3983: Conbee II - USB reconnection loop on Raspbbery Pi

TungstenE2, 2020-12-21:
## Describe the bug
Conbee II - USB reconnection loop on Raspbbery Pi

deCONZ GUI takes about 10 min to start.

In GUI the Conbee II keeps connecting and disconnection.

Once connected seems stable.

Renaming of a node is not persisted after a rebbot.

## Steps to reproduce the behavior

see above

## Expected behavior

## Screenshots

## Environment
Updated to 2.08.00 raspbian-stretch-beta

## deCONZ Logs

## Additional context

closed 2020-12-21 by Mimiix

TungstenE2, 2020-12-21:
to be honest, this is not how software is supposed to work.
There are so many bugs, you should be happy supporting you by reporting.

have a nice day

Mimiix, 2020-12-21:
Let me explain a bit. 

First, we need a detailed explanation on what you did. Because there is multiple ways to get this behaviour. From a broken PSU to a firmware issue.  Secondly, it is important to us how you run deconz(environment).

It's in your best interest to fill these fields, to get the help you need. 

_Help us help you_

TungstenE2, 2020-12-21:
by the way, the same issue was already reported, and people reporting the same behavior were asked to open new issue. 

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3001

Mimiix, 2020-12-21:
Yes, i don't mind opening issues. But you need to follow the template and provide all information that is asked.