dresden-elektronik/deconz-rest-plugin #3986: Develco Temperature and Humidity Sensor HMSZB-110

myka280, 2020-12-21:
## Device

- Product name: Temperature and Humidity Sensor
- Manufacturer: Develco Products A/S
- Model identifier: HMSZB-110
- Device type : 
  - Sensor
 
## Screenshots

[image]
[image]
[image]
[image]
[image]
[image]
[image]

Please add device support to Phoscon. Sensor should expose battery, temperature, humidity and link quality readings

SwoopX, 2020-12-22:
@myka280  Can you please update the 2nd screenshot? The information from the node info panel belongs to your coordinator ;)

myka280, 2020-12-23:
Sure.
[image]

closed 2020-12-23 by manup