dresden-elektronik/deconz-rest-plugin #3989: Issue controlling revolt thermostat

WelpeLP, 2020-12-21:
## Describe the question or issue you are having

I want to control my thermostats by revolt using deCONZ and its REST API.
Support was added in the pull request [https://github.com/dresden-elektronik/deconz-rest-plugin/pull/3889](url),
assuming the hardware revision didn't change anything but the manufacturer string.
When sending a PUT request to the API (.../api/.../sensors/7/config; `"heatsetpoint": 2500`), I get a `"success"` response.
A GET request now gives: `"heatsetpoint": 2500`.
Seems alright up to here.
Going to the thermostat I read: 21.5c, like set manually before.
Turning to a different temperature, e.g. 23c will also return `"heatsetpoint": 2300`.
The problem is, I can only read values from the device, and not ''write"/set them.

## Screenshots
[image]
[image]
[image]

JSON response:
```
{
    "config": {
        "battery": 100,
        "heatsetpoint": 2200,
        "locked": false,
        "offset": 0,
        "on": true,
        "preset": "manual",
        "reachable": true,
        "schedule": {},
        "schedule_on": false,
        "setvalve": true,
        "windowopen_set": false
    },
    "ep": 1,
    "etag": "e22e38409e2ba08a749b608a34d1f498",
    "lastseen": "2020-12-21T13:24Z",
    "manufacturername": "_TZE200_kfvq6avy",
    "modelid": "TS0601",
    "name": "Thermostat",
    "state": {
        "lastupdated": "none",
        "on": null,
        "temperature": 2320
    },
    "type": "ZHAThermostat",
    "uniqueid": "84:71:27:ff:fe:19:02:90-01-0201"
}
```

## Environment
Using fork by kgorszczyk: [https://github.com/kgorszczyk/deconz-rest-plugin](url)

 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: 26610700
 - deCONZ version: 2.07.01 / 12/8/2020
 - Device: RaspBee II
 - Is there any other USB or serial devices connected to the host system? No.

Mimiix, 2020-12-21:
@WelpeLP That one isn't merged yet. 

However, you are on the old version. @Smanar , can you explain a bit here?

WelpeLP, 2020-12-21:
> @WelpeLP That one isn't merged yet. 
> 
> However, you are on the old version. @Smanar , can you explain a bit here?
> 

I am using an older (and forked) version, because that has the added support for my thermostat and I'm not really that familiar with git and the c++ language.
The merge request for the 5 lines of code changed is still open since two weeks and hasn't been reviewed yet.

Smanar, 2020-12-22:
We have the same problem for another device, I think you have same issue than here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3922#issuecomment-749135655

WelpeLP, 2020-12-22:
> We have the same problem for another device, I think you have same issue than here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3922#issuecomment-749135655

So you mean they could've used the old enclosure with a new board/firmware inside?
If that's the case, I can test the code from the other issue when it is available and the detection for my device is added.

Smanar, 2020-12-22:
The code is not finished, but we have same issue, the device is in the api but no temperature report.

WelpeLP, 2020-12-22:
> The code is not finished, but we have same issue, the device is in the api but no temperature report.

Hello again,

There's something different to that thermostat, although I have a similar problem.
I **am** able to read the current set temp and the room temp using the integration for the essentials tuya thermostat added earlier. Maybe this is a "crossover" of these two device editions?
I just have problems setting the temperature on the thermostat. 
So when that's working, it would be enough for me.
I just want to integrate these thermostats into Home Assistant with schedules/automations, and I've bought 12 of these for my whole house (all of them work/have the same issue with deconz).

If you need additional information (e.g. debug logs, serial numbers, ...), I'd be glad to help you out with that.

Smanar, 2020-12-23:
ha sorry.
Can you check if you don't need to "wake up" the device before sending the command ?

Can you take a look in deconz log (deconz / help) with filter "info" and 'info_l2" for lines starting by "Tuya debug", try to change manualy the setpoint, and using automation pls.

WelpeLP, 2020-12-23:
> ha sorry.
> Can you check if you don't need to "wake up" the device before sending the command ?
> 
> Can you take a look in deconz log (deconz / help) with filter "info" and 'info_l2" for lines starting by "Tuya debug", try to change manualy the setpoint, and using automation pls.

Hi, I tested setting the temp, here's the log:

directly sending put request to api:
```
12:33:01:199 Tuya debug 4 : Address 0x847127FFFE190290 Payload 008102020004000000dc
12:33:01:200 Tuya debug 5 : Status: 0 Transid: 129 Dp: 514 (0x02,0x02) Fn: 0 Data 220
```

setting the temp in home assistant:
```
12:33:11:699 Tuya debug 4 : Address 0x847127FFFE190290 Payload 008202020004000000e6
12:33:11:700 Tuya debug 5 : Status: 0 Transid: 130 Dp: 514 (0x02,0x02) Fn: 0 Data 230
```

WelpeLP, 2020-12-23:
> > ha sorry.
> > Can you check if you don't need to "wake up" the device before sending the command ?
> > 
> > Can you take a look in deconz log (deconz / help) with filter "info" and 'info_l2" for lines starting by "Tuya debug", try to change manualy the setpoint, and using automation pls.
> 
> Hi, I tested setting the temp, here's the log:
> 
> directly sending put request to api (26c):
> ```
> 12:33:01:199 Tuya debug 4 : Address 0x847127FFFE190290 Payload 008102020004000000dc
> 12:33:01:200 Tuya debug 5 : Status: 0 Transid: 129 Dp: 514 (0x02,0x02) Fn: 0 Data 220
> ```
> 
> setting the temp in home assistant (22c):
> ```
> 12:33:11:699 Tuya debug 4 : Address 0x847127FFFE190290 Payload 008202020004000000e6
> 12:33:11:700 Tuya debug 5 : Status: 0 Transid: 130 Dp: 514 (0x02,0x02) Fn: 0 Data 230
> ```

And what do you mean with "waking up" the device?

Smanar, 2020-12-23:
You have not a line with "Send Tuya Request" when you use it with the API (direclty or with HA) ?

Data you receive for "data" are the value it s using for Thermostat heatsetpoint (0x02 0x02), so 22 C for the first one, and 23 C for the second one.

So it s realy not the value you are sending (26 and 22)

There is another issue ATM, some user need to use the device, to make it "awake" and after send a command to make it use it.
But for you the device answer so it react to request (but with bad values)

Can you make a try whith setting a value manualy to compare ?

WelpeLP, 2020-12-23:
> You have not a line with "Send Tuya Request" when you use it with the API (direclty or with HA) ?
>
> Data you receive for "data" are the value it s using for Thermostat heatsetpoint (0x02 0x02), so 22 [...]

I"m sorry but how should I set it manually?
And by the way, I didn't see any other Tuya messages in the log...

Smanar, 2020-12-23:
>I"m sorry but how should I set it manually?

Not possible using directly the device itself ?

>And by the way, I didn't see any other Tuya messages in the log...

when you make the command using HA, you need to have something starting by "Send Tuya Request"

WelpeLP, 2020-12-23:
> >I"m sorry but how should I set it manually?
> 
> Not possible using directly the device itself ?
> 

I did that during the recording of the log file, but it didn't show any events.

> >And by the way, I didn't see any other Tuya messages in the log...
> 
> when you make the command using HA, you need to have something starting by "Send Tuya Request"
> 
> 

Where do I do that? Could you tell me please? I'm new to deconz and home assistant, set them both up about a week ago...

P.S. Thanks for your support up to this point :)

Smanar, 2020-12-24:
You have already do it here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3989#issuecomment-750274431
Was perfect, you have used the good method.

For tuya, when they use the tuya cluster, absolutely all requests are logged, received request start by "Tuya debug" and send request start by "Send Tuya Request".
And as your device only have the tuya cluster, so I realy thing you have missed something, if you change the setpoint manualy, it s possible the device don't send request, but in this situation deconz can't change the value too.

And every time you make a request from deconz to the device (using HA or the API direclty), if it s a tuya request ofc, it will be logged for sure.

WelpeLP, 2020-12-25:
> You have already do it here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3989#issuecomment-750274431
> Was perfect, you have used the good method.
>
> For tuya, when they use the tuya [...]

I've checked again today, the device just sends the new set temp using a "tuya debug" request.
There never is a "send tuya request".
Do you have any idea left how to make this work?
And how can I send a "send tuya request" manually to the device with deconz?
Do I have to write C++ code for that?

Thanks in advance for your response :)

Smanar, 2020-12-25:
Yep Tuya use special format, so using the GUI for this cluster can't work exactly.

>And how can I send a "send tuya request" manually to the device with deconz?

Not possible, you need to send them using the gateway, so only using API (with or without third app) or GUI.

>I've checked again today, the device just sends the new set temp using a "tuya debug" request.

Can be the problem.

Edit:
Found something give me 10 mn.

Smanar, 2020-12-25:
Ok can you try the new code ?

Ttake a look here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only

So it will be for you

```
sudo apt install deconz-dev
git clone --branch revolt https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

WelpeLP, 2020-12-25:
> Ok can you try the new code ?
> 
> Take a look here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only
> 
> So it will be for you
> 
> ```
> sudo apt install deconz-dev
> git clone --branch revolt https://github.com/Smanar/deconz-rest-plugin.git
> cd deconz-rest-plugin
> qmake && make
> sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
> ```
> 
> The code is not compiled on my side, so I can be wrong, if you have error on compilation just tell me.

Hi, there's a compiler error:
```
database.cpp: In function ‘int sqliteLoadAllSensorsCallback(void*, int, char**, char**)’:
database.cpp:3554:27: error: base operand of ‘->’ has non-pointer type ‘Sensor’
                    (sensor->manufacturer().endsWith(QLatin1String("fvq6avy"))) || // Tuya
                           ^~
database.cpp:3566:27: error: base operand of ‘->’ has non-pointer type ‘Sensor’
                    (sensor->manufacturer().endsWith(QLatin1String("fvq6avy"))) || // Tuya
                           ^~
database.cpp:3577:27: error: base operand of ‘->’ has non-pointer type ‘Sensor’
                    (sensor->manufacturer().endsWith(QLatin1String("fvq6avy"))) || // Tuya
                           ^~
make[1]: *** [Makefile.Release:648: release/database.o] Error 1
make[1]: Leaving directory '/home/pi/deconz-rest-plugin'
make: *** [Makefile:40: release] Error 2

```

Smanar, 2020-12-25:
Corrected.

WelpeLP, 2020-12-25:
Nice, setting the temp works!
Nice to have would be that I can enable the "anti frost" mode and the child lock, maybe through the `"preset"`?
And now there's a new issue in Home Assistant: The temperature can be set but HA only gets the `"heatsetpoint"` when reloading the integration.

```
{
    "config": {
        "heatsetpoint": 2300,
        "locked": false,
        "offset": 0,
        "on": true,
        "preset": null,
        "reachable": false,
        "schedule": {},
        "schedule_on": false,
        "setvalve": null,
        "windowopen_set": true
    },
    "ep": 1,
    "etag": "bd7e64a9fc3231e30448c5756bb2bfe4",
    "lastseen": null,
    "manufacturername": "_TZE200_kfvq6avy",
    "modelid": "TS0601",
    "name": "Thermostat",
    "state": {
        "lastupdated": "none",
        "on": null,
        "temperature": 2450
    },
    "type": "ZHAThermostat",
    "uniqueid": "84:71:27:ff:fe:19:02:90-01-0201"
}
```

Smanar, 2020-12-25:
The child lock is here, it s the field "locked"
For the anti frost, I don't see it for your device. Can you enable it manualy on the device, and say me the return on deconz log ? (starting with "tuya debug")

For the 2 others problems :
- Can you do same when changing the "heatpoint" using HA ?
- I think it s because state/on = null, It reflect the valve position, but your device is a thermostat or a TRV ?

It seem it s a TRV https://zigbee.blakadder.com/Revolt_NX-4911.html

It don't go to "on" when the deive is working ?

WelpeLP, 2020-12-25:
Hi again,

Yes, the device is a TRV.
Here's the log for turning the child lock on/off (that doesn't work):
```
ON:
19:13:51:289 Tuya debug 4 : Address 0x847127FFFE190290 Payload 01380701000101
19:13:51:290 Tuya debug 5 : Status: 1 Transid: 56 Dp: 263 (0x01,0x07) Fn: 0 Data 1
```
```
OFF:
19:14:01:794 Tuya debug 4 : Address 0x847127FFFE190290 Payload 01390701000100
19:14:01:796 Tuya debug 5 : Status: 1 Transid: 57 Dp: 263 (0x01,0x07) Fn: 0 Data 0
```

The issue with HA fixed itself, but I suggest always returning `true` because you can't turn the device off (without taking the batteries out). No, it doesn't change the value.

When turning on/off anti frost mode, there's no log output. But I can emulate that by having a "virtual switch" in HA.

And by the way, the setting "window open" also does not work.

P.S. I really appreciate your work here, without your support I couldn't make it work.

Smanar, 2020-12-25:
For TRV it's, the valve state, so if the temperature is reached, the vanne is closed = "off".

>Status: 1 Transid: 57 Dp: 263 (0x01,0x07) Fn: 0 Data 0

It s the value I m using ATM, can you check again if in the JSON the value config/locked don't move ?

For windows open I m using 0x01 0x12 if you want to check again.

WelpeLP, 2020-12-25:
Window open now works. Don't know why.

The value `"on"` **can** be changed using a rest request.

Smanar, 2020-12-26:
And the "locked" one stll not working ?
And no, the value config/on can be changed, not the state/on.

Schedule is a long story ^^, better to rest the manual to check how it work on your device, it s realy device dependant.
Some device have a special preset to enable schedule mode, and with this mode you can set different schedule by days for the week.

But as you have just 
>"schedule": {},

Your device probably don't support it.

WelpeLP, 2020-12-27:
> And the "locked" one stll not working ?
> And no, the value config/on can be changed, not the state/on.
>
> Schedule is a long story ^^, better to rest the manual to check how it work on your device, it [...]

Hi again, 

will be at home again tomorrow,
I just want the child lock to work, then I'll close the issue. 
The anti frost mode is a seperate mode with temps from 5-15c, the normal manual mode is from 5-30c, so you can still set low temps in manual mode, that's clearly no issue if that's non functional.

Smanar, 2020-12-28:
For the "window open" some device can support some more params, like a timer, idk if your device too ?

WelpeLP, 2020-12-29:
> For the "window open" some device can support some more params, like a timer, idk if your device too ?

I don't know if it supports these parameters because you can't set the window mode on the device itself

Smanar, 2020-12-29:
Param are :
- Valve state
- Timer
- Temperature limit

You can take a look in the documentation if you see them, or in log, devices use 0x00 0x68.

But tell me for the "locked" command.

WelpeLP, 2020-12-29:
> Param are :
> - Valve state
> - Timer
> - Temperature limit
> 
> You can take a look in the documentation if you see them, or in log, devices use 0x00 0x68.
> 
> But tell me for the "locked" command.

Can you tell me which documentation you mean and how I can get the `"locked"` address?

Smanar, 2020-12-30:
The device documentation, in the box (if there is one)

For locked you already have it.
```
{
    "config": {
        "heatsetpoint": 2300,
        "locked": false,
        "offset": 0,
        "on": true,
        "preset": null,
        "reachable": false,
        "schedule": {},
        "schedule_on": false,
        "setvalve": null,
        "windowopen_set": true
    },
```

Just set the field "locked" to true, using third app or direclty the API, but I m almost sure you have in visible in HA.

AxS73, 2021-01-28:
Any progress on this? I also have a Revolt NX-4911-675 with the TS0601 modelid which is not recognized as a thermostat.
I'm on the official version 2.09.01 and would like to access it with Home Assistant.

Smanar, 2021-01-28:
Yep, I think you have it here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4087
If you can make a try ?

GR3ee3N, 2021-02-05:
@Smanar Hi , I have the NX-4911-675 (TS0601) as well. I am running #4087 right now and the thermostat is recognized as such but controlling it through Home Assistant is extremely glitchy. When setting the temperature, sometimes the thermostat does update, sometimes not, there are also constellations where the thermostat shows one set temperature but Home Assistant shows a different one. For Modes there is only 'Off' and 'Heating' available in HA which cannot be changed or at least there is no feedback that it was changed.
I am not sure if this might be an issue the the device itself or not.

Smanar, 2021-02-05:
You are using the last version ? it have a modification just 2 days ago.
But 2 things
- this device is battery powered, so slow, you probably need to wait some time for a setting change, you can take a look on logs to be sure, all tuya request are logged, and easy to find (all starting by "tuya debug").
- this device don't use "mode" but "preset", so it s not the same working mode in HA.

WelpeLP, 2021-02-05:
> Hello, you are using the last version ? it have a modification just 2 days ago.
> But 2 things
> - this device is battery powered, so slow, you probably need to wait some time for a setting change, you can take a look on logs to be sure, all tuya request are logged, and easy to find (all starting by "tuya debug").
> - this device don't use "mode" but "preset", so it s not the same working mode in HA.

Hi, I can confirm, sometimes the devices take up to a minute to report the correct temperature, you just have to be patient.

And a different question, will this be added to the master beta branch and if so, when?

Thanks for your support, the thermostats work almost flawlessly since about two months 👍

WelpeLP, 2021-02-05:
> @Smanar Hi , I have the NX-4911-675 (TS0601) as well. I am running #4087 right now and the thermostat is recognized as such but controlling it through Home Assistant is extremely glitchy. When [...]

It's true, HA thinks, the thermostat is off all the time, but setting the temperature works fine.
Btw, you can just ignore the "off" state, as these cannot be turned off. The closest thing to that is setting it to 5c, it's the antifrost feature.

Smanar, 2021-02-05:
Yep, just need to wait some time, but I think HA devs will soon manage too the preset field.

Not possible to be sure for the roadmap, but if all is fine it will be merged for the stable this WE, or in worst situation on the next beta just few days after

closed 2021-02-08 by manup