dresden-elektronik/deconz-rest-plugin #3991: Philips Hue app tries to update bridge software

ThomasDaheim, 2020-12-21:
## Describe the question or issue you are having
When I try to connect the Philips Hue app on android it finds the ConBee II stick but then tries to install an update of the bridge software - which fails (of course). Since this step can't be skipped in the Philips app I can't use it to control my hue lights :-(

## Screenshots

## Environment
- Host system: Synology NAS (modified to open port 80)
- Running method: Marthoc Docker container
- Firmware version: 26680700
- deCONZ version: 2.08.00 / 12/8/2020
- Device: ConBee II
- Do you use an USB extension cable: yes
- Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

## Additional context

Mimiix, 2020-12-21:
As far as I know, this is something we can't fix. Hue changed it on their end.

ThomasDaheim, 2020-12-21:
I would be willing to spend some time to dig deeper into this - if you could give me any starting point. I would really love to use the ConBee II with the Philips Hue app...

KodeCR, 2020-12-27:
The issue is deCONZ doesn't report the latest bridge version (1941132070) to the Hue app, so it wants to update and blocks until you do. Once https://github.com/dresden-elektronik/deconz-rest-plugin/pull/3696 is merged, this should be resolved.

closed 2021-02-08 by Mimiix