dresden-elektronik/deconz-rest-plugin #3994: Sensors continue to work but actuator control stops working spontaneously from time to time

tslivnik, 2020-12-21:
## Describe the bug

From time to time, deconz becomes unable to control actuators, while it continues to read from the sensors. For example, I have a SmartThings smart plug which is both a sensor (voltage, current, power) and actuator (on/off). The sensors continue updating, but control of the actuators on the same device (and all devices, e.g. Sonoff Zigbee relays) stops working. OpenHAB with the API key continues to show regular sensor updates. Restarting deconz makes no difference. Rebooting the system appears to cure the problem.

## Steps to reproduce the behavior

1) Install Debian Linux, deconz and OpenHAB.
2) Add the various sensors and actuators to deconz, then to OpenHAB.
3) Run it for a while.
4) Eventually (after several days or maybe 1-2 weeks), actuators stop responding to commands, while sensors continue to report data.

## Expected behavior

I would expect to be able to continue to control the actuators.

## Screenshots

I have no useful screen shots, the server is running headless.

## Environment

Version: 2.05.84 / 9/14/2020
Firmware: 26660700

 - Host system: Debian 10.5 running inside VirtualBox 6.1 on Debian 10.5
 - Running method: Virtual Machine
 - Firmware version: (26660700)
 - deCONZ version: (2.05.84)
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: Which? Yes, Aeotec Z-Wave Stick 5.

## deCONZ Logs

I ran deconz with all logging flags enabled and obtained an enormous log too big to upload, but happy to provide extracts or re-run it with limited debugging flags enabled and provide the whole such log if you tell me what flags to enable.

## Additional context

Zigbee devices on the network:
- Aqara temperature/humidity/pressure sensors (all continue to work) - about 10 of them
- Sonoff Zigbee relays - 2x
- Samsung SmartThings smart plug - 1x

J-N-K, 2020-12-22:
Is there anything in the openHAB log when this happens? On TRACE level you should see the request openHAB sends on the command and also immediately afterwards the response from the REST API plug-in. This could give some insight what‘s going on.

Smanar, 2020-12-22:
>Restarting deconz makes no difference. Rebooting the system appears to cure the problem

You mean restarting deconz change nothing but restarting deconz + openHAB do something ?
Have you try just with unplug / replug USB device ?

tslivnik, 2020-12-22:
> > Restarting deconz makes no difference. Rebooting the system appears to cure the problem
> 
> You mean restarting deconz change nothing but restarting deconz + openHAB do something ? Have you tried using phoscon if the device is visible on it ?
> Have you try just with unplug / replug USB device ?

Restarting /usr/bin/deCONZ does not cure the problem, but rebooting the virtual machine seems to. I don't think restarting openHAB will make any difference, since controls also don't work if I access the Phoscon gateway via http and try to switch the smart plug on and off from there. I have not tried unplugging and replugging the USB device. I'd like to understand what the problem is and solve it. Rebooting is not a solution, and unplugging hardware is even worse, since the system needs to continue working and controlling heating etc. when no one has physical access to the hardware (e.g. when travelling).

tslivnik, 2020-12-22:
> Is there anything in the openHAB log when this happens? On TRACE level you should see the request openHAB sends on the command and also immediately afterwards the response from the REST API plug-in. This could give some insight what‘s going on.
> 
> Edit: Which version of the openHAB binding are you using?

There are no error messages or anything unusual at the relevant time in /var/log/openhab2/openhab.log .

I cannot find anything unusual in /var/log/openhab2/events.log*, but those logs are very large and I'm not sure what I'm looking for.

 /var/log/openhab2/audit.log is empty.

I have deconz binding version 2.5.9 in OpenHAB.

Thanks.

Smanar, 2020-12-22:
No, It's just because I was not sure wheat mean "restarting" for you. And in some situation a reboot can restart the conbee and on other not (depend on USB powering), but if you just restart the virtual machine, you are right for me, nothing to see with the hardware.

J-N-K, 2020-12-22:
So you say, as long as you can control the actuator from openHAB, you can also control it from the Phoscon Software and as soon as the control from openHAB stops working, you can‘t control from Phoscon either?

And then after rebooting the deconz container, openHAB reconnects and it starts working again? In that case I doubt that it has anything to do with openHAB.

tslivnik, 2020-12-22:
> So you say, as long as you can control the actuator from openHAB, you can also control it from the Phoscon Software and as soon as the control from openHAB stops working, you can‘t control from [...]

So, the truth is, I only ever use openHAB to control the actuators. When this fails, I log into Phoscon, and then at that point, invariably I find that I also cannot control it from Phoscon. After I reboot the VM, it all seems to work again - openHAB can control the actuators, and if I try at that point, I can also do so from Phoscon. But then I do not use Phoscon any more, so I cannot claim that whenever I can control the actuators from openHAB, I can also do so from Phoscon - although I believe that most likely this is true.

Also, I should add, I have only had this problem occur twice. The first time, I was trying to debug the issue and rebooted the VM (actually, crashed it) accidentally, and found that everything was working fine after reboot. The second time is now. I don't want to reboot the VM while I am in this error state so I can try to identify what the problem is. I do not know how to reproduce the problem on purpose: it just happens after several days of running. I have tried restarting deconz, and the problem persists. I have not tried rebooting a second time. It is theoretically possible that the first time I was just lucky and reboot cured the issue, though if I had to bet, I'd say the reboot will cure the issue again, but then I won't be in a position to try to debug the issue while the problem is persisting. I'm happy to run tests or collect logs while the system is in this state if that will help us track down the problem.

tslivnik, 2020-12-28:
I would be grateful for any guidance as to what information to extract from the logs and provide to you, or what tests to run. I have not rebooted the system since it started exhibited this behaviour, which means my home automation has been down for the past 7 days. Tomorrow, I will reboot, at which point I will probably not be able to run any tests any more as the system will likely work normally again (until the problem reappears), although I will preserve the logs. Thanks. Happy New Year.

tslivnik, 2021-01-10:
I rebooted my system, which cleared the problem. For now, the solution I have adopted is to run `/sbin/shutdown -r` now via the crontab twice a week. We'll see if this solves the problem permanently. It's not a good solution though. There clearly remains a bug in deconz.

tslivnik, 2021-03-05:
The issue may be stale but it is not solved.

Mimiix, 2021-03-05:
I'll Re-Open for now.

@J-N-K  @Smanar Can you provide some guidance?

J-N-K, 2021-03-05:
@Mimiix As I said before: if both (openHAB and Phoscon) either work or don’t work, I doubt the issue is in openHAB or Phoscon but in the common part.

Probably the REST plugin is not able to control the ZigBee network at that time and therefore both fail. Another indicator for that is, that rebooting the deconz container brings both back online. I have next-to-no knowledge how the rest-API-Plugin works internally, so I can‘t help much here. 

Maybe it‘s not a deconz issue at all but the virtual machine fails to provide USB communication. I have seen a onewire USB adapter (DS9490R) failing in a VirtualBox  VM from time to time and never figured out why. It was still listed as connected but unresponsive. It was fixed by a reboot. I guess that an unresponsive ConBee stick would be similar and could create the described behavior. But this is just guessing.

Mimiix, 2021-03-05:
@J-N-K  This gives me some pointers!  Let's check on a zigbee level. 

@tslivnik I'd like to have some logs on whats going on. 

To enable logging. Open deCONZ and click Help. After that, click Debug View. The following debug levels need to be enabled for proper logging: INFO, ERROR,ERROR_L2,APS,APS_L2.

tslivnik, 2021-04-03:
The issue has not been solved. The issue seems to occur randomly after about a week of running time. When I enable all the logging flags (I ran `/usr/bin/deCONZ -platform minimal --http-port-80 --dbg-info=2 --dbg-error=2 --dbg-aps=2 --dbg-ota=1 --dbg-prot=2 --dbg-wire=1 --dbg-zdp=1 --dbg-zcl=1 --dbg-zcldb=1 --dbg-http=1 --dbg-tlink=1 >~/deconz-log 2>&1 &`, the logs grow extremely rapidly and would fill the virtual disk in my virtual machine before the problem would be triggered). I work around the issue by rebooting my VM every night at 4am. This mostly avoids the issue, but from time to time, the actuators still stop working. I haven't checked why, I just restarted the VM every time it happened. Basically I've got so many other priorities I have stopped looking at using deCONZ as a part of my home automation solution because of this problem. I'd be happy to run the system with logs enabled, but I need to limit the amount of logging options as logging everything produces too logs which are too large. Which options in the above command line can I take out? Thanks.

Mimiix, 2021-04-04:
I've only asked for 3 parameters..

Besides, if you feel deconz it at fault, but don't want to help to fix it : I can't do anything.

tslivnik, 2021-04-04:
Which three? dbg-info? dbg-error? dbg-aps? dbg-ota? dbg-prot? dbg-wire? dbg-zdp? dbp-zcl? dbg-zcldb? dbg-http? dbg-tlink? You haven't specified what you want anywhere. I have asked which ones and you answered not with information but with a snide comment. Deconz is certainly at fault because everything else is working in my home automation VM. If all you know to do is accuse people who report bugs of "not wanting to help", I am happy to stay where I am, which is having concluded that DeConz is not fit for purpose and that it is not properly supported. I reported the bug 4 months ago and I've only been ignored.

Smanar, 2021-04-04:
Can you make the try on a real machine, with same database, without logging ?

tslivnik, 2021-04-04:
Sorry, I don't have any suitable available real machine. I haven't tried this for a while (I've given up on using deconz/ConBee II, and switched everything to Z-Wave for now and plan to buy a different Zigbee USB stick when I have the time to research the market, which isn't now), but looking at my own comments from earlier on, it does seem that Phoscon also is unable to control the actuators when this happens. I don't remember any error message, I believe (if I recall correctly) the actuators, from within OpenHAB, accept the command and move to the new state, but the physical actuators don't do anything, and the state then flips back to the original state. But this may be wrong - I can take out the 4am reboot and wait for the problem to re-appear (probably after about 1 week) and try it again. But since it takes a while to trigger the issue, I might as well set up the logging before I do that as well - as long as I know what logging options to use and the logs don't fill up my disk over the space of a week.

Smanar, 2021-04-04:
If you can use the GUI (using OS with desktop, VNC, X-fowarding, or other) you will be able to read log on demand.

tslivnik, 2021-04-04:
Sorry I just have a headless VM which I access via ssh.

Mimiix, 2021-04-04:
> Which three? dbg-info? dbg-error? dbg-aps? dbg-ota? dbg-prot? dbg-wire? dbg-zdp? dbp-zcl? dbg-zcldb? dbg-http? dbg-tlink? You haven't specified what you want anywhere. I have asked which ones and [...]

To enable logging. Open deCONZ and click Help. After that, click Debug View. The following debug levels need to be enabled for proper logging: INFO, ERROR,ERROR_L2,APS,APS_L2.

Which translates to info2, error 2 and aps 2...

tslivnik, 2021-04-04:
Ok, thanks, I've just started deconz with those debugging options, will let you know what happens. It may take a while for the problem to materialize.

Smanar, 2021-04-05:
Will not be easier to install a second deconz installation on a full os with desktop time for test.
Or using "magic" like X-fowarding to access it on your headless server ?

tslivnik, 2021-04-05:
I don't think so, I'm not aware of magic which can forward X11 from a server which does not have X11 installed.

tslivnik, 2021-04-11:
Ok after 6 days, actuator control stopped working again. However, this time, the Zigbee sensors also stopped updating. However, I have explained why this is.

I logged into the VM running deconz and the deconz process is no longer running. There is nothing obvious in the deconz log file itself to tell me why the process would have crashed. The last few lines are:
```08:38:28:796 Idle timer triggered
08:38:28:796 Force read attributes for ZHATemperature SensorNode Environmental Sensor-Bar Room
08:38:28:796 don't create binding for attribute reporting of sensor Environmental Sensor-Bar Room
08:38:28:796 Force binding of attribute reporting for node Environmental Sensor-Bar Room
08:38:28:804 Master: read param with arg 0x19
08:38:28:818 Current channel 15
08:38:28:826 CTRL got nwk update id 0
08:38:28:832 Device TTL 3059 s flags: 0x7
08:38:28:837 Outgoing frame counter 22217858 (0x01530482)
08:38:29:173 Poll APS request to 0x00124B001E728842 cluster: 0x0006 dropped, values are fresh enough
```
However, `/var/log/messages` is more revealing. Ignore the "homeassistant" hostname: the VM used to run HomeAssistant, so I named the host "homeassistant", however I wiped out the hard drive and installed OpenHAB and HomeAssistant was never installed on this hard disk. I've redacted the ConBee II serial number.
```Apr 11 00:00:12 homeassistant rsyslogd:  [origin software="rsyslogd" swVersion="8.1901.0" x-pid="273" x-info="https://www.rsyslog.com"] rsyslogd was HUPed
Apr 11 08:38:29 homeassistant kernel: [621451.406074] QNetworkAccessM[27230]: segfault at 0 ip 00007f83f1be9b84 sp 00007f83e75148c8 error 4 in libQt5Network.so.5.11.3[7f83f1b8c000+113000]
Apr 11 08:38:29 homeassistant kernel: [621451.406089] Code: 1c 00 00 00 00 48 8b 7b 10 eb df 0f 1f 00 c7 47 1c 00 00 00 00 c6 47 50 00 c3 5b c3 66 2e 0f 1f 84 00 00 00 00 00 48 8b 7f 78 <48> 8b 07 ff 60 20 66 0f 1f 44 00 00 53 48 89 fb bf 18 00 00 00 e8
Apr 11 09:29:28 homeassistant kernel: [624510.502907] usb 1-3: USB disconnect, device number 4
Apr 11 09:29:29 homeassistant kernel: [624511.297264] usb 1-3: new full-speed USB device number 5 using xhci_hcd
Apr 11 09:29:29 homeassistant kernel: [624511.598648] usb 1-3: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
Apr 11 09:29:29 homeassistant kernel: [624511.598651] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Apr 11 09:29:29 homeassistant kernel: [624511.598651] usb 1-3: Product: ConBee II
Apr 11 09:29:29 homeassistant kernel: [624511.598652] usb 1-3: Manufacturer: dresden elektronik ingenieurtechnik GmbH
Apr 11 09:29:29 homeassistant kernel: [624511.598653] usb 1-3: SerialNumber: DExxxxxxx
Apr 11 09:29:29 homeassistant kernel: [624511.600942] cdc_acm 1-3:1.0: ttyACM1: USB ACM device
Apr 11 09:29:32 homeassistant kernel: [624514.309534] usb 1-3: USB disconnect, device number 5
Apr 11 09:29:33 homeassistant kernel: [624515.117148] usb 1-3: new full-speed USB device number 6 using xhci_hcd
Apr 11 09:29:33 homeassistant kernel: [624515.418425] usb 1-3: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
Apr 11 09:29:33 homeassistant kernel: [624515.418430] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Apr 11 09:29:33 homeassistant kernel: [624515.418432] usb 1-3: Product: ConBee II
Apr 11 09:29:33 homeassistant kernel: [624515.418434] usb 1-3: Manufacturer: dresden elektronik ingenieurtechnik GmbH
Apr 11 09:29:33 homeassistant kernel: [624515.418436] usb 1-3: SerialNumber: DExxxxxxx
Apr 11 09:29:33 homeassistant kernel: [624515.428547] cdc_acm 1-3:1.0: ttyACM1: USB ACM device
Apr 11 10:29:40 homeassistant kernel: [628122.599500] usb 1-3: USB disconnect, device number 6
Apr 11 10:29:41 homeassistant kernel: [628123.440325] usb 1-3: new full-speed USB device number 7 using xhci_hcd
Apr 11 10:29:41 homeassistant kernel: [628123.743031] usb 1-3: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
Apr 11 10:29:41 homeassistant kernel: [628123.743035] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Apr 11 10:29:41 homeassistant kernel: [628123.743036] usb 1-3: Product: ConBee II
Apr 11 10:29:41 homeassistant kernel: [628123.743037] usb 1-3: Manufacturer: dresden elektronik ingenieurtechnik GmbH
Apr 11 10:29:41 homeassistant kernel: [628123.743038] usb 1-3: SerialNumber: DExxxxxxx
Apr 11 10:29:41 homeassistant kernel: [628123.751699] cdc_acm 1-3:1.0: ttyACM1: USB ACM device
Apr 11 10:29:44 homeassistant kernel: [628126.436935] usb 1-3: USB disconnect, device number 7
Apr 11 10:29:45 homeassistant kernel: [628127.300243] usb 1-3: new full-speed USB device number 8 using xhci_hcd
Apr 11 10:29:45 homeassistant kernel: [628127.601273] usb 1-3: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
Apr 11 10:29:45 homeassistant kernel: [628127.601276] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Apr 11 10:29:45 homeassistant kernel: [628127.601277] usb 1-3: Product: ConBee II
Apr 11 10:29:45 homeassistant kernel: [628127.601278] usb 1-3: Manufacturer: dresden elektronik ingenieurtechnik GmbH
Apr 11 10:29:45 homeassistant kernel: [628127.601279] usb 1-3: SerialNumber: DExxxxxxx
Apr 11 10:29:45 homeassistant kernel: [628127.604996] cdc_acm 1-3:1.0: ttyACM1: USB ACM device
Apr 11 11:29:44 homeassistant kernel: [631726.538851] usb 1-3: USB disconnect, device number 8
Apr 11 11:29:45 homeassistant kernel: [631727.348671] usb 1-3: new full-speed USB device number 9 using xhci_hcd
[... cut ...]
The reason, presumably, that the sensors also stopped working on this occasion, whereas normally they resume working, is that on this occasion, I stopped the `deconz` service and ran deconz manually. Had the service been running, it would have automatically restarted `deconz` after it crashed. However, this doesn't tell me why only the sensors start working again after such a restart, but not the actuators. With the ConBee II disconnecting and reconnecting, I guess it's surprising anything at all works, however I don't think these "disconnects" are physical disconnects, but something to do with the segfault which suggests some kind of a driver problem to me.

tslivnik, 2021-04-18:
Any ideas, or anything else I should look for in the log files and provide here?

tslivnik, 2021-05-08:
The problem has reoccurred, but this time, `deconz` has not crashed, and there is no `QNetworkAccessM` segfault line in `/var/log/messages`. This time, sensors are continuing to provide readings, but actuators stopped working again. The log file is now 3GB in size and continuing to update, but I can't find anything obviously wrong in it. As before, a Zigbee smart plug is providing its power consumption reading, but sending it ON or OFF commands has stopped working.

tslivnik, 2021-05-09:
If you have any suggestions or any experiments I can run which might provide useful information, can you please let me know within the next 48 hours, after which I will restart deconz and/or the VM again, and probably get another USB stick because after 5 months, I've made no progress on this problem, the ConBee II / deconz is the weak link my home automation setup which is stopping me building an actual usable home automation system and I think it's time to switch to another Zigbee solution.

tslivnik, 2021-05-14:
Ok, I'm restarting deconz and deleting the log file because it's taken over my disk. It seems that this software is not being maintained, so I'm going to buy and try another Zigbee stick which uses OpenHAB's Zigbee binding rather than Deconz.

Mimiix, 2021-05-14:
Asked @manup  to check in :)

tslivnik, 2021-06-05:
Not really sure what I can do to keep the issue alive. It is not resolved, however.

tslivnik, 2021-06-29:
The robot may think it stale, but the bug continues to exist.

manup, 2021-06-29:
Sorry for the long delay, which deCONZ and firmware version are you currently using?

I have the Sonoff relay here and will add it to my test network for observation.
Unfortunately I don't have the Samsung SmartThings smart plug.

tslivnik, 2021-06-29:
Firmware 26660700. Phoscon says the firmware is up to date. deCONZ version 2.05.84-debian-buster-stable.

manup, 2021-06-29:
Ok, there have been quite some improvements in recent beta versions which might come into play here.
I'd suggest to either install the latest beta version [v2.12.1-beta](https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/v2.12.1-beta) — or wait for the next stable release.

Not sure if the firmware version is too important in this case, there are newer versions with various fixes as shown in the [Firmware&nbsp;Changelog](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Firmware-Changelog). New firmware versions can be installed manually as described in [Update deCONZ Manually](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually).

tslivnik, 2021-07-27:
Thanks. When do you think there will be a new stable release? I've currently installed deconz from the deconz apt repository using `apt`. If the wait is not too long, it'll be easier to wait for the new release rather than mix and match the apt installed packages and hand compiled ones, which might work or it might create a mess, especially if installing beta versions.

tslivnik, 2021-08-18:
I have updated deconz to 2.11.05 and now nothing works at all. To be honest, I've lost interest in deconz, this is just to say that this problem still has not been fixed. Obviously, this project is nowhere near deployment ready.

Mimiix, 2021-08-18:
@tslivnik I think that's a bit short to say. There's over 100k installations running without issues. 

Either way, I think i already see the issue here: Virtualbox.  Did you try any other system?

tslivnik, 2021-08-18:
It's a deconz issue, not a Virtualbox issue - Virtualbox works fine with my Z-Wave USB stick and all the other USB devices.

Mimiix, 2021-08-18:
I've seen more issues with VirtualBox then with any other platform. Either way, i just have 2 more suggestions:
- Install latest Conbee II firmware (Beta) Which has a lot of improvements on USB Serial Connections.
- Try the newest beta (or stable when it comes out later this week).

In addition, reading trough the topic a timeline:
- You reported issues with Actuators only (Zigbee connectivity is fine, as sensors continue to work, Indicates an issue with Software/deCONZ)
 - There's been requests for full logs, I haven't received any of them.
 - Suddenly there's USB connection issues(Which causes the Sensors to stop reporting/working too).  Funny thing: After 6 days you mention it. Before that, nothing has changed.  

We still haven't received any full logs of deCONZ with the flags i've asked:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3994#issuecomment-813078885

So please, provide those and perhaps we can check there.

tslivnik, 2021-08-18:
I have already updated the Conbee II firmware to the latest version.

Yes the issues are still only with the actuators. But now the actuators do not work at all (before the upgrade, they would work for about a week before breaking).

I've already advised that the full logs are several GB in size. I cannot upload several GB worth of logs.

I asked to be told what to grep for in the logs and I could upload the result of those searches. But I received no response.

Mimiix, 2021-08-18:
So they are on version http://deconz.dresden-elektronik.de/deconz-firmware/beta/deCONZ_RaspBeeII_0x266f0700.bin.GCF ?
That's the beta im talking about :)   Are you on that one? 
 
check the logs for errorcodes as mentioned on the wiki.

Mimiix, 2021-08-18:
Also, another thing, Are those ZB minis from sonoff?

tslivnik, 2021-08-18:
> So they are on version : RemovedURL
> Sorry: http://deconz.dresden-elektronik.de/deconz-firmware/beta/deCONZ_ConBeeII_0x26700700.bin.GCF
> 
> That's the beta im talking about :) Are you on that one?
> 
> check the logs for errorcodes as mentioned on the wiki.

I am on the beta you mentioned originally (266f) not the one in your updated post (2670) which only came out a few days ago.

I updated to 266f very recently (less than 3 weeks ago - I think 2 weeks ago), at the same time I upgraded to deconz 2.11.05, after I was advised by this forum to do so. It did not improve anything, indeed, the result of both those updates was to make actuators not work at all, rather than just stop working after a week. So 266f is all broken but 2670 is all working?

Grep does not find any of the Zigbee error codes mentioned in the Wiki (E1, E9, A7, D0) in the log.

I uploaded an extract of /var/log/messages on 11 April which pointed out at least one problem, though it seems to be a different one from the one that happens usually.

Mimiix, 2021-08-18:
> So 266f is all broken but 2670 is all working?

The latest beta (2670) contains a lot of improvements on serial issues. Perhaps it fixes yours, but i stand by my initial point: VB is not stable with USB stuff.  A lot of users mention stabbility issues with it , comparable with what you are experiencing. 

> Grep does not find any of the Zigbee error codes mentioned in the Wiki (E1, E9, A7, D0) in the log.

Good. 

> I uploaded an extract of /var/log/messages on 11 April which pointed out at least one problem, though it seems to be a different one from the one that happens usually.

Which was you running on an very old version of deCONZ right? Make sure your at least on the latest stable.

For now(forward):
- Update to the 2670 firmware
- Make sure you are on the latest stable or (even better) beta.
- Provide some logs of that running(see how that goes).
- Try and test to move away from VBox and see if that brings joy.

tslivnik, 2021-08-18:
I can upgrade to 2670 firmware. An environment other than VirtualBox is not an option. All my other USB devices including the USB Z-Wave stick, work fine, so frankly, blaming VirtualBox for the issue isn't very impressive.

Mimiix, 2021-08-18:
> I can upgrade to 2670 firmware. An environment other than VirtualBox is not an option. All my other USB devices including the USB Z-Wave stick, work fine, so frankly, blaming VirtualBox for the issue isn't very impressive.

I am just trying to help and point out solutions and see what works, not blaming anything. But hey, what do i know.

Either way: Please follow what i've suggested. If you wish not to follow them,which is fine, i don't feel the need to keep this issue open.

tslivnik, 2021-08-18:
Thanks.

I posted information about this bug 8 months ago.
I followed up several times.
Most if my posts were ignored.
Some were not. Those that made suggestions, I followed - I upgraded deconz, I upgraded the firmware, I posted logs, etc. etc.
The bug remains unfixed.

The platform I have is VirtualBox running on a server with redundant power supplies, redundant UPS's, ECC RAM and a redundant ZFS filesystem.

a) I do not have another platform available. I have an old Raspberry Pi, but it's not a platform reliable enough to run my building.
b) I interested running the system which will run the whole building on dinky unreliable hardware. I'm not prepared to buy new hardware which is unsuitable for deployment just to try and see if that might work (when the issue I believe is with the software) nor am I prepared to buy and set up a whole extra redundant server just to see if that will work - or even to set up a whole extra such a server just to run my building automation even if I was sure that would work.

c) A virtual machine, which can run either in VirtualBox, or otherwise concurrently with VirtualBox, is the system this has to run on. Everything else runs in such a virtual machine just fine, including USB subsystems like a Z-Wave stick and its drivers.

senilio, 2021-08-18:
You are not the first person to report issues with Virtualbox, and especially unstable USB passthrough. If this is an issue with Virtualbox or with Conbee, I will leave unsaid. But unless you are willing to put in the effort of ruling out known Virtualbox issues, this ticket no longer serves a purpose.

manup, 2021-08-18:
A note on the 0x26700700 firmware: This version is more robust for VMs or Docker setups as it's not so picky about the host application starting and running quickly enough. I can't tell if that is the problem here though.

But the problem visible in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3994#issuecomment-817289611 is addressed in this version.

Can you also please share a screenshot of the VM Settings for the USB port of the ConBee.

TheNON75, 2021-08-18:
Hi @tslivnik ,

I am really sorry that so many time is spent already with the issue by you and the others, I have no doubts it has left some bad taste in the mouth, unnecessarily.
Let’s try to move back to the right track, if you agree. 

Your machine is very impressive, also your idea is superb and very welcome by myself too. 
Despite all of the above, I recommend to try one of the official methods for installing and running deconz in the first rounds.
It can either work from the very first moment without issue or have the same/similar/other symptoms. 

Nevertheless, these are called official methods with purpose…. And that is: these are tested, supported and so, logically the conclusion is: anything else is at your own risk and if possible, a very best effort is provided by the experts. (Filling with incompatible fuel my car is not supported too, but i can try at my own risk, right?) I believe this is something that has to be accepted and understood by everyone.

Anyhow, once it is confirmed it works, you may continue tinkering on any not officially supported combinations, knowing that it has to work, just there is something to sort out.

I would recommend the docker approach as in one hand it is a kind of virtualization, it provides you the capability to access the deconz GUI also on headless, minimal systems with vnc. 

Does that sound reasonable and feasible by you?

tslivnik, 2021-08-20:
I installed the 2670 firmware. After reboot, the actuators started working and worked for 24 hours, then stopped working. It's an improvement on 266f firmware, which didn't work at all, but it's still a regression from my original firmware, which worked for approximately a week at a time.

I will look at a Docker installation, however, I can't do that in the short term for several reasons:

1) I have my entire setup with over 100 devices (only some of them Zigbee) operating the building at present. Everything else works fine, including other USB devices, only deconz/ConBee II/Zigbee is broken. Migrating may involve a lot of work and could break things.

2) I don't have enough knowledge and experience with Docker yet including with how to virtualize things like USB hardware, how secure it is etc. to deploy it to run the building. I will try to educate myself over time.

TheNON75, 2021-08-20:
Docker is a very easy approach and is in use by many enterprises.
Backing up the device database with the phoscon app and restoring it elsewhere is a well working method (you can avoid adding again everything etc)

please note, there is a one day old new stable release and also a new stable firmware.

TheNON75, 2021-08-24:
Hi @tslivnik, do you need any help or is everything on the track?

tslivnik, 2021-09-01:
Thank you. Currently, the Zigbee sensors are always working and the Zigbee actuators are never working. I.e. reflashing the firmware and upgrading deconz made things worse - from the actuators working for about a week at a time before stopping working to never working at all. I don't have the time currently to fully investigate using Docker, though my initial investigation is that this won't be a good solution in my case. I will also look into KVM and Xen, but neither of those is a short term project.

TheNON75, 2021-09-02:
Well, inak really sorry the things are not getting better in your environment.
As I understand you are still on a vm other than docker.
Don’t get me pushy, but I really recommend to try one of the official ways. 
feel free to join our discord to get some more interactive help on firing up a dedicated docker container for deconz. It takes really few minutes only.

tslivnik, 2021-09-02:
Sorry, I'm using OpenHAB to run a building, I'm not tinkering with this as a hobby and I don't have a lot of time to tinker with it. I am looking at Docker, Xen and KVM with the appropriate level of priority (low - I don't have the time, the VirtualBox setup works for all purposes other than deconz, it's doing useful things and trying to replace the entire system would cause disruption and take time which I don't have). On what I have seen so far, Docker will not meet my requirements, but Xen or KVM may, so I could try to switch to Xen or KVM as the virtualization platform on this server - but again this won't happen soon because, a) the server runs a lot of VMs, and it's not clear that migrating to a new platform would not break things, or would work better, b)  it would require me to migrate to a new system, which means downtime, disruption, and c) it will take up a lot of my time for not a lot of benefit. For now, since I'm not using a lot of Zigbee actuators and the Z-Wave USB stick is working fine, the simplest thing for me to do is to replace the ZigBee actuators with Z-Wave ones. I'd like to get Zigbee to work, but for now, my conclusion is that deconz does not work, at least not in my environment.

TheNON75, 2021-09-02:
I fully understand that you don’t want to harm your production environment, I wouldn’t do that either. My suggestion was only about deconz.
VirtualBox is confirmed to be good for many purposes, but also to have issues in certain cases.

Nevertheless, if you have a spare machine for a test environment for that purpose, it could also help in the investigations, as you said you are now switching to zwave, it could not be a big issue to miss zigbee devices ;)

Apart from the above, it is fully your decision how you would or wouldn’t like to proceed. 
“Things” can work this or that way without issues even if officially not stated as supported, but getting help on resolution in such a case is always very difficult. Here I am not talking only about deconz, but in general.

Should you decide to continue the investigations, we are here (and on discord) to help.
Please note that unless you continue keeping this topic open, it will be automatically closed after certain days or you/we can close it manually even there is no solution found. Just let us know.

regards
THN

tslivnik, 2021-09-02:
Thanks. Currently, I don't have a suitable spare computer - the only spare Linux machines I have are a couple of Raspberry Pi's, which themselves crash all the time. Maybe it's the power supply, although I've replaced the power supply with an "original" one twice already. I think it's just dinky consumer-grade hardware. But then it's cheap. All other available machines would only be able to run a Linux distribution in a VM. Also, I am using the Zigbee stick for the sensors which continue to work fine using deconz; only the Zigbee actuators do not work. I'd have to get another ConBee II stick. Which I can easily do, but at this point, getting another brand of Zigbee stick and trying to use the OpenHAB Zigbee binding seems like a better investment of time.

So in other words, I will continue to look at using Xen, KVM and maybe even Docker (though, like I said, from what I have gathered so far, I don't think Docker will work for me) at some point in the future. I suspect Xen may well prove to be a better solution all round than VirtualBox, for other reasons too. But I don't have the time to do any of this in the short term. In the short term, other solutions seem to be more promising and/or a better investment of time.

I will continue to monitor deconz, because if the bug is fixed and a simple "apt install" to upgrade to a new version can get my actuators to work, in the immediate future, that would be the simplest solution of all for me.

TheNON75, 2021-09-02:
Well, to sum it up:
- you’ve discovered an unexpected behavior when using deconz in a non supported environment
- You received some hints, questions, suggestions and have tried several attempts to sort it out without success
- There was sadly no possibility to try the things with one of the officially supported methods and so trying to get the things sorted out if that way the things are working well or there is a bug that can be confirmed and allows the deconz experts to sort out the root cause.
- we can confirm bugs which the official methods only

In case you agree, I am recommending to close the issue and if you can manage your time and technical possibilities to give a try to the official methods (fixing one of your rpi or else), please return to us and let us pick up the topic together again in a new thread (starting with a “blank sheet”).

I hope this sounds also reasonable and acceptable to you.

We look forward to hearing from you soon

@Mimiix, may you please close the thread?

[14 more replies]

closed 2021-09-02 by Mimiix