dresden-elektronik/deconz-rest-plugin #3997: Completely unreliable attribute reporting

aryelevin, 2020-12-22:
Hi Everyone,

After a long time trying to investigate what is the issue with my setup, I decided to open this issue in order to get some tips maybe (I hope) on how to improve or solve the issue.

The issue is that many attribute reporting is not reaching the Coordinator which is ConBee II with up to date firmware (26680700), thus the state os many sensors and lights is not up to date, for example: Aqara Door sensor is not updates many times and by itself is very close to the Coordinator, and also I have Danalock V3 which is not always updates its state, also have a ubisys D1 dimmers which also seems to not being always up to date state, so long story short: Attributes reporting in completely unreliable, and the mystery for me is that it works, but only about 7 of a 8 times.

My network consist:
Hue Bulbs (E14 candles and GU10), Plug-in sockets, Dimmer Switches and Tap
Tint (Muller Licht) E27 Bulbs (Both A19 and G80 sizes)
Lutron Aurora
Aqara Door and Window Sensor
Innr E27 Bulbs and Plugin socket
Develco Plugin Sockets and Smoke sensor
Heiman Smoke sensor, CO Sensor, Siren and Gas sensor
Mijia-Honeywell smoke sensor
LEDVANCE Z3 II RGBW E27 Bulbs (The new up to date version which is HA based not ZLL)
Schweiger E27 Bulbs (shown as LDS in the manufacturer name on the Basic cluster)
OWON AC201 thermostats and SLC603 (A rotary dimmer remote which I working on its integration to the rest-plugin)
GLEDOPTO RGB+CCT
Sunricher LED Dimmer
Ubisys D1, S1 and S2
EcoDim Rotary Dimmer
AduroSmart Plugin Socket and Dimmer module
A cheap Dimmer shown as HOMA1031 (Its a dimmer module)
A cheap sliding dimmer

My only suspect is that the last 2 devices in my list causing this issue, and couldn't prove this theory.

Any ideas?

P.S. The general commands (turn on/off etc) is also not the performs in the greatest performance it supposed to be...
In addition, it seems that if I restart my main electricity breaker (power cycling everything at once), it gets better for few hours and then it becomes worse again.
Another thing, I bought an Aqara M2 Hub and paired it another Aqara Door and Window sensor which works flawless!

Thanks a lot in advance

closed 2020-12-22 by close-issue-app[bot]

Mimiix, 2020-12-22:
In addition to the bot: Please attach logs in your new issue report(together with the filled template). That would help greatly.

To enable logging. Open deCONZ and click Help. After that, click Debug View. The following debug levels need to be enabled for proper logging: INFO, ERROR,ERROR_L2,APS,APS_L2.

aryelevin, 2020-12-22:
@Mimiix ,
OK, Opened a new issue and soon will update it with some logs

Mimiix, 2020-12-22:
Great! See you there 👍