dresden-elektronik/deconz-rest-plugin #3998: Completely unreliable attribute reporting

aryelevin, 2020-12-22:
## Describe the question or issue you are having
Hi Everyone,

After a long time trying to investigate what is the issue with my setup, I decided to open this issue in order to get some tips maybe (I hope) on how to improve or solve the issue.

The issue is that many attribute reporting is not reaching the Coordinator which is ConBee II with up to date firmware (26680700), thus the state os many sensors and lights is not up to date, for example: Aqara Door sensor is not updates many times and by itself is very close to the Coordinator, and also I have Danalock V3 which is not always updates its state, also have a ubisys D1 dimmers which also seems to not being always up to date state, so long story short: Attributes reporting in completely unreliable, and the mystery for me is that it works, but only about 7 of a 8 times.

My network consist:
Hue Bulbs (E14 candles and GU10), Plug-in sockets, Dimmer Switches and Tap
Tint (Muller Licht) E27 Bulbs (Both A19 and G80 sizes)
Lutron Aurora
Aqara Door and Window Sensor
Innr E27 Bulbs and Plugin socket
Develco Plugin Sockets and Smoke sensor
Heiman Smoke sensor, CO Sensor, Siren and Gas sensor
Mijia-Honeywell smoke sensor
LEDVANCE Z3 II RGBW E27 Bulbs (The new up to date version which is HA based not ZLL)
Schweiger E27 Bulbs (shown as LDS in the manufacturer name on the Basic cluster)
OWON AC201 thermostats and SLC603 (A rotary dimmer remote which I working on its integration to the rest-plugin)
GLEDOPTO RGB+CCT
Sunricher LED Dimmer
Ubisys D1, S1 and S2
EcoDim Rotary Dimmer
AduroSmart Plugin Socket and Dimmer module
A cheap Dimmer shown as HOMA1031 (Its a dimmer module)
A cheap sliding dimmer

My only suspect is that the last 2 devices in my list causing this issue, and couldn't prove this theory.

Any ideas?

P.S. The general commands (turn on/off etc) is also not the performs in the greatest performance it supposed to be...
In addition, it seems that if I restart my main electricity breaker (power cycling everything at once), it gets better for few hours and then it becomes worse again.
Another thing, I bought an Aqara M2 Hub and paired it another Aqara Door and Window sensor which works flawless!

Thanks a lot in advance

## Screenshots
NA

## Environment

 - Host system: PC
 - Running method: Ubuntu
 - Firmware version: 26680700
 - deCONZ version: 2.07.01
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? Yes, Aeotec Z-Stick

## deCONZ Logs
NA

## Additional context
See above

Mimiix, 2020-12-22:
Hi!

As i asked in your last issue, can you provide some logs?

To enable logging. Open deCONZ and click Help. After that, click Debug View. The following debug levels need to be enabled for proper logging: INFO, ERROR,ERROR_L2,APS,APS_L2.

aryelevin, 2020-12-22:
Yes, now capturing some logs and soon will upload it here...

aryelevin, 2020-12-22:
OK, attached a txt with some log, which includes a situation of the Aqara Door sensor is not updated to the correct value.

Thanks a lot in advance

Arye
[New text file.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5728847/New.text.file.txt)

Mimiix, 2020-12-22:
Hi!

I've spotted 2 issues:
11:04:13:513 0x00124B0019E75810 error APSDE-DATA.confirm: 0xA7 on task
11:05:35:302 0x001FEE0000004BD5 error APSDE-DATA.confirm: 0xE9 on task

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Zigbee-Error-Codes-in-the-Log

What devices are these?

aryelevin, 2020-12-22:
OK, so the 0x00124B0019E75810 Is the cheap slider dimmer, and the 0x001FEE0000004BD5 is Ubisys S1R unit.
Check the link you sent

aryelevin, 2020-12-22:
It seems that both errors related to a command which sent from the Coordinator which failed in the routing.
My issue is less about commands which not received (It also an issue but not that much in compare to the attributes reporting issues), but unreliable attributes reporting such as sensors states and lights state.

Mimiix, 2020-12-22:
They often fail when the network has weak connections. Are you able to provide a screen of deCONZ and see the links?

Another reason for a "slow" network could be the speed of the PC that hosts deconz. For example, Raspberry PI's with slow SD's show the same behaviour. What type of Hard drive are you using?

aryelevin, 2020-12-22:
Yes, attached the screenshot.
About the HW, I'm using the Intel Compute stick which seems to have utilizing only 20-30% of the CPU and have 25% of free RAM.
The storage device is the internal SSD (32GB).
[image]

Mimiix, 2020-12-22:
I'll foward it To Manuel. I'm out of options. He'll know how to fix it.

aryelevin, 2020-12-22:
@Mimiix Great, Thanks a lot for your help and support.
I'll wait for Manuel response.

Warm regards and stay safe

Mimiix, 2020-12-22:
Last thing i can advice: See what happens if you turn off the devices (remove them from hte power) that i mentioned earlier. Maybe  that changes a thing (Maybe they are broken?)

> Warm regards and stay safe

Happy holidays and #staysafe :)!

aryelevin, 2020-12-22:
In general, if I power cycle my main fuse breaker in the house then everything working much better for only for limited time.
I can try what you suggested, I know for Sure that the S1R is functioning properly in terms of turn off/on, but the other slider dimmer maybe is not working well for routings or something.
I will try to remove them from power for just in case it might help and will report here.

Thank a lot and of course, Happy holidays.

aryelevin, 2020-12-22:
OK, so after powering them off, no difference...
I'll wait for Manuel response...

Thanks a lot

aryelevin, 2020-12-22:
@Mimiix I see you reacted "Confused" emoji, In my response I just reported that your suggestion of disconnecting the error nodes from power didn't helped and the issue continues.
In addition I said I'm waiting for response from Manuel as you wrote he is the doctor for such issues and he'll have solution for this.
Last but not least I thanked you a lot 👍 

Again, happy holidays and stay safe

Mimiix, 2020-12-22:
Haha, no i did that because of the no difference :( I hoped it couldlve helped you. The smiley itself expressed my face on it, i don't think confused is the right word. either just a bit sad. 

Waiting on @manup it is :)

aryelevin, 2020-12-22:
Ah, OK.
So we're all good... Hahaha...

aryelevin, 2020-12-27:
@Mimiix How are you?
I see Manuel is not responding to this... yet...
I thought maybe its an issue with using Hue devices which is not support Attribute reporting as routers for attribute reports, what do you think? and in case this is the problem, how would you solve this? (I thought using another ConBee for Hue devices... not sure...)

manup, 2020-12-28:
Hard to reason what exactly is causing the lost commands, the log itself looks quite good.
The routers seem to have strong connections, but not the end-devices.

To proceed we need to look closer what parent nodes are selected by the problematic sensors/switches.
(Some of the end-devices in the screenshot show very weak links, for example the SLC603)

Can you please check to which nodes these connect and if the nodes are not to far away of the end-device?

aryelevin, 2020-12-28:
@manup Thanks for the reply,
OK, so the problem is super wide across the all house, the major of cases we see is the Hue dimmer buttons commands misses, sometimes we need to press the buttons 6-7 times before it happens to response on the lights (I'm using custom handling of the commands in my Homebridge-hue fork) and the Aqara door sensor...
As you can see most Hue switches (on the right side) have good signal so its not a signal issue most probably, in addition to that the house is full of routers, many bulbs and sockets.

The SLC is no being used now, I'm just investigating its behavior for future use, so this is not a good example...
Any other idea of what to check?

P.S. the Aqara door sensor is 60cm away from the ConBee II...

closed 2021-02-08 by Mimiix