dresden-elektronik/deconz-rest-plugin #3999: Cause of the known Rapsbee II USB 3.0 issue?

m0wlheld, 2020-12-22:
## Describe the question or issue you are having

As stated on https://phoscon.de/en/raspbee2/install#connection, there is known issue with Raspbee II module and USB 3.0 devices being nearby. I faced this issue when trying to setup a new Zigbee smart home hub using an RPI4B, Raspbee II and a USB 3.1 pen drive.

Question is: It that a general issue with Raspberry Pi 4B or is it related to Raspbee II only and can (hopefully) be resolved by Phoscon?

Thanks in advance!

## Screenshots

## Environment

 - Host system: Raspberry Pi 4B 4G
 - Running method: Raspbian, Home Assistant Core @ Docker, Home Assistant "native" ZHA integration, OSRAM Smart+ bulbs
 - Firmware version: 26690700
 - deCONZ version: (not used)
 - Device: RaspBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs

n/a
## Additional context

n/a

Mimiix, 2020-12-22:
Hi!

That's something we can't do much about:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Zigbee-Error-Codes-in-the-Log (check the 0xe1). 

That's a bug with the USB implementation. Every device that works with 2.4GHZ has issues with it unfortunately.

m0wlheld, 2020-12-22:
Okay, then. Going to use SD-card then. Thanks for the fast reply.

Angenehme Feiertage.

closed 2020-12-22 by m0wlheld

Mimiix, 2020-12-22:
@m0wlheld Fijne feestdagen!

ebaauw, 2020-12-22:
> It that a general issue with Raspberry Pi 4B or is it related to Raspbee II only and can (hopefully) be resolved by Phoscon?

It's a general issue with USB3 interfering with 2.4GHz radio, see https://www.intel.com/content/www/us/en/products/docs/io/universal-serial-bus/usb3-frequency-interference-paper.html.  Zigbee is more susceptible to interference than other 2.4GHz radio protocols (due to its low-power design), so you're bound to experience issues with Zigbee sooner than with Bluetooth or WiFi.