dresden-elektronik/deconz-rest-plugin #4001: Installation of Conbee II under Debian-based OpenMediaVault

tomillr, 2020-12-22:
## Describe the question or issue you are having
Trying to get a Conbee II stick setup on a Debian-based system running OpenMediaVault. Unable to access Phoscon webinterface via http://192.168.2.126/pwa/login.html.

Tried following the instruction for a headless installation using root over SSH:

```
root@homeserver:~# sudo gpasswd -a $USER dialout
Benutzer root wird zur Gruppe dialout hinzugefügt.

root@homeserver:~# wget -O - http://phoscon.de/apt/deconz.pub.key | \
>            sudo apt-key add -
--2020-12-22 10:38:42--  http://phoscon.de/apt/deconz.pub.key
Auflösen des Hostnamens phoscon.de (phoscon.de)… 144.76.96.194
Verbindungsaufbau zu phoscon.de (phoscon.de)|144.76.96.194|:80 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 200 OK
Länge: 1692 (1,7K) [application/octet-stream]
Wird in »STDOUT« gespeichert.

-                   100%[===================>]   1,65K  --.-KB/s    in 0s      

2020-12-22 10:38:42 (75,2 MB/s) - auf die Standardausgabe geschrieben [1692/1692]

OK

root@homeserver:~# sudo sh -c "echo 'deb [arch=amd64] http://phoscon.de/apt/deconz \
>             $(lsb_release -cs) main' > \
>             /etc/apt/sources.list.d/deconz.list"

root@homeserver:~# sudo apt update
Holen:1 file:/var/cache/openmediavault/archives  InRelease
Ign:1 file:/var/cache/openmediavault/archives  InRelease
Holen:2 file:/var/cache/openmediavault/archives  Release
Ign:2 file:/var/cache/openmediavault/archives  Release
Holen:3 file:/var/cache/openmediavault/archives  Packages
Ign:3 file:/var/cache/openmediavault/archives  Packages
Holen:4 file:/var/cache/openmediavault/archives  Translation-de
Ign:4 file:/var/cache/openmediavault/archives  Translation-de
Holen:5 file:/var/cache/openmediavault/archives  Translation-en
Ign:5 file:/var/cache/openmediavault/archives  Translation-en
Holen:6 file:/var/cache/openmediavault/archives  Translation-de_DE
Ign:6 file:/var/cache/openmediavault/archives  Translation-de_DE
Holen:3 file:/var/cache/openmediavault/archives  Packages
Ign:3 file:/var/cache/openmediavault/archives  Packages
Holen:4 file:/var/cache/openmediavault/archives  Translation-de
Ign:4 file:/var/cache/openmediavault/archives  Translation-de
Holen:5 file:/var/cache/openmediavault/archives  Translation-en
Ign:5 file:/var/cache/openmediavault/archives  Translation-en
Holen:6 file:/var/cache/openmediavault/archives  Translation-de_DE
Ign:6 file:/var/cache/openmediavault/archives  Translation-de_DE
Holen:3 file:/var/cache/openmediavault/archives  Packages
Ign:3 file:/var/cache/openmediavault/archives  Packages
Holen:4 file:/var/cache/openmediavault/archives  Translation-de
Ign:4 file:/var/cache/openmediavault/archives  Translation-de
Holen:5 file:/var/cache/openmediavault/archives  Translation-en
Ign:5 file:/var/cache/openmediavault/archives  Translation-en
Holen:6 file:/var/cache/openmediavault/archives  Translation-de_DE
Ign:6 file:/var/cache/openmediavault/archives  Translation-de_DE
Holen:3 file:/var/cache/openmediavault/archives  Packages
Ign:3 file:/var/cache/openmediavault/archives  Packages
Holen:4 file:/var/cache/openmediavault/archives  Translation-de
Ign:4 file:/var/cache/openmediavault/archives  Translation-de
Holen:5 file:/var/cache/openmediavault/archives  Translation-en
Ign:5 file:/var/cache/openmediavault/archives  Translation-en
Holen:6 file:/var/cache/openmediavault/archives  Translation-de_DE
Ign:6 file:/var/cache/openmediavault/archives  Translation-de_DE
Holen:3 file:/var/cache/openmediavault/archives  Packages
Ign:3 file:/var/cache/openmediavault/archives  Packages
Holen:4 file:/var/cache/openmediavault/archives  Translation-de
Ign:4 file:/var/cache/openmediavault/archives  Translation-de
Holen:5 file:/var/cache/openmediavault/archives  Translation-en
[... 572 lines cut ...]
10:49:24:699 enqueue event config/localtime for /config/
10:49:25:699 enqueue event config/localtime for /config/
10:49:26:699 enqueue event config/localtime for /config/
10:49:27:198 command queue EVENT_TIMEOUT, cmd: CMD_FEATURE, seq: 7
10:49:27:699 enqueue event config/localtime for /config/
10:49:28:699 enqueue event config/localtime for /config/
10:49:28:702 has feature: ethernet (0x04), status: 0x04
10:49:29:699 enqueue event config/localtime for /config/
10:49:30:198 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
10:49:30:198 device state timeout ignored in state 3
10:49:30:699 invalid current channel 0 (TODO)
10:49:30:699 enqueue event config/localtime for /config/
10:49:31:699 enqueue event config/localtime for /config/
10:49:32:699 enqueue event config/localtime for /config/
10:49:33:199 command queue EVENT_TIMEOUT, cmd: CMD_FEATURE, seq: 7
10:49:33:699 enqueue event config/localtime for /config/
10:49:34:699 enqueue event config/localtime for /config/
10:49:34:701 has feature: ethernet (0x04), status: 0x04
10:49:35:699 enqueue event config/localtime for /config/
10:49:36:198 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
10:49:36:198 device state timeout ignored in state 3
10:49:36:699 enqueue event config/localtime for /config/
```

## Additional context

Mimiix, 2020-12-22:
Can you try on port :8080 ?

tomillr, 2020-12-22:
Already tried the following without any success:

```
http://192.168.2.126:8080
http://192.168.2.126:8080/pwa/index.html
http://phoscon.de/pwa/login.html
```

tomillr, 2020-12-23:
So I've tried setting the whole thing up using a new user ("tomillr") instead of root.

Now I can finally access the Phoscon web interface, but unfortunately the Conbee stick **doesn't detect any devices**. In the advance settings there is **no Zigbee Chanel and network id** listed:

[image]

[image]

Also, when I reboot the system and access the Phoscon app again afterwards, it brings up the initial setup wizard again. Looks like the settings are not saved properly?

[image]

tomillr, 2020-12-28:
Just to verify that the stick itself is working, I've run the software in a docker container and am able to detect all the Zigbee devices. So it's definitely a software issue. Any idea where I should start to look for the issue?

tomillr, 2021-01-08:
Anyone? How can I go forward to identify the problem? Any assistance would be appreciated.

closed 2021-03-05 by github-actions[bot]