dresden-elektronik/deconz-rest-plugin #4003: only one items discovered by Alexa App

bilou95, 2020-12-22:
## Describe the question or issue you are having
  hi.

I have deployed recently a Raspbee II with Domoticz. All is working fine.
To try to link to ALexa, I do it directly on the Phoscon App.
pairing to Alexa was succesfull but only one device of the list were added.

I was even able to change the name and now my christmas tree can be power on by voice :-).

but I have one light with color and dimmer which is working fine in Domoticz and Phoscon but cannot link it to alexa. I try multiple time to discover but nothing change.

Any idea ?

Thanks.

## Screenshots

## Environment

 - Host system: (Raspberry Pi 4
 - Running method: (Raspbian / / Home Assistent deCONZ Add-on 
 - Firmware version: (2610700)
 - deCONZ version: (2.07.01)
 - Device: RaspBee II)

## deCONZ Logs

## Additional context

Mimiix, 2020-12-22:
@Smanar

bilou95, 2020-12-28:
Anyone to Help ?
I see no option on the Gateway website, only to make the gateway discoverables for 60 seconds and again, alexa only found one of the 2 ?

bilou95, 2020-12-29:
I finally remove the discovered item in alexa on my phone.
then I rename the 2 devices in Phoscon APP with shorter name and I launch again the discovery.
This time both were disovered.
I still have issue setting the color of the light but power ON/OFF is working .

I have 2 devices.
One wall plug switch on/off and one light with dimmer and color.

closed 2021-02-08 by Mimiix