dresden-elektronik/deconz-rest-plugin #4004: Define the used IP ddresses of the System.

talun2075, 2020-12-22:
## Describe the question or issue you are having
Is is possible to define the IP Address that the rest Plugin is used? default it takes all ip addresses of the system and use the port for all ip addresses.

I found a paramater for the http and websocket Port but not for the IP.

## Screenshots
[image]

## Environment

 - Host system: Windows
 - Running method:  Windows
 - Firmware version: (26680700)
 - deCONZ version: (2.07.01)
 - Device: ConBee II
 - Do you use an USB extension cable: (yes ) -- only relevant for ConBee I/II 
 - Is there any other USB or serial devices connected to the host system? If so: Keyboard Mouse

## deCONZ Logs

## Additional context

SwoopX, 2020-12-22:
This is currently not possible. Closing this in favor of #3948

closed 2020-12-22 by SwoopX

talun2075, 2020-12-26:
Perfect. Thanks.