dresden-elektronik/deconz-rest-plugin #4005: Cannot change channel

Monnem, 2020-12-22:
Hi,

Background: 
Existing and working Conbee (one) in a Qnap Docker. Network on channel 25 with 50+ nodes
Moved to another hardware to have enough power in combination with IOBroker.
Before able to restore old config, conbee selected automatically a channel != 25. 
Even with the hidden menu an restoring a config with channel 25 the channel will not be set.
Right now I'm not able to switch back to channel 25. 
## Steps to reproduce the behavior

Smanar, 2020-12-22:
>22:47:53:695 enqueue event config/localtime for /config/

I don't think this message in loop is normal ....

Just by curiosity, can you show logs when you try to change the channel using deconz ?

Monnem, 2020-12-22:
For sure. Here is the output with all checked options.
Clicked 21:21:30 in Phoscon on the change button after selecting channel 25. Round about 21:21:48 I get the error message, that channel could not be changed
https://pastebin.com/2TQ3My0J

Smanar, 2020-12-23:
BTW nothing to see, but you have generated so much api key, one time you have register to phoscon , put the page in bookmark just after the login, everytime you set a new login/pass, it generate a new api key.

And yes the channel change on the config, but not on the device. You have same problem on different machine ? If yes, have you tried to upload again the firmware ?

Monnem, 2020-12-23:
May this happens while resetting the gateway trying to get the disired channel. After that I restored always the old config. Is it possible to get rid of the not used api keys?
Yes, also in the docker and a real Win 10 Machine.
I installed yesterday the lastest firmware again. After that I saw some green connections, today in the morning they are all gone, now some of then are reestablished, but far away from the old functional state

Smanar, 2020-12-23:
Yep, you can delete them just using Curl command, on my side I m using a python script, not realy a problem.

Your channel problem is more worrying ...

manup, 2020-12-23:
To change the channel manually please try the following in deCONZ GUI:

* Press **Leave** button;
* Open *Edit &rarr; Network Settings*;
* Select only channel 25;
* Press **Save** button;  (the channel might jump around, ignore it for now)
* Close Dialog;
* Press **Join** button.

Note for this to work you need a valid configuration.

[image]

Important here:
* MAC Address = TC Address
* Security Mode: High – No Master but TC Link Key
* TC Link key as shown.

kugelkopf123, 2020-12-23:
Same Problem. 
I use a Raspbee Module. Tried to change it like @manup said but doesnt work.

Monnem, 2020-12-27:
Hope you all had nice XMas. :-)
I tried it again, but no luck. The only difference was APS Acks was not ticked. Ticked it, but no difference. Channel 15 AND 25 is ticked (15 is surounded with a grey box. When i uncheck 15 and save, 15 comes back again (see my screenshot in the first entry)

Monnem, 2020-12-28:
Finally I made it. I'll try to note all steps as I'm not sure which of them are really needed (I believe the downgrade to this specific version is the key):
Note: I downgraded earlier to the previous and previous version without success...

- fresh actual Win 10
- plugged Conbee. Driver has to be updated via Win driver Update  
- Installed latest Deconz from Homepage (8 I think)
- tried to switch channel -> failed
- downgraded firmware  as descripted on their page to 26330500
- first time tried to switch channel -> failed
- second time tried to switch channel -> SUCCESS (only ch25 was ticked even after "save" and "read"
- Plugged Conbee back to main installation (virtual ubuntu)
- Channel jumped back to 15! 
- first time tried to switch channel -> failed
- second time tried to switch channel -> SUCCESS (only ch25 was ticked even after "save" and "read"
- Connections to nodes came back very fast
- "Sing Hallejuja" ;-)

Status so far: Nearly all nodes (lights, plugs, Ikea dimmer, window sensors (aqara) and most of the temp sensors (aqara) are back, the motion sensors (Aqara) are right now a little bit picky...

- Next step will be upgrading the firmware to see, if my actual issues between deconz and IOBroker will be fixed.

Thx to all for the help. If I can offer logs or whatever to help, please come back to me

Smanar, 2020-12-28:
So you need a Firmware roolback to have a chance of succed ?

Monnem, 2020-12-28:
I think so. But it seems to be that 1 or 2 versions earlier is not enough...

kugelkopf123, 2020-12-29:
I gave up after downgrading 1 version. Didn't downgrade the fw from my Raspbee. Had no time to wait for a solution. So I connected all devices with ch11. :(

pboschi, 2021-01-04:
Same problem here with a Raspbee 1, can't change to channels 20 and 25, only 11 and 15. In Phoscon I get the same error, in Deconz I simply can't set 25 alone, at least one among 11 and 15 must be set. I'll try to revert to a previous version.

pboschi, 2021-01-04:
Ok so the only chance to solve the problem is a rollback to a previous firmware. This is the procedure if you are using Home Assistant:
- Stop deCONZ plugin
- Login via SSH to HA (follow HA instructions on how to do it)
- docker run -it --rm --entrypoint "/firmware-update.sh" --privileged --cap-add=ALL --device=/dev/ttyAMA0 -v /lib/modules:/lib/modules -v /sys:/sys marthoc/deconz
- Follow the on-screen instructions and input this string as update file (it will be downloaded): deCONZ_Rpi_0x26330500.bin.GCF  
- Wait for firmware flash to finish
- Start deCONZ plugin
- Go to deCONZ (VNC), press "Leave"; enable channel 25 (or whatever) and save. Then disable other channels and save again. Check with "read" that 25 is now the only selected channel; then "Join".

SAOPP, 2021-01-05:
Exactly the same problem with raspbee on the latest firmware 26390500, I can set the 15th channel, but I can’t set the 20th or 25th. The instructions above also didn't work. Is it really the firmware?

pboschi, 2021-01-05:
I can sistematically reproduce it by upgrading to 26390500 and solve it by downgrading to 26350500. I get this when i upgrade:

21:00:33:596 channel is 25 but should be 15, start channel change
21:00:33:597 Skip automatic channel change, TODO warn user
21:00:43:596 channel is 25 but should be 15, start channel change
21:00:43:597 Skip automatic channel change, TODO warn user
21:00:53:595 channel is 25 but should be 15, start channel change
21:00:53:596 Skip automatic channel change, TODO warn user

SAOPP, 2021-01-05:
Thw same homie:

```
,00:14:52:218 Current channel 15
,00:14:52:240 Device TTL 4742 s flags: 0x7
,00:14:53:170 channel is 15 but should be 25, start channel change
,00:14:53:172 Skip automatic channel change, TODO warn user
,00:15:03:172 Skip automatic channel change, TODO warn user
,00:15:03:170 channel is 15 but should be 25, start channel change
,00:15:13:170 channel is 15 but should be 25, start channel change
,00:15:13:172 Skip automatic channel change, TODO warn user
,00:15:23:170 channel is 15 but should be 25, start channel change
,00:15:23:172 Skip automatic channel change, TODO warn user

```

SAOPP, 2021-01-05:
@pboschi so I correctly understood u, I need to downgrade now and change the channel, and after it go to update again to the latest firmware, and after this steps, 25th channel is saved?

SwoopX, 2021-01-06:
I'll see if I can reproduce that somehow in my test network tomorrow.

SAOPP, 2021-01-08:
At the moment I cannot check the channel change on conbee2, but I get the impression that only raspbee has the current incomprehensible problems with the transfer to the 25th channel.

jh83, 2021-01-18:
Same issue for me.

Have been running conbee with deConz for around 2+ years and have 50 nodes installed. The day before yesterday it all stopped, and it seemed as if the conbee was not talking to any of the nodes. All the "links" between the nodes and the conbee device was missing when checking in the deConz GUI.

After some time, i noticed that it looked like the conbee was no longer running on channel 25. The gui said that it was unable to change back to channel 25, and at one time it worked, but then the deconz GUI still said that it was running on channel 11 or 15.

After downgrading the FW two versions i was able to change back to channel 25, and the network started to rebuild it self.

I have no idea why it all of a suddenly stopped and changed channel by it self from channel 25 which i have been running on for about 2 years...

SAOPP, 2021-01-18:
Downgrade to what the version?

SAOPP, 2021-01-18:
> After downgrading the FW two versions i was able to change back to channel 25, and the network started to rebuild it self.
Hey guys @manup @SwoopX please сould you tell in more detail if you have this information, what happened on the latest firmware, why is it impossible to set the channel above the 15th? 

For example, for me this is very important now, another conbee2 is coming to me to distribute devices in the house, and I really need to put channel 25 and transfer half of the devices there, I'm afraid of the prospect of not possible to set the channel. It's really issue with latest firmware for raspbee and conbee?

jh83, 2021-01-18:
> Downgrade to what the version?

deCONZ_Rpi_0x26350500.bin.GCF
And after downgrading to that version, I performed the steps described above where you uncheck all channels except for channel 25 thru the gui. 

After the downgrade, it actually saved the channel 25 without reverting to what ever checkbox that was selected before.

SAOPP, 2021-01-18:
and u stayed on the old firmware after it?

jh83, 2021-01-18:
> and u stayed on the old firmware after it?

So far, yes. Since I don't know what made it stopped working in the first place....

pboschi, 2021-01-18:
If you upgrade again to latest version it seems to work, but after some minutes it switches back to channel 11 o 15.

SAOPP, 2021-01-19:
I understood, thanks, the behavior is the same as when changing channel on a new firmware, the devil's shit.

SAOPP, 2021-01-19:
My current one of conbee 2 sticks work on 25 channel now at 264A0700 firmware, and I set channel at summer, and all work fine.

CherAlban, 2021-01-20:
I am running a ConBee2 with 26680700 firmware on Windows 10. I changed from channel 25 to 20 months ago, but when checking the debug log I am also seeing this: 

13:42:18:663 channel is 20 but should be 25, start channel change
13:42:18:664 Skip automatic channel change, TODO warn user

My network in general is working very unreliable, with lots of messages (e.g. Xiaomi door contacts) getting lost.

Tscherno, 2021-01-29:
> I am running a ConBee2 with 26680700 firmware on Windows 10. I changed from channel 25 to 20 months ago, but when checking the debug log I am also seeing this:
>
> 13:42:18:663 channel is 20 but should [...]

I'm currently seeing the excact same issue. I changed the channel today by mistake and now cannot change it back to 15. I plugged the stick into my Windows PC and it seems to stick in Deconz but log still gives 

16:45:47:016 channel is 15 but should be 20, start channel change
16:45:47:017 Skip automatic channel change, TODO warn user

Anything i can do?

Smanar, 2021-01-29:
ATM, I think the only workaround is downgrading the firmware, according to previous posts

SAOPP, 2021-01-30:
Conbee 2 with the latest at current moment firmware version `26680700` is changing its own channel without issues also.

ichbaum, 2021-02-22:
Same problem with 26330500. Only 11, and 15 work. Worked after the downgrade.

SAOPP, 2021-02-22:
> Same problem with 26330500. Only 11, and 15 work. Worked after the downgrade.

Try `26680700` and let some feedback.

ichbaum, 2021-02-24:
> > Same problem with 26330500. Only 11, and 15 work. Worked after the downgrade.
> 
> Try `26680700` and let some feedback.
Forget to mention that i use ConBee Stick 1.

SAOPP, 2021-02-24:
ah ok, thanks for feedback of issue

jjlawren, 2021-03-02:
I was unable to use channel 25 with a RaspBee, but reverting to deCONZ_Rpi_0x26350500 now allows me to select it.

ntcong, 2021-03-07:
Can confirm that the latest firmware automatically check channel 11 after Save, but just downgrade to deCONZ_Rpi_0x26350500 uncheck it (I don't have to uncheck and Save again)

kugelkopf123, 2021-04-05:
Still exists...

mbooij, 2021-04-24:
> Same issue for me.
>
> Have been running conbee with deConz for around 2+ years and have 50 nodes installed. The day before yesterday it all stopped, and it seemed as if the conbee was not talking to [...]

Had exactly the same experience. Was able to get my network working again based on this post. Thats a lot. Was about to throw away my RaspBee and buy a ConBee II. Any word on a fix for firmware 0x26390500?

pboschi, 2021-05-16:
How to reopen this issue?

elratt0r, 2021-08-01:
Still exists... Thanks for pointing to the still working firmware version here!

SwoopX, 2021-08-01:
We'll make a new issue for this to keep things clean

closed 2021-08-01 by SwoopX