dresden-elektronik/deconz-rest-plugin #4006: LELLKI WP33, aka E220-KR4N0Z0-HA

yruss972, 2020-12-22:
## Device

- Product name: Smart Power Strip WP33
- Manufacturer: <Empty>
- Model identifier: E220-KR4N0Z0-HA
- Device type :
  - Light
  - Other: Shows as On/Off light x4 but really it's 4 outlets  

## Screenshots

 - Basic Cluster attributes in the Cluster Info panel.
[image]

  - Node Info panel
[image]

  - Endpoints and clusters of the node
[image]

SwoopX, 2020-12-23:
Can you please read the manufacturer name from basic cluster as well please? Other than that and a small addition to ensure on/off state is reported, the device should be fully functional out of the box.

yruss972, 2020-12-23:
Nothing comes up for the manufacturer. It's just empty.
The device works partially out of the box.
If the 'lights' are assigned to a group in phoscon they can be switched on and off. Trying to turn them on without a group results in them cycling on and off several times.
Renaming them in phoscon has a similar effect.

SwoopX, 2021-01-17:
@yruss972 Can you please give me a screenshot where and how exactly you try to switch them in Phoscon?

cagnulein, 2021-01-19:
i guess i have a little sister of yours @yruss972 

[image]

closed 2021-03-05 by github-actions[bot]