dresden-elektronik/deconz-rest-plugin #4009: RaspBee and SSD Interference

vantino, 2020-12-22:
## Describe the bug

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: (Raspberry Pi / PC / NAS)
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
 - Firmware version: (26xxyy00)
 - deCONZ version: (2.xx.yy)
 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II)
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs

## Additional context

Mimiix, 2020-12-23:
This is known. Please read the wiki.

Also, this is not how the template works.

closed 2020-12-23 by Mimiix

vantino, 2020-12-23:
ok - can i get a link to your wiki ?

ebaauw, 2020-12-23:
An SSD in itself shouldn't be causing any issues.  How do you connect the SSD to the Pi4?  USB3 is a known source of interference.

Mimiix, 2020-12-23:
@vantino https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Zigbee-Error-Codes-in-the-Log

vantino, 2020-12-23:
danke

unwrittenlaw, 2021-01-14:
sorry for the stupid question, but can i never boot the pi from an ssd and use the raspbee i? or does it also work with an extension cable? or do I have to take a usb2 case?

i would use pi4+ssd+raspbee i+home assistant 64bit. thanks a lot.

Mimiix, 2021-01-14:
@unwrittenlaw you need to make sure the USB interference can't reach it.