dresden-elektronik/deconz-rest-plugin #4010: Sending commands to Thermostat TS0601 are not working when display is off

Alex66955, 2020-12-23:
## Describe the question or issue you are having

- Product name: Tuya Smart TRV HY369RT.
- Manufacturer: _TZE200_ckud7u2l.
- Model identifier: TS0601.

I  bought 6 of these thermostats and integrate them with deconz in my "home assistant". I did the binding via Phoscon Sensor without any problem.
But when I try to change the temperature set-point via "home assistant" the thermostat did not react. The only way I can set the "temperature" is to do it manually or to press a button on the termostat (display turn on) and then repeating setting the temperature via "home assistant".

It seems that the commands are only received and accepted when the thermostat is not in a kind of sleeping mode.

## Screenshots
[image]

## Environment

 - Host system: Raspberry Pi
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: 26660700
 - deCONZ version: 2.07.01 / 8.12.2020
 - Device: ConBee II
 - Do you use an USB extension cable: no

## deCONZ Logs

Smanar, 2020-12-23:
We already have this issue https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3452#issuecomment-745476261

Can you check with another device or retry to include it (2 inclusion without deleting previous entry)

Alex66955, 2020-12-24:
> We already have this issue [#3452 (comment)](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3452#issuecomment-745476261)
> 
> Can you check with another device or retry to include it (2 inclusion without deleting previous entry)

All 4 devices have the same problem. You mean I should rebind them without deleting the present binding?
Can I provide additional information for support?

Smanar, 2020-12-24:
Take care "bind" have a special meaning for zigbee ^^
Try to re-pair them without deleting the previous one.

Else if it don't work, go in deconz / help to view log, "info" and "info_l2".
- awake your device (manualy)
- send the request just after

You will have in log some lines starting by "debug tuya", perhaps we will see a special to command to wake up the device before sending the request.

Alex66955, 2020-12-25:
Okay, thanks for the additional information. 
I did the re-paring but without any changes how the thermostat behave.

I trieto repeat what I observed until now:

- Changing the temperature at the thermostat 
  - Does not update the temperature on "Home Assistant"
  - (in rare situations it will update something but not the correct set temperature)
- Changing the temperature in "Home Assistant" (thermostat is sleeping - display is off)
  - Thermostat does not set the requested temperature
- Changing the temperature in "Home Assistant" (thermostat is not sleeping - display is on)
  - Thermostat does update the requested temperature as expected

Attached the logs (I was not able to connect via VNA to export the logs as text file. The requested VNC password was not accepted - I am on it..)

Steps what the logs should observed:
- I activated the display (display on)
- I send a temperature request to the thermostat which was accepted as expected

[image]
[image]
[image]

Smanar, 2020-12-25:
ok so

>Changing the temperature at the thermostat 

On the first log we can see "tuya debug ... 0x02 0x02" it mean deconz is receiving the setpoint 21,5, and it send the websocket with the value, so you need to see it in HA ...

But for the 2 others, I dont see return from the device ( no "tuya debug") so the device don't answer, for both. HA is updated because deconz do it, but it s not because of device.
And on the third log, the device don't answer at all.

Alex66955, 2020-12-25:
> Okay, thanks for the additional information.
> I did the re-paring but without any changes how the thermostat behave.
>
> I trieto repeat what I observed until now:
>
> * Changing the temperature at the [...]

Some updates for the part "Changing the temperature at the thermostat":

-  The update works, I have to wait some minutes until the manual changed temperature was updated on HA

vegetate7, 2020-12-25:
> @vegetate7 it s your device too ?

Yes, one of my devices is same:
```
        "manufacturername": "_TZE200_ckud7u2l",
        "modelid": "TS0601",
        "name": "TS0601",
        "state": {
            "lastupdated": "none",
            "lowbattery": null,
            "on": false,
            "temperature": 2550,
            "valve": 0
        },
        "type": "ZHAThermostat",
        "uniqueid": "bc:33:ac:ff:fe:6e:ba:40-01-0201"

```

But look on the starting picture: one of 4 thermostats have different MAC prefix.
And WRT logs: see 'delay request' lines. In my setup this mean device will not receive command.

Smanar, 2020-12-25:
>And WRT logs: see 'delay request' lines. In my setup this mean device will not receive command.
Yep, but if I m right it s the case "Changing the temperature in "Home Assistant" (thermostat is not sleeping - display is on)" and this one is working.

@Alex66955 do you have the log return (starting with "tuya debug") just when you awake the device ?
Perhaps we can found a special command to awake it before sending the command.

Alex66955, 2020-12-25:
> Yep, but if I m right it s the case "Changing the temperature in "Home Assistant" (thermostat is not sleeping - display is on)" and this one is working.
>
> @Alex66955 do you have the log return [...]

I tried to record the right 2 times section

**First try:**

```
14:16:27:490 poll node 7c:b0:3e:aa:0a:0b:9c:7a-03
14:16:27:492 Poll light node Switch07
14:16:27:543 Poll APS request to 0x7CB03EAA0A0B9C7A cluster: 0x0006 dropped, values are fresh enough
14:16:27:665 DB save zll database items 0x00000001
14:16:27:669 DB sql exec REPLACE INTO nodes (id, state, mac, name, groups, endpoint, modelid, manufacturername, swbuildid, ritems) VALUES ('14', 'normal', '58:8e:81:ff:fe:ae:c4:11-01', 'Lampe01', '65520', '1', 'TS0505A', 'Heiman', '', '{"attr/id":"14","attr/lastannounced":null,"attr/lastseen":"2020-12-25T13:16Z","attr/manufacturername":"Heiman","attr/modelid":"TS0505A","attr/name":"Lampe01","attr/swversion":null,"attr/type":"Extended color light","attr/uniqueid":"58:8e:81:ff:fe:ae:c4:11-01","config/colorcapabilities":0,"config/ctmax":65279,"config/ctmin":0,"state/alert":null,"state/bri":254,"state/colormode":"ct","state/ct":424,"state/effect":"1","state/hue":0,"state/on":true,"state/reachable":true,"state/sat":254,"state/x":21167,"state/y":21561}')
14:16:27:689 DB saved in 24 ms
14:16:27:692 enqueue event config/localtime for /config/
14:16:28:253 Tuya : Schedule command
14:16:28:255 Tuya debug 4 : Address 0xBC33ACFFFE45849B Payload 1ff57100001286001408000f0b1e0f0c1e0f111e1416000f
14:16:28:259 Tuya debug 5 : Status: 31 Transid: 245 Dp: 113 (0x00,0x71) Fn: 0 Data 0
14:16:28:261 enqueue event config/schedule for /sensors/6
14:16:28:264 enqueue event config/schedule for /sensors/6
14:16:28:266 enqueue event config/schedule for /sensors/6
14:16:28:269 enqueue event config/schedule for /sensors/6
14:16:28:270 enqueue event config/schedule for /sensors/6
14:16:28:272 enqueue event config/schedule for /sensors/6
14:16:28:301 Websocket 172.30.32.1:59518 send message: {"config":{"heatsetpoint":1950,"mode":"auto","offset":-100,"on":true,"preset":"manual","reachable":true,"schedule":{"W124":[{"heatsetpoint":20,"localtime":"T06:00"},{"heatsetpoint":15,"localtime":"T08:00"},{"heatsetpoint":15,"localtime":"T11:30"},{"heatsetpoint":15,"localtime":"T12:32"},{"heatsetpoint":20,"localtime":"T17:30"},{"heatsetpoint":15,"localtime":"T22:00"}],"W3":[{"heatsetpoint":20,"localtime":"T134:00"},{"heatsetpoint":15,"localtime":"T08:00"},{"heatsetpoint":15,"localtime":"T11:30"},{"heatsetpoint":15,"localtime":"T12:30"},{"heatsetpoint":20,"localtime":"T17:30"},{"heatsetpoint":15,"localtime":"T22:00"}]}},"e":"changed","id":"6","r":"sensors","t":"event","uniqueid":"bc:33:ac:ff:fe:45:84:9b-01-0201"} (ret = 721)
14:16:28:317 discard sensor config push for config/schedule (already pushed)
14:16:28:442 discard sensor config push for config/schedule (already pushed)
14:16:28:447 don't create binding for attribute reporting of sensor Thermostat
14:16:28:451 don't create binding for attribute reporting of sensor Thermostat
14:16:28:454 don't create binding for attribute reporting of sensor Thermostat
14:16:28:473 discard sensor config push for config/schedule (already pushed)
14:16:28:488 discard sensor config push for config/schedule (already pushed)
14:16:28:500 poll node 7c:b0:3e:aa:0a:0c:25:8e-03
14:16:28:504 Poll light node Switch08
```

**Second try:**

```
14:23:41:665 Daylight now: solarNoon, status: 170, daylight: 1, dark: 0
14:23:41:669 enqueue event config/localtime for /config/
14:23:41:672 poll node 7c:b0:3e:aa:0a:0c:26:ae-03
14:23:41:675 Poll light node Switch04
14:23:41:677 Idle timer triggered
14:23:41:680 don't create binding for attribute reporting of sensor Thermostat
14:23:41:683 Force binding of attribute reporting for node Thermostat
14:23:41:971 Poll APS request to 0x7CB03EAA0A0C26AE cluster: 0x0006 dropped, values are fresh enough
14:23:42:665 enqueue event config/localtime for /config/
14:23:42:840 poll node 58:8e:81:ff:fe:ae:c4:11-01
14:23:42:843 Poll light node Lampe01
14:23:42:910 Poll APS request to 0x588E81FFFEAEC411 cluster: 0x0006 dropped, values are fresh enough
14:23:43:221 don't create binding for attribute reporting of sensor Thermostat
14:23:43:331 Poll APS request to 0x588E81FFFEAEC411 cluster: 0x0008 dropped, values are fresh enough
14:23:43:440 read attributes of 0x588E81FFFEAEC411 cluster: 0x0300: [ 
14:23:43:442 0x0000 
14:23:43:443 0x0001 
14:23:43:445 0x0003 
14:23:43:446 0x0004 
14:23:43:448 0x0007 
14:23:43:449 0x0008 
14:23:43:450 ]
14:23:43:451 add task 59426 type 19 to 0x588E81FFFEAEC411 cluster 0x0300 req.id 15
14:23:43:453 Poll APS request 15 to 0x588E81FFFEAEC411 cluster: 0x0300
14:23:43:558 Poll APS confirm 15 status: 0x00
14:23:43:560 Erase task req-id: 15, type: 19 zcl seqno: 103 send time 0, profileId: 0x0104, clusterId: 0x0300
14:23:43:582 Node data 0x588e81fffeaec411 profileId: 0x0104, clusterId: 0x0300
14:23:43:586 0x588E81FFFEAEC411: update ZCL value 0x01/0x0300/0x0000 after 0 s
14:23:43:587 0x588E81FFFEAEC411: update ZCL value 0x01/0x0300/0x0001 after 0 s
14:23:43:593 0x588E81FFFEAEC411: update ZCL value 0x01/0x0300/0x0003 after 0 s
[... 2 lines cut ...]
14:23:43:597 0x588E81FFFEAEC411: update ZCL value 0x01/0x0300/0x0008 after 0 s
14:23:43:664 enqueue event config/localtime for /config/
14:23:43:666 poll node 7c:b0:3e:aa:0a:0c:2b:2a-03
14:23:43:669 Poll light node Switch05
14:23:43:671 Idle timer triggered
14:23:43:673 Force read attributes for ZHATemperature SensorNode MS02
[... cut ...]
14:23:44:244 don't create binding for attribute reporting of sensor Thermostat01
14:23:44:664 enqueue event config/localtime for /config/
14:23:44:698 Master: read param with arg 0x19
14:23:44:881 poll node 7c:b0:3e:aa:0a:0c:18:94-03
14:23:44:889 Poll light node Switch06
14:23:44:922 Current channel 25
14:23:44:942 Poll APS request to 0x7CB03EAA0A0C1894 cluster: 0x0006 dropped, values are fresh enough
14:23:44:967 Device TTL 2400 s flags: 0x7
14:23:44:985 Outgoing frame counter 770685 (0x000BC27D)
```

Smanar, 2020-12-25:
On the first one, it s a schedule command, so I don't thnk it s the good one.
But on the second one it s an "auto lock" command 0x01 0x74, not implemented in deconz yet. There is somethng about this feature on the manual ?

jul1an353, 2020-12-27:
I had the same issue and needed to update the firmware of the thermostat. 
https://github.com/Koenkk/zigbee2mqtt/issues/3821#issuecomment-696011507

Smanar, 2020-12-27:
And there is a way to find the firmware if we haven't the original gateway ?

Alex66955, 2020-12-28:
> On the first one, it s a schedule command, so I don't thnk it s the good one.
> But on the second one it s an "auto lock" command 0x01 0x74, not implemented in deconz yet. There is somethng about this feature on the manual ?
> 
> And one some device the "heatsetpoint" can take hours before change, are you sure It don't change on your ?

It is not fully reproducible. Sometimes it updates the temperature Set-Point and sometimes not. I already tested it and waited several hours but the behavior of the thermostat is not very reliable.

Attached the manual but I found nothing interesting:
[image]
[image]

Alex66955, 2020-12-28:
> And there is a way to find the firmware if we haven't the original gateway ?

Maybe there is a way to record the network traffic during a firmware flashing process with the origin gateway?

Smanar, 2020-12-28:
IDK you have something called "auto lock" on the doc, but with the description it s the children lock for me, and from my memory it s already working on this device ?

>Maybe there is a way to record the network traffic during a firmware flashing process with the origin gateway?

Yep we are using it ATM for Legrand firmware, but my question was to know if someone already have do that ?

Alex66955, 2020-12-30:
What do I need to record the firmware except the origin gateway? I would invest the equipment to get it working,

Alex66955, 2020-12-30:
> I used for update the original gateway. in this post you find the firmware update file. [Koenkk/zigbee2mqtt#3821 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/3821#issuecomment-702825657)
> update with zigbee2mqtt is also possible.

Thanks for this information update. 

First, I updated one thermostat via deconz:
- Connected to docker container via portainer
- download firmware file via "wget" command to the correct folder
- Triggered the update process via deconz with the downloaded file
- After update version changed from 0x53 -> 0x55

What I observed since update:
- Thermostat gets the temperature set-point even if display is off
- Thermostat informs about manually set temperature and this information received back to HASS

Now I will update the others and I will test the next days how the the firmware behaves.

Seems pretty nice until now :)

Smanar, 2020-12-30:
@jul1an353 big thx for tips (and the file)

Alex66955, 2021-01-03:
After several days of testing everything works as expected.

Thank you all!

closed 2021-01-03 by Alex66955