dresden-elektronik/deconz-rest-plugin #4012: Tuya Zigbee 4 gangDevice name

Simonk007, 2020-12-23:
Tuya Zigbee 4 gang

- Product name: Wireless Scene Switch 20/36
- Manufacturer: Tuya
- Model identifier: As per deCONZ GUI Basic cluster.
- Device type : 4 switch 

Feature

- Name：4 Gang Zigbee smart switch
- Compatible：Tuya,Smartlife,Smartthing
- Operability：Control other Wifi or Zigbee intelligent devices and scene.
- Push the button：single click,double click,long press

Specification

Communication:Zigbee 802.15.4
Button lifecycle:500k
Dimension:86*86*13mm
Material:Mechanical Push On/Off Button
Input power:CR 2430 button battery（Need to buy it yourself）
Battery life:730days
Standby current:10uA
Receive sensitivity:-101dBm
Indicator light:LED
Standby consumption:<=20mW
Transmit power:19dBm
Operating frequency:2.400~2.484GHz

https://www.aliexpress.com/item/1005001302383925.html?spm=a2g0o.productlist.0.0.6570621ddJVL11&algo_pvid=5b5846c7-c06c-4c98-9c0d-8ed2526e1e49&algo_expid=5b5846c7-c06c-4c98-9c0d-8ed2526e1e49-0&btsid=0bb0623d16086744509461424e4973&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_

[image]

closed 2020-12-23 by close-issue-app[bot]

dmontes46, 2021-02-18:
is this working with deconz ?

AssetBurned, 2021-03-29:
For me it is detected in Phoscon "2.09.03 / 23/12/2020" running on a Raspberry 2B with RaspBee 1 and Homebridge 1.3.4.
But every button click is detected twice.